nebrius/raspi-io #12: Cannot run PWM0 and PWM1 Concurrently

virgilvox, 2015-02-13:
When running

```

var raspi = require('raspi-io');
var io = new raspi();
io.on('ready', function() {
 var servo = 24;
 var servo2 = 26;

 this.pinMode(servo, io.MODES.PWM);
 this.pinMode(servo2, io.MODES.PWM); 

 this.analogWrite(servo, 255);
  this.analogWrite(servo2, 255);
});

```

I get

```
/home/pi/raspi/node_modules/raspi-io/node_modules/raspi-pwm/lib/index.js:19
      throw new Error('Attempted to write to a destroyed peripheral');
            ^
Error: Attempted to write to a destroyed peripheral
    at PWM.write (/home/pi/raspi/node_modules/raspi-io/node_modules/raspi-pwm/lib/index.js:19:13)
    at Raspi.Object.defineProperty.value [as analogWrite] (/home/pi/raspi/node_modules/raspi-io/lib/index.js:307:28)
    at Raspi.<anonymous> (/home/pi/raspi/test.js:8:7)
    at Raspi.emit (events.js:92:17)
    at /home/pi/raspi/node_modules/raspi-io/lib/index.js:237:10

``

bryan-m-hughes, 2015-02-14:
It should be fixed now with the release of 1.1.0 of raspi-pwm. I left my B+ at home, so I can't test at the moment, but I'm pretty confident in this fix. You may need to remove your existing install of raspi-io and reinstall it to get it to pick up the new dependency.

If you could, please test it and let me know if it works.

bryan-m-hughes, 2015-02-16:
Confirmed fixed.

closed 2015-02-16 by bryan-m-hughes