nebrius/raspi-io #14: Reading back board.io.pins mode after setting the mode does not reflect proper mode

MrYsLab, 2015-02-21:
In my application I am setting a pin pin mode with: board.pinMode(pin, five.Pin.OUTPUT);

Before I go to do a digital write, I check the to make sure that the pin mode was set to output, before committing the write. I do so by checking the board.pin mode. In previous releases (I am not sure when this broke) this worked, but apparently, the pin mode is no longer being reflected properly.

I do this for all pins and modes, and the mode does not seem to be set.

I am using the latest and greatest and j5 and raspi-io (updated today). The code works fine for Arduino and BeagleBone Black, but no longer works on RPi.

bryan-m-hughes, 2015-02-22:
Fixed with v2.2.1. That was a really embarrassing bug...beginner JS error. I actually made this mistake (simplified):

```
for (var i in obj) {
  asyncOp(function () {
    obj[i].doStuff();
  });
}
```

closed 2015-02-22 by bryan-m-hughes

MrYsLab, 2015-02-22:
Thanks for the quick fix. You have just made one of my user's in France very happy ;-).

bryan-m-hughes, 2015-02-22:
:)