nebrius/raspi-io #15: Exceptions when opening up a RPi 2 Session

MrYsLab, 2015-02-22:


bryan-m-hughes, 2015-02-22:
Looking at the stack trace and code, that's a johnny-five bug, not a raspi-io bug (I don't use es6-shim myself). Can you open a bug report there?

FWIW I was testing on an RPi2 yesterday and I didn't encounter this bug.

As a workaround, they can run ```npm install es6-shim``` in the project directory.

closed 2015-02-22 by bryan-m-hughes

MrYsLab, 2015-02-22:
Just curious, why wouldn't I see this behavior on an Arduino installation if the problem is within johnny-five?

To bring up raspi-io, here is what I do:
        var raspi = require('raspi-io');

        //noinspection JSCheckFunctionSignatures
        board = new five.Board({
            io: new raspi()
        });

For Arduino, I do:
board = new five.Board();

rwaldron, 2015-02-22:
What version of node is running on the Raspberry Pi?

mfreard, 2015-02-22:
Hi
i'm the fench user.
The version is v0.10.36 on my Rpi2.
Thanks
Michel

bryan-m-hughes, 2015-02-22:
Also, can you test running the code on an arduino connected to your pi? All you need to do is change

```
board = new five.Board({
  io: new raspi()
});
```

to 

```
board = new five.Board();
```

This will help test your install of johnny-five.

MrYsLab, 2015-02-22:
I am not following you. What do you mean by "arduino connected to your pi"? How is it connected?

MrYsLab, 2015-02-22:
@mfreard Are you running an Arduino in conjunction with RPi when you get the problem, or is this the same configuration you ran on the B+?

bryan-m-hughes, 2015-02-22:
@MrYsLab the Raspberry Pi is a full fledged computer with USB ports, so you can use the Raspberry Pi the same way you would use a laptop. So you connect the Arduino to the Raspberry Pi with a USB cable.

bryan-m-hughes, 2015-02-22:
Also, just to make sure, @mfreard, have you tried completely deleting your ```node_modules``` folder and reinstalling johnny-five?

mfreard, 2015-02-22:
i installed the same configuration i ran on my B+.
Could you remember  me the exact command to delete the folder?
thanks

MrYsLab, 2015-02-22:
if you cd into the xipi directory, you can then:
sudo rm -rf node_modules

After that you can install each modue individually:

sudo npm install node-gyp
sudo npm install johnny-five
sudo npm install raspi-io
sudo npm install websocket
sudo npm install open

mfreard, 2015-02-23:
I Installed each module and nothing has changed

bryan-m-hughes, 2015-02-23:
@mfreard did you test connecting an arduino to your raspberry pi?

bryan-m-hughes, 2015-02-23:
Also, can you post the output of ```npm list```?

mfreard, 2015-02-23:
I didn't test with arduino
the results of npm list
pi@raspberrypi ~ $ cd xipi
pi@raspberrypi ~/xipi $ npm list
/home/pi/xipi
├─┬ gyp@0.5.0
│ ├─┬ gyp-conditions@0.0.4
│ │ └── UNMET DEPENDENCY ometajs@~3.2.4
│ ├── gyp-expansions@0.0.2
│ ├─┬ gyp-load@0.5.0
│ │ └─┬ jsonic-ometajs@0.0.0
│ │   ├── UNMET DEPENDENCY json-ometajs@0.0.1
│ │   └── UNMET DEPENDENCY ometajs@~3.2.4
│ └── gyp-merge@0.2.1
├─┬ johnny-five@0.8.43
│ ├─┬ array-includes@1.0.6
│ │ ├── UNMET DEPENDENCY define-properties@^1.0.1
│ │ └── UNMET DEPENDENCY es-abstract@^1.1.0
│ ├── color-convert@0.5.2
│ ├── colors@1.0.3
│ ├── descriptor@0.1.0
│ ├── ease-component@1.0.0
│ ├── es6-shim@0.25.3
│ ├─┬ firmata@0.4.0
│ │ ├── UNMET DEPENDENCY browser-serialport@*
│ │ └── UNMET DEPENDENCY object-assign@~1.0.0
│ ├─┬ galileo-io@0.8.1
│ │ ├── UNMET DEPENDENCY es6-promise@latest
│ │ └── UNMET DEPENDENCY remapped@~0.2.1
│ ├── lodash@3.3.0
│ ├── nanotimer@0.3.1
│ ├─┬ serialport@1.5.0
│ │ ├── UNMET DEPENDENCY async@0.9.0
│ │ ├── UNMET DEPENDENCY bindings@1.2.1
│ │ ├── UNMET DEPENDENCY debug@^2.1.1
│ │ ├── UNMET DEPENDENCY nan@~1.6.2
│ │ ├── UNMET DEPENDENCY node-pre-gyp@0.6.x
│ │ ├── UNMET DEPENDENCY optimist@~0.6.1
│ │ └── UNMET DEPENDENCY sf@0.1.7
│ └── temporal@0.4.0
├── node@0.0.0
├── open@0.0.5
├─┬ raspi-io@2.2.1
│ ├─┬ raspi@1.3.0
│ │ ├── nan@1.6.2
│ │ └── raspi-wiringpi@1.0.4
│ ├── raspi-board@2.1.0
│ ├─┬ raspi-gpio@1.2.0
│ │ ├── nan@1.6.2
│ │ └── raspi-peripheral@1.1.0
│ └─┬ raspi-pwm@1.3.0
│   ├── nan@1.6.2
│   └── raspi-peripheral@1.1.0
└─┬ websocket@1.0.17
  ├─┬ debug@2.1.1
  │ └── ms@0.6.2
  ├── nan@1.0.0
  └─┬ typedarray-to-buffer@3.0.1
    └── is-typedarray@0.0.0

npm ERR! missing: ometajs@~3.2.4, required by gyp-conditions@0.0.4
npm ERR! missing: json-ometajs@0.0.1, required by jsonic-ometajs@0.0.0
npm ERR! missing: ometajs@~3.2.4, required by jsonic-ometajs@0.0.0
npm ERR! missing: define-properties@^1.0.1, required by array-includes@1.0.6
npm ERR! missing: es-abstract@^1.1.0, required by array-includes@1.0.6
npm ERR! missing: browser-serialport@*, required by firmata@0.4.0
npm ERR! missing: object-assign@~1.0.0, required by firmata@0.4.0
npm ERR! missing: async@0.9.0, required by serialport@1.5.0
npm ERR! missing: bindings@1.2.1, required by serialport@1.5.0
npm ERR! missing: debug@^2.1.1, required by serialport@1.5.0
npm ERR! missing: nan@~1.6.2, required by serialport@1.5.0
npm ERR! missing: node-pre-gyp@0.6.x, required by serialport@1.5.0
npm ERR! missing: optimist@~0.6.1, required by serialport@1.5.0
npm ERR! missing: sf@0.1.7, required by serialport@1.5.0
npm ERR! missing: es6-promise@latest, required by galileo-io@0.8.1
npm ERR! missing: remapped@~0.2.1, required by galileo-io@0.8.1
npm ERR! not ok code 0

MrYsLab, 2015-02-23:
Very different from what I see on the clean load I did on my B+. Something is very strange. If you don't have anything you need to save on the SD card for the Pi2, perhaps it is best to do a complete reinstall.
Reload the OS, load node 0.10.36, and reinstall xipi.

@bryan-m-hughes 
Here is the install script that I use to create diretories, load my stuff, and do npm installs. It worked for the B model 2, and B+. Is there anything special about the 2?

```
#!/bin/sh
# XiPi install script
echo "Installing XiPi ..."
echo
echo "creating xipi directory"
mkdir -p ~/xipi

echo "change permissions on xipi.sh"
sudo chmod ugo+x ./xipi.sh

echo "copying xipi.sh"
cp ./xipi.sh ~/xipi/.

echo "copying xiserver.js"
cp ../xiserver/xiserver.js ~/xipi/.

echo "Installing npm modules for XiPi"

echo "loading node-gyp"
npm install --prefix ~/xipi node-gyp

echo "loading johnny-five"

npm install --prefix ~/xipi johnny-five

echo "loading raspi-io"

npm install --prefix ~/xipi raspi-io

echo "loading websocket"

npm install --prefix ~/xipi websocket

echo "loading open"
npm install --prefix ~/xipi open

echo "!!!! Install Complete !!!"
echo
echo
echo "To start XiPi: "
echo "  cd ~/xipi"
echo "  bash xipi.sh"
echo
echo "You may be prompted for sudo priveleges".
```

mfreard, 2015-02-23:
I'll do a complete reinstall tonight.

bryan-m-hughes, 2015-02-23:
OK, looking at your list, I see:

```
├─┬ johnny-five@0.8.43
...
│ ├── es6-shim@0.25.3
```

So it's installed. This means node isn't finding it because...reasons...? Seems like something is borked with your node.js installation. I think you should try reinstalling node.js, and if that doesn't fix it, reinstall Raspbian.

bryan-m-hughes, 2015-02-23:
@MrYsLab That install script looks fine to me. The Raspberry Pi 2 is amazingly non-different. In my testing it behaves _exactly_ like the Raspberry Pi 2 Model B+, just faster.

mfreard, 2015-02-23:
I tested twice
with two different SD cards. But the result is the same 
pi@raspberrypi ~ $ cd xipi
pi@raspberrypi ~/xipi $ ls
node_modules  xipi.sh  xiserver.js
pi@raspberrypi ~/xipi $ bash xipi.sh
Starting XiPi server...

module.js:340
    throw err;
          ^
Error: Cannot find module 'es6-shim'
    at Function.Module._resolveFilename (module.js:338:15)
    at Function.Module._load (module.js:280:25)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/pi/xipi/node_modules/johnny-five/lib/johnny-five.js:1:63)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)

bryan-m-hughes, 2015-02-23:
@mfreard have you had a chance to test with an arduino connected to your pi?

mfreard, 2015-02-23:
I don't understand why do you want i test with an arduino?
I just want control the pins of my RPi2 with scratch 2.0 like my Rpi B+

bryan-m-hughes, 2015-02-23:
I'm trying to narrow down the source of the problem, and this should help with that. To be honest I don't know why you're consistently getting this issue, or how to fix it, and this is my last idea.

MrYsLab, 2015-02-23:
@bryan-m-hughes Can an sd card flashed and configured for a B+ be safely placed in a PI2 without fear of corruption when placed back in the B+? The reason I ask, is that Michel had a working B+ image. Perhaps that could be tried to see what the results are.

bryan-m-hughes, 2015-02-23:
@MrYsLab that's a good idea, and yes it should work just fine AFAIK. @mfreard can you try that?

MrYsLab, 2015-02-23:
@mfreard In case you had deleted my earlier email, here are the instructions that worked to create the B+ image:
1. I flashed Raspbian 2015-02-16 on to an SD card.
2.After booting up, I 
	expanded rootfs
	enabled boot to desktop

3. I executed:
		sudo apt-get update
		sudo apt-get upgrade
	
	Nothing was needed to be upgraded because this is a new image.

4. I installed xrdp to allow me access the graphical interface so I could download xipi using the Rpi browser. Xrdp will start up each time you boot automatically.

	sudo apt-get install -y xrdp

5. Install node:
	
	sudo wget http://node-arm.herokuapp.com/node_0.10.36_armhf.deb

6. Download xipi from github.

7. Run xipiInstall.sh

mfreard, 2015-02-24:
I still have the mail and I follow the instructions to install my SD card.

if I understand correctly, you want that i  put the sd card of the Rpi B + in the Rpi2 , that's right?

bryan-m-hughes, 2015-02-24:
Yes, that's right

mfreard, 2015-02-24:
Hi, great news : it works... 
Sorry, I follow yet the same directions as for installation on my Rpi B +
I have a question : I would like to share the contents of my card to a friend on Github .
But when I save the image created is the size of my 8GB card while the OS doesn't take the whole card . Do you know how to back up only the part that contains the files ?
I don't know if I'm clear? 

Thanks a lot for your help.

bryan-m-hughes, 2015-02-24:
I don't really know to be honest, perhaps [GParted](http://gparted.sourceforge.net/) could help?

MrYsLab, 2015-02-24:
Here are a couple of  links that might help: http://www.raspberrypi.org/forums/viewtopic.php?f=26&t=22524
http://www.raspberrypi.org/forums/viewtopic.php?f=41&t=49305

@bryan-m-hughes Thanks for all the help.