nebrius/raspi-io #16: I2C not implemented

nwalters512, 2015-02-24:
Are there any plans to implement I2C communication in the near future?

bryan-m-hughes, 2015-02-24:
Yes there are! Before I can do that, I need to get a good unit testing framework in place, which I am working on right now. I don't have an exact timeline, but I think 1-2 months is a reasonable timeframe for when I2C support will be implemented

nwalters512, 2015-02-24:
Ok, thank you! I looks forward to being able to use I2C :)

cybertim, 2015-03-03:
:+1: 

@rwaldron tipped me about this framwork, but when I tried my PCA9685 to run some servo's all i got was 'sendI2CConfig is not yet implemented' :)

Looking forward to the i2c impl. for the rasp pi.

fivdi, 2015-03-07:
@bryan-m-hughes This is a continuation of our conversation on gitter yesterday about using i2c-bus for I2C. 

Good news, things are looking good.
The current implementation can be seen here [https://github.com/fivdi/raspi-io](https://github.com/fivdi/raspi-io)

The implementation is being tested with the following code at the moment:

```js
var five = require("johnny-five");
var RasPi = require("raspi-io");
var board = new five.Board({
  io: new RasPi()
});

board.on("ready", function() {
  var accelerometer = new five.Accelerometer({
    controller: "ADXL345"
  });
  var compass = new five.Compass({
    controller: "HMC5883L"
  });
  var x, y, z;
  var heading, bearing;
  var accelerometerReadings = 0;
  var compassReadings = 0;

  setInterval(function () {
    console.log(compassReadings + ' '
      + heading + ' '
      + bearing + ' '
      + accelerometerReadings + ' '
      + x + ' '
      + y + ' '
      + x + ' '
    );
  }, 500);

  accelerometer.on("data", function() {
    x = this.x;
    y = this.y;
    z = this.z;

    accelerometerReadings += 1;
  });

  compass.on("data", function() {
    heading = Math.floor(this.heading);
    bearing = this.bearing.name;

    compassReadings += 1;
  });
});
```

The test code requests data from an ADXL345 and a HMC5883L as often as possible and prints summary information twice a second. The bus is running at 400KHz. I'll leave it running for a few hours to see what happens.

There are a few things that should be mentioned:
* During the initialization process raspi-io currently assumes that all pins are inputs and creates DigitalInput objects for each pin [here](https://github.com/fivdi/raspi-io/blob/2.2.2/index.js#L143). This "breaks" the pins that are required for I2C. They no longer function correctly afterwards. For example, Linux tools like `i2cdetect -y -r 1` no longer function. The solution (workaround?) chosen was not to create DigitalInput objects for the I2C pins by returning from the forEach [here](https://github.com/fivdi/raspi-io/blob/master/index.js#L152-L154). I hope this doesn't break something else. I can't think of a better solution for this at the moment. Perhaps the Raspi constructor should have an option to say whether or not I2C is being used.
* I didn't modify anything in [raspi-i2c](https://github.com/bryan-m-hughes/raspi-i2c/blob/master/index.js#L29-L43). Currently it looks like class I2C in raspi-i2c should support multiple devices, but it does things which would indicate that it will only function correctly with one device. For example, the call `super([ 'SDA', 'SCL' ])` [here](https://github.com/bryan-m-hughes/raspi-i2c/blob/master/index.js#L31) registers pins, but may also destroy existing pins that are needed for an existing device wired to the bus. At least that's what I think will happen. The constructor also has an optional baudRate. Normally the baudrate applies to the I2C bus, not a single device on the bus. I guess the design might need a bit of work here and it would be better if you took over from here and decided what needs to be done.
* The npm module execsync which is used by raspi-i2c doesn't compile with io.js v1.3.0. It may no longer be needed anyway.

So, now for the important question :), assuming your ok with what's there at the moment, can I make a PR with what's there now and leave the decisions and implementation in raspi-i2c up to you? 

[Configuring I2C on the Raspberry Pi](https://github.com/fivdi/i2c-bus/blob/master/doc/raspberry-pi-i2c.md) explains how to configure I2C on recent versions of Raspbian.

bryan-m-hughes, 2015-03-07:
@fivdi So...

The reason that raspi-io assumes all pins are DigitalInputs is because they are all DigitalInputs by default according to the J5 io plugin design, and this logic needs to be preserved. This is actually totally valid legit code:

```
...
board.on('ready', function() {
  var button = new five.Button('SDA0');
});
```

Wiring Pi has the capability to change the mode back to I2C after changing it to GPIO. Does i2c-bus have this capability? If not, we may have to stick with Wiring Pi after all, which makes me sad.

This is also part of the reason why we need to wrap the i2c code in raspi-i2c (in addition to all the other reasons I mentioned yesterday). Wrapping in raspi-i2c enables the pin contention mechansim for swapping these modes. That said, J5 doesn't call ```board.pinMode``` for I2C calls, so we'll need to have something like:

```
import { I2C } from 'raspi-i2c';
...
i2cWrite(address, cmdRegOrData, inBytes) {
  var pinInstance = this[getPinInstance](pin);
  if(!(pinInstance.peripheral instanceOf I2C) {
     // Do a pinMode-like switch
  }
```
```

You're notes on raspi-i2c are pretty spot on...like I said, that code is _very_ rough and incomplete, so don't believe anything in it :) It should look a lot more like the code in raspi-pwm.

I'm going to be off-grid for most of the day (I'm going hiking). I'll be working on this a lot tomorrow, so it's up to you how you want to handle this. If you want to work on it today, I recommend tracing through all the details on how raspi-pwm works, since the i2c code will be very similar. Start with raspi-peripheral (and raspi-board) and work you're way up to raspi-io.

Thanks for working on this, I really appreciate it. We'll get there!

bryan-m-hughes, 2015-03-07:
Oh also, forgot to mention, after some though, I think the raspi-i2c constructor should look like:

```
constructor(baudRate, pins) {
  super(pins || [ 'SDA0', 'SCL0' ]);
}
```
If pins are supplied, it would probably be a good idea to make sure there are two pins listed, and they are SDAx and SCLx. The 'SDA' and 'SCL' names were removed a while ago.

fivdi, 2015-03-07:
I can't find any information in the [IO-Plugins documentation](https://github.com/rwaldron/johnny-five/wiki/IO-Plugins) that states that all pins must be explicitly configured as digital inputs per software.

Yes, I agree, the following code is totally valid legit code:
```
board.on('ready', function() {
  var button = new five.Button('SDA0');
});
```
However, it is not necessary to explicitly configure all pins as digital inputs to achieve this.

WiringPi doesn't have a way to change the mode back to I2C after it has been a digital input. Explicitly configuring the pins to be digital inputs also breaks WiringPi. This can be seen in the output below. The following steps are performed on a freshly booted Pi:

* Display a J5 program that strobes an LED on P1-11 (P1-11has nothing to do with I2C)
* Load the I2C kernel modules using the WiringPi gpio utility
* Display the devices on the I2C bus using the WiringPi gpio utility (it works correctly this time)
* Run led.js to strobe the led. The led strobes :)
* Load the I2C kernel modules using the WiringPi gpio utility
* Display the devices on the I2C bus using the WiringPi gpio utility (it doesn't works correctly this time)

WiringPi doesn't function correctly after led.js has been executed because explicitly configuring the pins as digital inputs rather that leaving them in the state that the user requested them to be in in `/boot/config.txt` breaks them for everyone.
```
pi@raspberrypi ~/dev/j5 $ cat led.js
var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi()
});

board.on('ready', function() {
  (new five.Led('P1-11')).strobe();
});
pi@raspberrypi ~/dev/j5 $ gpio load i2c
gpio: Warning (not an error): File not present: /dev/i2c-0
pi@raspberrypi ~/dev/j5 $ gpio i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- 1e -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- 39 -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- -- 
50: -- -- -- 53 -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@raspberrypi ~/dev/j5 $ sudo $(which node) led.js
1425753536351 Device(s) RaspberryPi-IO 
1425753536425 Connected RaspberryPi-IO 
1425753536456 Repl Initialized 
>> 
(^C again to quit)
>> 
1425753539830 Board Closing. 
pi@raspberrypi ~/dev/j5 $ gpio load i2c
gpio: Warning (not an error): File not present: /dev/i2c-0
pi@raspberrypi ~/dev/j5 $ gpio i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@raspberrypi ~/dev/j5 $ 
```
The modularity that you want in Raspi-IO is perfectly ok in my opinion. I've no objections to any of it :). And yes, we'll get there. However, if we try to achieve too much in one step, we'll be a lot longer getting there.

rwaldron, 2015-03-07:
StandardFirmata initializes all digital pins as digital inputs, but that's specific to that platform. There's no hard rule about initial state of the actual gpio, as long as the code in the program can reason about the relevant state as told by the io-plugin.
For example: galileo-io no longer initializes anything up front. It keeps a list of pin states in memory and then initializes the actually io control instances as needed. Hope that's helpful. We can update the io plugin protocol to specify all this as well

bryan-m-hughes, 2015-03-08:
Ah, things make sense now, thanks @rwaldron. When I started writing raspi-io almost a year ago (!!), the IO plugin spec was largely incomplete, necessitating lots of testing to figure things out. I do remember getting the "init to input" tidbit of information from firmata, not the spec now :)

@fivdi That's too bad about Wiring Pi. I could have sworn it worked, but then again I haven't looked at their I2C stuff in months. Given that initializing to input isn't actually in the J5 IO spec, I'm OK with changing this now.

So here's what I propose we do:

I'll work on removing the DigitalInput initialization for _all_ pins, so that it behaves like galileo-io. I prefer this approach to initializing some to GPIO and some to I2C. This will require regression testing, bug fixing, etc and generally prepping for the now 3.0.0 release of raspi-io.

@fivdi, you work on implementing this work inside of raspi-i2c. We still need this code implemented in raspi-i2c not raspi-io. This will enable the pin contention mechanism for i2c, which we can use to check for the case where someone changes the pinMode to GPIO and then tries to write to I2C and display a proper warning message. I also really don't want there to be peripheral support in J5 that isn't also in Raspi.js. To test, you can do something like this:

```
var raspi = require('raspi');
var I2C = require('raspi-i2c').I2C;

raspi.init(function() {
  var i2c = new I2C();
  i2c.doStuff();
});
```

Once that's done, we'll regroup on integrating raspi-i2c into raspi-io. Does that sound good to you?

fivdi, 2015-03-08:
Yes, that sounds good to me.

Here are a few points where I need some feedback:

#### i2cWrite
In a post above it says that i2cWrite should be implemented something like this:
```
import { I2C } from 'raspi-i2c';
...
i2cWrite(address, cmdRegOrData, inBytes) {
  var pinInstance = this[getPinInstance](pin);
  if(!(pinInstance.peripheral instanceOf I2C) {
     // Do a pinMode-like switch
  }
```
There are a number of issues here:
* None of the J5 I2C methods are passed a pin argument so the pinInstance will not be found.
* There are two pin involved, not one.
* It's not necessary to do a pinMode like switch.

I'd suggest that we take a standard approach for configuring I2C. Typically this happens at boot time. So, if someone wants to use I2C in Raspi-IO, they need to make sure it's configured at boot time. There's a description of how to do this [here](https://github.com/fivdi/i2c-bus/blob/master/doc/raspberry-pi-i2c.md).

Is this ok for you?

#### Baudrate
In a post above it says that the raspi-i2c constructor should look something like this:
```
constructor(baudRate, pins) {
  super(pins || [ 'SDA0', 'SCL0' ]);
}
```
Baudrate is another aspect that is typically configured at boot time. For example, the baudrate on Raspbian can be set to 400KHz by adding `dtparam=i2c1_baudrate=400000` to `/boot/confix.txt`. I suggest we take this approach and remove the baudRate from the constructor.

Is this ok for you?

#### Raspi-i2c
In order to implement the J5 I2C methods, the following four primitives are required:
* i2cRead(addr, length, buffer, cb) - Asynchronous plain I2C read
* i2cWriteSync(addr, length, buffer) - Synchronous plain I2C write
* readI2cBlock(addr, cmd, length, buffer, cb) - Asynchronous I2C block read
* writeByteSync(addr, cmd, byte) - Synchronous write byte.

These are the initial four I2C methods required by class I2C, correct?

fivdi, 2015-03-08:
The test code posted above that requests data from an ADXL345 and a HMC5883L as often as possible has being running for over 24 hours now. Everything looks ok :)

bryan-m-hughes, 2015-03-08:
### i2cWrite

Don't worry about all this, we'll cross that bridge when we get there (I've got a few ideas already). The design for raspi-i2c shouldn't be based on what's needed of J5, unless the design for raspi-i2c makes it impossible to conform with the J5 interface, which is not the case here.

Let's focus on getting a good implementation of raspi-i2c done first, which does require passing in an a single pin _or_ array of pins (so the multiple pins thing has already been solved). I already paid the price for just assuming one port for PWM, and I don't want to make that mistake again.

### Baud Rate

One of my key goals for raspi-io is that users don't have to configure _anything_ outside of JavaScript, it conflicts with my and @monteslu vision for what we want to do with this platform long term (our own variant of Raspbian that is plug-and-play with j5-chrome).

Changing the baud rate is really easy with the Wiring Pi ```gpio``` command line tool, and that code has already been written anyways, so why throw it out? We should stick with this approach.

I know this is not the "Linux way", but I prioritize that much lower than ease of use. Given J5 isn't necessarily a Linux thing, and a most of the people using the Raspberry Pi are not well-versed in Linux sysadmin, I just don't value doing things the "Linux way" with raspi-io or Raspi.js.

### raspi-i2c

No, that's not the purpose of raspi-i2c. As I mentioned above, the design of raspi-i2c is _independent_ of the needs of J5. The design of raspi-i2c should hew close to the design of i2c-bus, except where it conflicts with the general design of raspi-gpio and raspi-pwm, at which point you should base it on the design of raspi-gpio and raspi-pwm.

It is perfectly OK for Raspi.js to have more functionality than J5. This was always the goal in fact. J5 presents a generic interface across _all_ hardware platforms, but this approach will, by definition, mean not including support for everything in the base platform.

A good analogy here is that Raspi.js is like OS X's Cocoa, and J5 is like Java's AWT. AWT consumes Cocoa (I assume), but the design of Cocoa is independent of AWT, and Cocoa has support for a lot more things than AWT has.

fivdi, 2015-03-08:
The following four methods are not related to J5, they are implemented by i2c-bus:

* i2cRead(addr, length, buffer, cb) - Asynchronous plain I2C read
* i2cWriteSync(addr, length, buffer) - Synchronous plain I2C write
* readI2cBlock(addr, cmd, length, buffer, cb) - Asynchronous I2C block read
* writeByteSync(addr, cmd, byte) - Synchronous write byte.

They are the four i2c-bus methods that the J5 specific I2C methods call.

If raspi-i2c does not implement these or almost identical methods, then I'm afraid I have no idea what raspi-i2c is.

To help me understand please describe three or four concrete examples of methods that raspi-i2c should implement. Please be a concrete as possible.

bryan-m-hughes, 2015-03-08:
@fivdi, I see, I misunderstood you, I thought you meant those were the J5 methods we wanted to implement, and raspi-i2c doesn't know or care about J5.

Here's my proposal for the raspi-i2c interface:
```
export class I2C extends Peripheral {
  constructor(baudRate, pins) {}
  
  close(cb) {} // function cb(err), returns undefined
  closeSync() {} // returns undefined

  i2cRead(address, register, length, cb) {} // function cb(err, buffer), returns undefined, register is optional
  i2cReadSync(address, register, length) {} // Returns a buffer, register is optional
  i2cWrite(address, register, buffer, cb) {} // function cb(err), returns undefined, register is optional
  i2cWriteSync(address, register, buffer) {} // returns undefined, register is optional

  readByte(address, register, cb) {} // function cb(err, value), returns undefined, register is optional
  readByteSync(address, register) {} // returns the value, register is optional
  readWord(address, register, cb) {} // function cb(err, value), returns undefined, register is optional
  readWordSync(address, register) {} // returns the value, register is optional

  writeByte(address, register, byte, cb) {} // function cb(err), returns undefined, register is optional
  writeByteSync(address, register, byte) {} // returns undefined, register is optional
  writeWord(address, register, word, cb) {} // function cb(err), returns undefined, register is optional
  writeWordSync(address, register, word) {} // returns undefined, register is optional
}
```

fivdi, 2015-03-08:
Ok, that looks fine. Some additions are required. Plain I2C methods where no register is specified are needed by J5. To be more specific, equivalents of the i2c-bus methods i2cRead and i2cWriteSync are needed.

bryan-m-hughes, 2015-03-08:
That's why every method says ```register is optional``` ;-)

fivdi, 2015-03-09:
Right, I missed that.

One instance of the I2C class is required for each device wired to the bus. Each of these will have it's own i2c-bus instance. Each i2c-bus instance will in turn have a file descriptor for that device.

This makes the baudRate argument questionable where it is at the moment. The baudRate should apply to the bus as a whole, not to each device.

bryan-m-hughes, 2015-03-09:
We discussed that already and I thought we came to a conclusion, but maybe not?

The API I designed above is for a single instance per _bus_, not per attached device. That's why each method takes an ```address``` argument.

bryan-m-hughes, 2015-03-09:
I just published v3.0.0 of raspi-io, which changes the default pin mode from INPUT to UNKNOWN and removes the DigitalInput instantiation.

Turns out, my code was a lot more resilient to this change than I expected, the changes were small :D https://github.com/bryan-m-hughes/raspi-io/commit/60feb4645c5a48461e5813dda6154bf106ffb86b

fivdi, 2015-03-09:
That's good news :)

bryan-m-hughes, 2015-03-09:
@fivdi FYI I also pushed some changes to the raspi-i2c repo. I upgraded the dev dependencies to the latest version, switched from traceur to babel, and stubbed out the methods according to the spec above. This way I can start integrating it into raspi-io while you work on implementing the details.

fivdi, 2015-03-10:
@bryan-m-hughes ok. It will be Saturday before I can start working on this again.

bryan-m-hughes, 2015-03-10:
@fivdi, no worries, I'm swamped this week and going on a snoeshowing trip this weekend, so it will be Tuesday before I can I work on it again :)

fivdi, 2015-03-13:
Do you have any objections if I start the integration of raspi-i2c into raspi-io? In theory, if raspi-i2c implements it's interface correctly, everything will function as expected, but it would be more effective for me if I can see that it works in combination with J5. To see this, the integration of raspi-i2c into raspi-io is needed.

fivdi, 2015-03-14:
Modifications of raspi-i2c and raspi-io have just been pushed. They can be seen here:

* [https://github.com/fivdi/raspi-i2c](https://github.com/fivdi/raspi-i2c)
* [https://github.com/fivdi/raspi-io](https://github.com/fivdi/raspi-io)

The following points are worth mentioning:

* raspi-i2c no longer depends on execsync as execsync doesn't support io.js or Node.js v0.12. For io.js and Node.js v0.12 child_process.execSync is now used. child_process.execSync was introduced with Node.js v0.12. Is an alternative solution needed for older versions of Node.js?
* Tested with io.js v1.3.0 on a Raspberry Pi 2
* For testing rasp-i2c the tests [here](https://github.com/fivdi/raspi-i2c/tree/master/test) were implemented
* For testing J5 [this](https://github.com/fivdi/raspi-io/blob/master/test/two-devices.js) was used
* The dependency of raspi-io on raspi-i2c needs to be added package.json
* raspi-i2c hasn't been published on npm yet
* I'm not convinced that `gpio load i2c baudRate` actually changes the baudrate on my system
* The JavaScript files generated by gulp were not pushed
* The ideas you have regarding i2cWrite (mentioned above) need to be taken into account

How should we proceed from here?

bryan-m-hughes, 2015-03-16:
For the most part this sounds good, I just two comments.

We need to continue supporting Node 0.10 for now, so we'll need to install/require execSync conditionally

Regarding `gpio load i2c baudRate`...I really want to confirm _conclusively_ whether or not it works, or if it only works when using Wiring Pi, etc. I've got an o-scope I can throw on it if you don't have a way to test it. If it turns out that we can't support changing the baud rate programmatically, then we will just kinda pretend that baud rate doesn't exist for now, and enterprising hackers can figure out ways around it.

Speaking of, I realized that there is a bit of an inconsistency with raspi-gpio, and fixing it actually works to our favor. If you take a look at raspi-gpio's constructor, you can see that it always takes exactly one argument, and that argument is either a primitive (representing a single pin) or an object (representing a pin configuration). For raspi-i2c, we can say that it takes a single (optional) argument as well: a pin array will be all that's supported for now, and eventually we can also say it takes a configuration object with a baud rate, assuming we figure out how to implement that. This way, it's a backwards compatible design that won't require a major version rev down the line.

My thoughts on proceeding from here is that you should work on getting these two issues ironed out, and then, if you're happy with the code, submit a PR. We can work out the remaining details there.

fivdi, 2015-03-16:
Yes, it's a bit too early for dropping support for v0.10 :) Installing execsync conditionally should be fairly easy. I'll add a conditional install and take another look at `gpio load i2c baudRate` to see if I can figure out what's going on. If I can't figure it out you'll probably have to take a look with your scope as I don't have one. It'll be tomorrow evening before I get around to doing it.

fivdi, 2015-03-16:
@bryan-m-hughes do you have a a few I2C devices supported by J5 that can be wired up to the Pi for testing?

bryan-m-hughes, 2015-03-16:
@fivdi I might have an extra I2C device lying around, not sure. I was planning on testing against another microcontroller in slave mode and/or using the logic analyzer built into my o-scope that understands I2C

fivdi, 2015-03-17:
@bryan-m-hughes 

So, the PRs have been submitted :)

The PR for raspi-i2c conditionally installs execsync. If Node.js v0.10 is being used, execsync is installed. This was tested with Node.js v0.10.37 on a Pi 2 running the latest version of Raspbian (2015-02-16)

I'm still not sure if `gpio load i2c baudRate` changes the baud rate.

It's still necessary to enable the ARM I2C interface using the raspi-config utility on the latest version of Raspbian. `gpio load i2c` doesn't appear to do this.

bryan-m-hughes, 2015-03-17:
@fivdi sorry, I haven't had a chance to get my oscope on my pi yet to test. I'll try and get that done some time this week. For now though, I say we just skip baud rate control. We can always add it as an extra in the config object passed to the constructor.

I'll take a peak at your PR tomorrow morning (St Paddy's day stuff tonight 😁)

bryan-m-hughes, 2015-03-20:
@fivdi reviews are in. Just a few more code style nits and then I think the code will look good. Once those are fixed, I'll do some quick sanity checks on my end with my test array (all the Pi versions and all the node versions), and then we'll be ready to land the patches!

bryan-m-hughes, 2015-03-23:
Just a status update, I landed @fivdi's raspi-i2c patch, and I think I'm done with [ini-builder](https://github.com/bryan-m-hughes/ini-builder), which will be used to parse /boot/config.txt. Once that's done and integrated into raspi-i2c, we'll be ready to publish that to NPM. At that point I'll get back to reviewing the raspi-io PR.

fivdi, 2015-03-23:
@bryan-m-hughes you've already reviewed the raspi-io PR three times. Are you not finished yet? :) You're hard to please. Hopefully we can get this completed before the first of us retires from the software business :)

bryan-m-hughes, 2015-03-23:
@fivdi the code looks fine in raspi-io, I just want to do an end-to-end test, including installing raspi-i2c from NPM with all its installation shenanigans, before I publish everything :)

I'm being picky to avoid getting stuck further down the line. Once an API is defined and released, it's _really_ hard to change it without facing backlash (and raspi-i2c is an API, as is raspi-io). Given that raspi-io is starting to get kinda popular, I think it's a lot more important to get it done right than to get it done fast.

Also, we don't have proper unit tests or integration tests for raspi-io, and without those I'm wary about making changes of this scope without lots of testing and consideration.

fivdi, 2015-03-23:
@bryan-m-hughes I hope you didn't take my last comment seriously as it wasn't meant so. Everything is going fine, we're nearly there. Yes, the lack of tests is a bit of an issue. I've only tested the J5 side of things with an ADXL345 and a HMC5883L.

bryan-m-hughes, 2015-03-26:
I2C support is now in with v3.1.0 of raspi-io

closed 2015-03-26 by bryan-m-hughes