nebrius/raspi-io #21: Pin aliases not working

danielmahon, 2015-03-06:
Using the Raspi2 in the following example only the WiringPi pin numbers work. With `GPIO-18` and `P1-12` the `motionstart` event is not fired and no errors occur. I have tried with other pins as well.

```
  // Create a new `motion` hardware instance.
  var motion = new five.IR.Motion(1); // "GPIO-18" and "P1-12" doesn't work

  // "calibrated" occurs once, at the beginning of a session,
  motion.on("calibrated", function() {
    console.log("calibrated");
  });

  // "motionstart" events are fired when the "calibrated"
  // proximal area is disrupted, generally by some form of movement
  motion.on("motionstart", function() {
    console.log("motionstart");
    intruder = true;

    sendMessage('Motion Detected!');

  });
```

bryan-m-hughes, 2015-03-06:
Which version of raspi-io are you using?

danielmahon, 2015-03-07:
@bryan-m-hughes v2.2.2

bryan-m-hughes, 2015-03-08:
I just published v2.2.3 which fixes this issue. Thanks for the bug report!

closed 2015-03-08 by bryan-m-hughes