nebrius/raspi-io #24: Add support for other OSes

bryan-m-hughes, 2015-03-27:
I think that Raspbian is the only OS raspi-io works on, although I'm not certain. We should confirm and make sure it does work on others, especially pidora.

frxnz, 2015-03-27:
Installing [Ubuntu](https://wiki.ubuntu.com/ARM/RaspberryPi) on my Pi 2 now. Will report back soon!

wadtech, 2015-06-11:
Ubuntu Core aside, we were going to use [Ubuntu Mate](https://ubuntu-mate.org/raspberry-pi/) to get easier access to a newer version of chromium.

Did you have any success with this @frxnz ?

frxnz, 2015-06-11:
@wadtech This was a few months ago, but I had pretty bad luck with Ubuntu. Hopefully things have gotten better since then.

wadtech, 2015-06-12:
Cheers, I'll let you know how I get on then

cardimajs, 2015-10-11:
please make rasp-io work on Arch Linux

wadtech, 2015-10-13:
Just for an update: it went really badly. Although it seemed that copying the vendor folder over from raspbian actually worked fine, it's just the install itself that wasn't right.

ymmv obviously.

nebrius, 2015-10-13:
@wadtech thanks for checking that out. If I were to guess, I would say it's because of https://github.com/nebrius/raspi-i2c/blob/master/script/enable_i2c.js

We'll probably need a different version of that script for each OS 😖

andrewn, 2015-11-15:
I tried to install an app using rapsi-io on a Raspberry Pi using the [Resin](https://resin.io) platform.

I think this script caused the install to hang and then fail because it waits for user input.

It would be great to have a way to optionally disable these scripts from running, perhaps by setting an environment variable on install? The user would then be responsible for ensuring I2C is enabled on their platform.

nebrius, 2015-11-15:
Good idea @andrewn, I like it.

Is this something you need asap? I'm really swamped, and don't really have time to work on raspi-io at the moment, sadly.

Would you be willing to implement it? It should be pretty straightfoward. Just add a `process.exit` call towards the top of the enable_i2c script linked above

ekryski, 2015-12-03:
I had the same issue there are potentially 2 problems with the i2c post install work:

1) If you are not installing using sudo or are not root then you will be prompted for your password

2) It assumes that `/boot/config.txt` is present. When running inside docker it won't be. I hacked around this by `touch /boot/config.txt` inside the container.

I'm going to check out resin.io right now but I might be able to get a PR in tomorrow, doing pretty much what you suggested @nebrius.

nebrius, 2016-01-07:
Ubuntu support is in!

ekryski, 2016-01-07:
Woooot! :tada:

philraj, 2016-03-17:
Got it running beautifully in Arch ARM, for future reference.

Swizz, 2016-04-01:
Hi @philraj,
It will work out of the box ?

philraj, 2016-04-01:
@Swizz I guess it depends on your definition of 'out of the box'. :)

I had some weird issues with the build process (npm was installing modules into the wrong folders), but running `rm -rf node_modules` and reinstalling raspi-io fixed it.

Swizz, 2016-04-04:
@philraj I tried it.

Something weird happened. All gyp packages have needed to be build separately.
It is truly blocking in a push-to-deploy philosophy.

nebrius, 2017-09-25:
This issue has been open for a long time now, and I haven't made any progress on adding other OS's. Sorry, but it's not a priority for me.

That said, raspi-io has been reworked so that it has a much more modular architecture now. If someone needs a new OS supported that isn't supported out of the box, you can roll your own a lot more easily now. Take a look at https://github.com/nebrius/raspi-io/blob/master/src/index.js to see how to inject the OS-specific dependencies into raspi-io-core.

I'm closing this issue now, thank you everyone who participated and left feedback!

closed 2017-09-25 by nebrius

WebReflection, 2018-06-20:
In case anyone uses ArchLinux, I've written up a little how to:
https://medium.com/@WebReflection/how-to-raspi-io-on-archlinux-5d59229e5763

jtpower12, 2018-08-26:
use it make Last few GCs