nebrius/raspi-io #27: Troubles when using hypriot images and docker

Snorlock, 2015-05-15:
Im trying to use Johnny Five as a base for my project. We are going to have a lot of raspberry Pi's so its also based on using docker with raspbian pi, by using the images from http://blog.hypriot.com/.

Unfortunalty raspi-io doesnt work even if i use an image with wiringPi and everything inside.
Its worth telling that using raspi and raspi-gpio inside the same image and top of the same stack, works perfectly. I have gotten my led to light up with these libraries.

But when i use JohnnyFive and raspi-io i recieve this error:
```
root@3c17965c9979:/home/johnny# node test.js 
libkmod: ERROR ../libkmod/libkmod.c:554 kmod_search_moddep: could not open moddep file '/lib/modules/3.18.11-hypriotos+/modules.dep.bin'
child_process.js:1382
    throw err;
          ^
Error: Command failed: modprobe i2c-dev
libkmod: ERROR ../libkmod/libkmod.c:554 kmod_search_moddep: could not open moddep file '/lib/modules/3.18.11-hypriotos+/modules.dep.bin'

    at checkExecSyncError (child_process.js:1339:13)
    at execSync (child_process.js:1379:13)
    at new I2C (/home/johnny/node_modules/raspi-io/node_modules/raspi-i2c/lib/index.js:116:5)
    at /home/johnny/node_modules/raspi-io/lib/index.js:147:16
    at new Raspi (/home/johnny/node_modules/raspi-io/lib/index.js:182:7)
    at Object.<anonymous> (/home/johnny/test.js:4:7)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
```

My Johnny-Five code is: 
```
var five = require("johnny-five");
var Raspi = require("raspi-io");
var board = new five.Board({
  io: new Raspi()
});

board.on("ready", function() {
  var led = new five.Led("P1-11");
  led.blink();
});
```

And my working raspi code is: 
```
var raspi = require('raspi');
var gpio = require('raspi-gpio');

raspi.init(function() {
  var output = new gpio.DigitalOutput('P1-11');

  output.write(0);
});
```

If there is anything more I can do to aid let me know. The reason we are using Johnny-Five is because we are going to have more different modules and want to try to keep the codebase very similar.

bryan-m-hughes, 2015-05-15:
It looks like this is an issue with I2C, which is initialized as soon as raspi-io is initialized. Can you try running ```sudo modprobe i2c-dev``` for me? This is what raspi-i2c is calling when it crashed, and it should give the same message when run directly.

Snorlock, 2015-05-15:
root@3c17965c9979:/data# sudo modprobe i2c-dev
libkmod: ERROR ../libkmod/libkmod.c:554 kmod_search_moddep: could not open moddep file '/lib/modules/3.18.11-hypriotos+/modules.dep.bin'

fivdi, 2015-05-15:
Hmm, `sudo modprobe i2c-dev` should work. I think there's something wrong with the image being used. Googling for `libkmod: ERROR ../libkmod/libkmod.c:554 kmod_search_moddep: could not open moddep file` often finds people with broken images.

Snorlock, 2015-05-15:
Its weird though because the image is very functional otherwise. And it works calling this on the root image that is on the SDcard, but not inside the docker container.

bryan-m-hughes, 2015-05-18:
I think @fivdi is right. Can you try redownloading the hypriot image and reflashing your SD card? If that doesn't fix it, I would bring it up with the Hypriot folks that `sudo modprobe i2c-dev` gives you that error (that raspi-io is calling it isn't actually relevant). I think https://github.com/hypriot/talk is the best place.

closed 2015-05-18 by bryan-m-hughes

Snorlock, 2015-05-18:
But again, its not the image on the SD card that is broken, when running this command on the image it works. Its the container im trying to run inside. But il ask the hypriot guys and see what might result in that.

Snorlock, 2015-05-18:
What solved this problem was this blogpost: http://dummdida.tumblr.com/post/117157045170/modprobe-in-a-docker-container

I had to give access to that part of the kernel.

They way I ran the docker container for full access was: docker run --device /dev/ttyAMA0:/dev/ttyAMA0 --device /dev/mem:/dev/mem --device /lib/modules:/lib/modules --cap-add=ALL --privileged -ti snorlock/node-gpio /bin/bash

bryan-m-hughes, 2015-05-18:
Glad you got it working!

fivdi, 2015-05-18:
And thanks for describing the solution :+1: