nebrius/raspi-io #28: Raspi-IO broken on Node 0.10

bryan-m-hughes, 2015-05-31:
Grrr, it turns out that Babel doesn't shim ```Symbol``` like traceur does, and of course I switched to it without testing. Node 0.12 and io.js both have native support for symbols, so I didn't even notice until now.

Raspi-IO crashes on startup in Node < 0.12 since it doesn't have support for Symbol. Need to add a shim

bryan-m-hughes, 2015-06-01:
Fixed with v3.2.2

closed 2015-06-01 by bryan-m-hughes

punitganshani, 2016-04-09:
Faced this issue as well

```
/home/pi/code/matrix/node_modules/raspi-io/node_modules/raspi-i2c/lib/index.js:96
var devices = Symbol('devices');
              ^
ReferenceError: Symbol is not defined
    at Object.<anonymous> (/home/pi/code/matrix/node_modules/raspi-io/node_modules/raspi-i2c/lib/index.js:96:15)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/pi/code/matrix/node_modules/raspi-io/lib/index.js:21:17)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
```