nebrius/raspi-io #29: Error: Pin "9" does not support mode "3"

riaancornelius, 2015-06-07:
I'm not sure if this is the right place for it, but as far as I can tell, I'm not doing anything stupid.

I'm running a Raspberry Pi B+ with a Pololu DRV8835 motor driver board. I've tested the motor driver using the sample Python code and all the hardware works fine.

I'm using the latest versions of everything (Johnny-five, Raspi-io, wiring-pi) as of about a week ago.

When I  run the following:
```javascript
var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi()
});

board.on('ready', function() {

  var motor1 = new five.Motor({
    pins: {
      pwm: 9,
      dir: 7
    }
  });

  var motor2 = new five.Motor({
    pins: {
      pwm: 10,
      dir: 8
    }
  });

  motor1.forward(64);
  motor2.forward(64);
});
```
I get this output:
pi@raspberrypinb ~/motortest $ sudo node basic.js
1433678531584 Device(s) RaspberryPi-IO
1433678533746 Connected RaspberryPi-IO
1433678533823 Repl Initialized
 /home/pi/motortest/node_modules/raspi-io/lib/index.js:328
          throw new Error("Pin \"" + pin + "\" does not support mode \"" + mod
                ^
Error: Pin "9" does not support mode "3"
    at Raspi.pinMode (/home/pi/motortest/node_modules/raspi-io/lib/index.js:328:17)
    at Motor.initialize (/home/pi/motortest/node_modules/johnny-five/lib/motor.js:484:11)
    at new Motor (/home/pi/motortest/node_modules/johnny-five/lib/motor.js:449:10)
    at Board.<anonymous> (/home/pi/motortest/basic.js:9:16)
    at Board.emit (events.js:104:17)
    at process._tickDomainCallback (node.js:381:11)

From what I've been able to see, this happens if you try to re-assign a pin, but as you can see, that's not the case here. 

The following works fine and blinks an LED on my setup:
```javascript
var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi()
});

board.on('ready', function() {
  (new five.Led('P1-7')).strobe();
});

Any help would be appreciated.

bryan-m-hughes, 2015-06-07:
The error message is correct, pins 9 and 10 aren't PWM pins, they are GND and GPIO-8, respectively :)

Pin numbering on the Raspberry Pi is, unfortunately, kinda complex since the different versions have different pinouts. See https://github.com/bryan-m-hughes/raspi-io/wiki for the full list of pins and pin names.

Try using pins 'P1-32' and 'P1-35', those are the two PWM pins on the RPi B+ that can be used to drive motors.

Let me know if you run into any more issues

closed 2015-06-07 by bryan-m-hughes

riaancornelius, 2015-06-07:
Guess I was wrong about doing something stupid... The reason I made this mistake was because I used the configuration in the Johnny-five example https://github.com/rwaldron/johnny-five/wiki/Motor#pololu-drv8835-shield

The weird thing is that it looks like the motor driver (https://www.pololu.com/product/2753) is actually wired to pins 29,31,32 and 33... And looking at the example Python code (https://github.com/pololu/drv8835-motor-driver-rpi/blob/master/pololu_drv8835_rpi.py), it seems to be expecting PWM on GPIO12 and GPIO13 and direction on GPIO5 and GPIO6. Either way, I'm confused.

Thanks for the quick response in any case. Now that I know what the problem is, I can try to make it work. I might just wire up a L293 and not bother with this motor driver...

bryan-m-hughes, 2015-06-07:
The Python code is probably emulating PWM on those pins, not doing actual, hardware PWM (you basically set timers and manually toggle the pin values). This is a trick that works somewhat well, but usually causes twitch in the motors, which hardware PWM doesn't do. This is why I decided not to enable software emulation of PWM in raspi-io.

rwaldron, 2015-06-08:
> The reason I made this mistake was because I used the configuration in the Johnny-five example https://github.com/rwaldron/johnny-five/wiki/Motor#pololu-drv8835-shield

The DRV8835 shield config is for the Arduino Uno (Leo, Yun, etc) pinout, but we should absolutely add one for the RPiB+ model https://www.pololu.com/product/2753.

younes200, 2015-11-28:
So, what is the pins to use we RPiB+ in this case ?

nebrius, 2015-11-30:
@younes200 the full pinout information can be found at https://github.com/nebrius/raspi-io/wiki/Pin-Information#p1-header-2

Those are the correct pin numbers. A few things to double check. Have you connected the PWM pins to the correct headers? The RPi headers aren't well labeled, and I plug into the wrong pins far more often than I'd like to admit 😅. Also, are you powering your motors from an external power source? The 5V supply on the RPi is notoriously fickle. Finally, do you have a proper common ground between all of your systems?