nebrius/raspi-io #30: Is Johnny-five mandatory?

roccomuso, 2015-07-02:
Does it require Johnny-five to work? can't it work without it?

monteslu, 2015-07-02:
it certainly can work without johnny-five as a stand alone IO class.

That said,  use Johnny-Five !

roccomuso, 2015-07-02:
Not exactly, just read that:

"Note: This module is not intended to be used directly. If you do not want to use Johnny-Five, I recommend taking a look at Raspi.js, which underpins this library and is a little more straightforward to use."

monteslu, 2015-07-02:
raspi.js makes it easier for sure, but if you want to use the raw IO class you can.  In fact, I'm using it for https://github.com/monteslu/node-red-contrib-gpio and https://github.com/monteslu/remote-io

Definitely use something like johnny-five or raspi.js.  I'm just saying it's possible to instantiate raspi-io directly and call its methods :)

roccomuso, 2015-07-02:
I'm trying to use the onoff library, but i have this issue:
https://github.com/fivdi/onoff/issues/35

nebrius, 2015-07-02:
I'll echo what @monteslu said, I wou recommend using raspi.js instead of raspi-io.

There is nothing that raspi-io supports that raspi.js does not support, but there are a few odds and ends that raspi.js supports that raspi-io does not. Indeed raspi-io is just a compatibility shim between raspi.js and Johnny-five.

Looking at the onoff and rp-gpio libraries, they use the same approach under the hood, which may explain why you're seeing the same behavior. Raspi.js uses a different approach that _may_ be more stable.

Let me know if you run into any issues with raspi.js, and I'd be more than happy to help!

closed 2015-07-13 by nebrius