nebrius/raspi-io #31: After power up,' board.io.pins[2].mode' returns 99, used to return 0.

MrYsLab, 2015-08-27:
The change has broken [Xi](https://github.com/MrYsLab/xi). Is it possible to restore the initial value to 0?
Also, after starting raspi-io, am I assured that the mode for all pins will be the same value (hopefully 0).
I need a consistent mode value in board.io.pins so that I can  determine if a Xi user has already assigned a pin mode to the pin. For the Arduino and BeagleBone Black, consistent values are returned.

Thanks.

rwaldron, 2015-08-27:
@nebrius `99` is an intentionally bogus mode. For compatibility, digital IO should be output and low by default.

nebrius, 2015-08-27:
We originally changed this behavior because once you change the mode of the i2c pins, you can't change it back which rendered the i2c pins unusable for i2c.

Perhaps another approach is to blacklist the i2c pins and simply not allow them to be used for GPIO?

MrYsLab, 2015-08-27:
The example I gave is for physical pin physical pin 13 - wiring pi pin 2 - GPIO27 (I just love having three different mapping designations for the same thing ;-). So where does i2c come into play for that pin. I can easily be wrong here, but aren't the i2c pins for the P1 connector  physical pins 3 and 5? 

Also, shouldn't it be the job of the application to manage the pins, that is, prevent a user from reassigning a pin that can't be reassigned?  An application author should have knowledge of the limitations of the library that she is using (as long as those limitation are well documented).

nebrius, 2015-08-28:
Yes, the i2c pins are p1-3 and p1-5, and they are also pins GPIO2 and GPIO3 (on the newer versions of the RPi at least).

If we initialized all GPIO pins to input, low on startup, as we previously did, then that means the mode on GPIO2 and GPIO3 would be switched to input, low and no one would be able to use i2c after that.

This is actually why the change was made, because i2c was broken without the change.

In order for i2c to work, we must initialize it to mode 99. The question then becomes "what about the other pins?" I think having some initialized to mode 99 and others to mode 0 is an undesirable solution. This is why I originally changed them all to mode 99.

I'm now thinking that if we simply say you can't use those pins as pins (not even expose them), then we're free to initialize them to whatever we want. The RPi has tons of GPIO as it is :)

MrYsLab, 2015-08-28:
Having pins 1-3 and 1-5 initialized to 99 and not exposing them, and having all other pins initialized to 0 would solve my problems.

nebrius, 2015-09-04:
Fixed with raspi-io v4.0.0. Testing these changes out also uncovered a _long_ standing bug with raspi-board, so that makes me happy :)

Let me know how it works!

MrYsLab, 2015-09-04:
Works great. Thanks!.

closed 2015-09-04 by MrYsLab

nebrius, 2015-09-04:
Hey @rwaldron, following up. The default mode is _output_, not _input_, right? @fivdi expressed some concern over this. What if a pin is going to be used as input, and there's a moment where the thing driving the pin is set high, and the microcontroller tries to set it low on startup? This could potentially damage the device.

rwaldron, 2015-09-04:
I guess that's a possibility, but that argument could easily go both ways. Maybe it's my inexperience, but I've never encountered an issue such as this in which setting a pin to output and low caused damage? The behavior has been baked into StandardFirmata since well before Johnny-Five ever used it: 

- During the reset: https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L620-L646
- Setting to output and low: https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L270-L275

I want to believe that the safest strategy was used here

nebrius, 2015-09-04:
Setting to output is definitely not the safest strategy. This is why virtually all microcontrollers default their pins to input at startup at the hardware level, it makes it impossible for two sources to try and drive the same piece of metal to different values.

That said, I think damage could only happen when the pin is being strongly driven high, e.g. connecting it straight to the battery/wall. Once it's CMOS to CMOS, then I'm don't _think_ it would hurt anything, and if it's a pull up resistor, well then obviously it'll be fine.

I can see why firmata would want to default to output, it avoids a read on every single pin at startup.

If this is how firmata does it, and the other platforms, then changing that would be a big uphill battle, and is certainly not a hill I want to die on.

I mostly wanted to check and make sure that wasn't a typo :)

fivdi, 2015-09-04:
I'd imagine things go well a lot of the time without causing damage due to timing.

Let's say someone uses [this toggle switch circuit](http://johnny-five.io/examples/toggle-switch/), but the switch is flipped to the left, rather than to the right as shown in the fritzing. If the program shown is run, pin 8 will be configures as an output, set low, and wired to 5V initially, but probably only for a very short amount of time. `new five.Switch(8)` results in pin 8 being configured as an input fairly quickly, so all probably goes well.

However, if someone uses the same circuit, also with the switch flipped to the left rather than the right, and then runs a program that does nothing with pin 8, then I can't imagine damage not being caused. I'm almost tempted to give it a try :)