nebrius/raspi-io #33: Allow GPIO exclusions

dexterlabora, 2015-09-22:
I'm writing an Arduino/Raspberry Pi/Lego blog (www.internetoflego.com) and was going to include some of your work. As I was testing a project, I realized that I have a major conflict and was hoping you could point me in the right direction.

Setup:
Raspberry Pi v2 
Johnny-Five
LIRC for Lego IR Power Functions control 
     https://github.com/dspinellis/lego-power-scratch
My Code (still needs tested)
      https://gist.github.com/dexterlabora/49ab4509ca4ef5edc8fd

The issue I am having is that in order to use the LIRC infrared driver/commands directly from Linux, pins 22 & 23 are reserved for this service. As I try to run a J5 app, I'm getting the following error, likely because the pins are already in use.
(Error: Pin "22" does not support mode "2"). 

I had a look through the index.js file, and assume I could write some code to bypass the initialization of these pins, but before I go down that rabbit hole, I was hoping you could guide me on this. I want a clean (less hacky) way of doing this as I write my instructional blog for the general public. Currently, I have to use the Rasp Pi for only the LIRC functions and all other GPIO related stuff on an Arduino. This works, but obviously I'm limiting my self here and requiring two controller boards.

Thanks for you help and the work you've done so far. I will be sure to credit you in the blog post!

~~ Console Error ~~

1442855141613 Device(s) RaspberryPi-IO  

1442855141665 Connected RaspberryPi-IO  

1442855141681 Repl Initialized  

>> 

Error: Pin "22" does not support mode "2"

    at Raspi.pinMode (/root/IoL/node_modules/raspi-io/lib/index.js:345:17)

    at new Sensor (/root/IoL/node_modules/johnny-five/lib/sensor.js:79:11)

    at new module.exports.Digital (/root/IoL/node_modules/johnny-five/lib/johnny-five.js:87:10)

    at Board.<anonymous> (/root/IoL/citystation.js:32:28)

    at Board.emit (events.js:92:17)

    at process._tickCallback (node.js:448:13)

Thanks again,

Cory

www.InternetOfLego.com

nebrius, 2015-09-23:
Looking at it more in depth, it's not because the pins are reserved, it's because Johnny-Five is trying to set the pin mode to analog input, which the Raspberry Pi does not support. So, the error message is technically correct.

Turns out, this is a bug in Johnny-Five itself. I submitted a PR to fix this at https://github.com/rwaldron/johnny-five/pull/908

closed 2015-09-23 by nebrius

dexterlabora, 2015-09-23:
Thank you very much!