nebrius/raspi-io #34: Getting started

wmmihaa, 2015-10-01:
Hi, 
I'm trying to get started with Johnny-five. I got a Raspberry Pi 2 model B, and a RGB Led called KY-016. I've connected it to 5V, GPIO17, GPIO21 & GPIO22, and whenever I try to do:
var a = new five.LED.RGB([ 'GPIO017', 'GPIO021', 'GPIO022' ]);

.. it fails with "PIN Error: 0 is not a valide PWM pin (Led.RGB)"

I understand that GPIO17 is pin #0, but I've tried other pins as well, but all with the same result. What am I doing wrong?
```javascript
var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi()
});
board.on("ready", function() {
  var a = new five.Led.RGB([ 'GPIO17', 'GPIO21', 'GPIO22' ]);  
});
```

Thanks
Mikael

nebrius, 2015-10-01:
Hi @wmmihaa! So, you're not going to like what I have to say. You can't use the Led.RGB class on the Raspberry Pi :(

(in depth explanation of how RGB LEDs work, feel free to skip if you already know). To do RGB leds, we have to be able to control the brightness of the LED. This is done using Pulse Width Modulation (PWM), which sends pulses at regular intervals. We set it up so that the pulses are sent so fast that the human eye can't see them, making their brightness correlate with the length of pulses. This is why we **require** all of the pins to support PWM.

The Raspberry Pi only has 2 PWM ports on it, not 3+ like the other boards do, which means you can't control all of the LEDs in the RGB. You can see what each pin supports at https://github.com/nebrius/raspi-io/wiki#model-abraspberry-pi-2

One thing you can do is create three separate LED instances and toggle them on and off. You'll only get 8 colors, but it's something at least.

Let me know if this helps.

wmmihaa, 2015-10-02:
Thank you Bryan, for taking the time to explain this. I'm pretty new to this, so I really appreciate the lesson :-).
I'll just continue to work through my box of thirty-seven sensors to find something that works. BTW you guys should sell a start-kit of sensors. I really like the API, but I struggle a bit with which sensors that works, as they don't always come with model numbers...

Anyway, thanks for the reply and for the work you've done.
Mikael

nebrius, 2015-10-02:
FWIW sparkfun.com has some good kits for working with sensors for the Arduino, maybe they have one for the Raspberry Pi too?

Best of luck!

closed 2015-10-02 by nebrius

rwaldron, 2015-10-02:
It should be possible using an `Expander` instance and a PCA9685: 

```js
var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi()
});
board.on("ready", function() {
  var expander = new five.Expander("PCA9685");
  var a = new five.Led.RGB({
    board: expander, 
    pins: [ 0, 1, 2 ]
  });  

  // ...
});
```

Here's a basic example of wiring the PCA9685, just swap out the Arduino: http://johnny-five.io/api/expander/#pca9685

rwaldron, 2015-10-02:
Actually, I completely forgot that we had a built-in PCA9685 controller for `Led.RGB`: http://johnny-five.io/examples/led-rgb-anode-PCA9685/

nebrius, 2015-10-02:
woo expanders!

rwaldron, 2015-10-02:
> woo expanders!

Eventually the internals of `Led.RGB` will just use an `Expander`, but not yet. Either way: Expanders are totally awesome!