nebrius/raspi-io #35: Upgrade to NAN2

nebrius, 2015-10-09:
We need NAN2 in order to get raspi-io to work on Node 4.

One thing to note is that, practically speaking, we need Raspbian Jesse for this due to the hard dependency on gcc 4.8. It is possible to install gcc 4.8 in wheezy alongside 4.6, but this provides a pretty high barrier to entry for people that we should avoid, in my opinion.

Part of this upgrade should be to add friendly error messages when attempting to install on wheezy.

It is also possible that NAN1 is the reason for https://github.com/rwaldron/johnny-five/issues/920. Either way, that issue will hopefully be resolved once the upgrade to NAN2 is complete.

Something to consider in the process: should this cause the bump to raspi-io 5 or raspi-io 4.1? Would this be considered a breaking change?

fivdi, 2015-10-12:
> Part of this upgrade should be to add friendly error messages when attempting to install on wheezy.

If the appropriate C++11 compiler isn't installed, NAN v2.1.0 will generate a compile error [here](https://github.com/nodejs/nan/blob/v2.1.0/nan.h#L34-L42). Not sure if Wheezy is the only relevant OS either.

nebrius, 2015-10-12:
probably not...I was thinking about including instructions in the raspi-* error message for installing the right version of gcc, or perhaps linking to a wiki doc. I would imagine that the instructions will be detailed for raspbian, with a "if you're using a different distro, do something similar" type message, since most people use raspbian.

nebrius, 2015-10-28:
Fixed with Raspi IO 5.0.0

closed 2015-10-28 by nebrius