nebrius/raspi-io #38: raspi-pwm REPLACE_INVALID_UTF8 when installing

francesco-vannini, 2015-11-29:
Steps:
1) On a brand new Jessie install npm
2) npm install raspi-io

Versions:
HW: Pi B
OS: Jessie
node -v v0.10.29
node-gyp -v v0.12.2
npm -v 1.4.21

Error:
> raspi-pwm@2.0.0 install /home/pi/node_modules/raspi-io/node_modules/raspi-pwm
> node-gyp rebuild

make: Entering directory '/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/build'
  CXX(target) Release/obj.target/addon/src/addon.o
In file included from ../src/addon.cc:26:0:
../node_modules/nan/nan.h:328:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
addon.target.mk:84: recipe for target 'Release/obj.target/addon/src/addon.o' failed
make: *** [Release/obj.target/addon/src/addon.o] Error 1
make: Leaving directory '/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.1.13+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/node_modules/raspi-io/node_modules/raspi-pwm
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian

francesco-vannini, 2015-11-29:
It would seem that all I needed to do was to run the command as sudo which I had assumed not to be necessary.

closed 2015-11-29 by francesco-vannini

nebrius, 2015-11-30:
Glad you got it working!

francesco-vannini, 2015-12-01:
Thanks and sorry if I had rushed into wrong conclusions a bit too quickly.