nebrius/raspi-io #40: Cannot find '../build/Release/addon'

ekryski, 2015-12-03:
I have a few moving parts here so bear with me... I get this error when trying to run the basic example:

```
root@281e58f4e6b1:/srv/pi-led-test# node led.js
module.js:338
    throw err;
    ^

Error: Cannot find module '../build/Release/addon'
    at Function.Module._resolveFilename (module.js:336:15)
    at Function.Module._load (module.js:286:25)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/srv/pi-led-test/node_modules/raspi-io/node_modules/raspi/lib/index.js:30:29)
    at Module._compile (module.js:434:26)
    at Object.Module._extensions..js (module.js:452:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
```
I'm running this using raspi-io v5.0.0 on a Pi Zero inside a docker container. The OS is [4.1.12-hypriotos+](http://blog.hypriot.com/post/just-in-time-for-DockerCon-EU-a-shipload-of-new-Docker-ARM-goodies/), which is basically a custom Raspbian Jessie build.

The funny thing is, I managed to install all the dependencies successfully. The only thing that had a soft failure to compile natively is execSync, but the whole process was fine. The error is below.

```
...
npm info install serialport@2.0.5

> execSync@1.0.2 install /srv/pi-led-test/node_modules/raspi-io/node_modules/raspi-i2c/node_modules/execSync
> node install.js

[execsync v1.0.2] Attempting to compile native extensions.
[execSync v1.0.2]
    Native code compile failed!!
npm info postinstall raspi-gpio@2.0.0
...
```

If anyone has any ideas, I'd greatly appreciate some direction.

ekryski, 2015-12-04:
So I've figured out that installing and attempting to run my script as **root** is the problem. Not entirely sure why it's causing a problem but if I `su pi` and install my deps and run my script in the pi user's home directory it's all good.

nebrius, 2015-12-05:
That's...strange. I would guess it has something to do with docker shenanigans, but honestly that's just a guess. I don't really know. tbqh, I only test on Raspbian Jesse directly, so I wouldn't be too surprised if there are issues with other configurations (even if they're just derivations).

ekryski, 2015-12-05:
So it doesn't appear to be a docker issue. When I install as root it's a no go, but if I install as the `pi` user all good. Honestly, I have no idea why but basically don't install as the root user and you are all good.

closed 2015-12-05 by ekryski

nebrius, 2015-12-05:
kk good to know. Sorry I don't have any ideas/leads for you to work off of.

mwittig, 2016-01-23:
I am having the same issue here. It would be great to add a trouble shooting hint as part of the README. Besides this I think it would be great, if a solution could be found for the issue.

nebrius, 2016-01-23:
@mwittig were you able to get it working using the workaround @ekryski mentioned above?

mwittig, 2016-01-24:
@nebrius Kind of ... if I install with an unprivileged user it works as ecpected. However, in my case raspi-io is used as part of a home-automation server which provides an update servive for plugins (nodejs packages). As the server runs as root (some packages like wiringPI require root privileges on wheezy at least) the package installation is performed as root. I have also tried to add `"config": { "unsafe-perm":true }` to the package.json but this did not help. So, I guess, the only solution at the moment is to change the update procedure of the home automation server to perform a change user at the beginning of the installation procedure.

Regarding this issue ticket my suggestion is to add the note in the README as it seems to be pitfall for a few users.

nebrius, 2016-01-26:
@mwittig can you submit that as a PR?

mwittig, 2016-01-26:
Sorry, I missed this. Done now.