nebrius/raspi-io #41: Install failiure

bigandy, 2015-12-10:
I am trying into npm install raspi-io but getting the following error warnings when I run `sudo npm install raspi-io --save`

> raspi-gpio@2.1.0 install /var/www/html/raspberry-js/node_modules/raspi-gpio
> node-gyp rebuild

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/4.2.1"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/var/www/html/raspberry-js/node_modules/raspi-gpio/.node-gyp"
make: Entering directory '/var/www/html/raspberry-js/node_modules/raspi-gpio/build'
  CXX(target) Release/obj.target/addon/src/addon.o
  CXX(target) Release/obj.target/addon/src/init.o
../src/init.cc:27:22: fatal error: wiringPi.h: No such file or directory
 #include <wiringPi.h>
                      ^
compilation terminated.
addon.target.mk:91: recipe for target 'Release/obj.target/addon/src/init.o' failed
make: *** [Release/obj.target/addon/src/init.o] Error 1
make: Leaving directory '/var/www/html/raspberry-js/node_modules/raspi-gpio/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.1.13+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /var/www/html/raspberry-js/node_modules/raspi-gpio
gyp ERR! node -v v4.2.1
gyp ERR! node-gyp -v v3.2.0
gyp ERR! not ok
npm WARN raspberry-js@1.0.0 No description
npm WARN raspberry-js@1.0.0 No repository field.
npm ERR! Linux 4.1.13+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "raspi-io" "--save"
npm ERR! node v4.2.1
npm ERR! npm  v3.5.1
npm ERR! code ELIFECYCLE

npm ERR! raspi-gpio@2.1.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the raspi-gpio@2.1.0 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the raspi-gpio package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs raspi-gpio
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls raspi-gpio
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /var/www/html/raspberry-js/npm-debug.log

bigandy, 2015-12-14:
yes that worked. Turns out issue is to do with `serial` module not supporting node 5. Will try with node 4.

cprior, 2016-01-11:
On a Raspbian Jessie Lite 2015-11-21 npm install raspi-wiringpi failed because git wasn't installed.
An apt-get install git solved it,
then npm install raspi-wiringpi pulled in i2c-tools

nebrius, 2016-01-12:
hmm...sounds like the Git check is failing for some reason: https://github.com/nebrius/raspi-wiringpi/blob/master/install-wiringpi.sh#L35

IgnatG, 2016-02-09:
try 
sudo npm install -g --unsafe-perm raspi
and then 
sudo npm install -g --unsafe-perm raspi-i2c

lukechilds, 2016-02-10:
I've just had the same problem, fresh install of Rasbian Lite (2016-02-09) with latest stable node (v5.6.0) installed via nvm. I was getting the following error:
```
../src/init.cc:27:22: fatal error: wiringPi.h: No such file or directory
 #include <wiringPi.h>
                      ^
compilation terminated.
```
Tried installing Git and re-installing but got the same error.
I've just manually installed `raspi-wiringpi` which ran fine and installed WiringPi, then tried `raspi-io` again and it works. It seems like `raspi-io` isn't actually installing WiringPi by itself.

Full terminal output if it helps:
```
pi@raspberrypi:~/projects/car $ npm install raspi-io --save

> raspi-gpio@2.1.0 install /home/pi/projects/car/node_modules/raspi-gpio
> node-gyp rebuild

make: Entering directory '/home/pi/projects/car/node_modules/raspi-gpio/build'
  CXX(target) Release/obj.target/addon/src/addon.o
  CXX(target) Release/obj.target/addon/src/init.o
../src/init.cc:27:22: fatal error: wiringPi.h: No such file or directory
 #include <wiringPi.h>
                      ^
compilation terminated.
addon.target.mk:91: recipe for target 'Release/obj.target/addon/src/init.o' failed
make: *** [Release/obj.target/addon/src/init.o] Error 1
make: Leaving directory '/home/pi/projects/car/node_modules/raspi-gpio/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/.nvm/versions/node/v5.6.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:100:13)
gyp ERR! stack     at ChildProcess.emit (events.js:185:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.1.17+
gyp ERR! command "/home/pi/.nvm/versions/node/v5.6.0/bin/node" "/home/pi/.nvm/versions/node/v5.6.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/projects/car/node_modules/raspi-gpio
gyp ERR! node -v v5.6.0
gyp ERR! node-gyp -v v3.2.1
gyp ERR! not ok
npm WARN car@1.0.0 No repository field.
npm ERR! Linux 4.1.17+
npm ERR! argv "/home/pi/.nvm/versions/node/v5.6.0/bin/node" "/home/pi/.nvm/versions/node/v5.6.0/bin/npm" "install" "raspi-io" "--save"
npm ERR! node v5.6.0
npm ERR! npm  v3.6.0
npm ERR! code ELIFECYCLE

npm ERR! raspi-gpio@2.1.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the raspi-gpio@2.1.0 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the raspi-gpio package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs raspi-gpio
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls raspi-gpio
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/projects/car/npm-debug.log
pi@raspberrypi:~/projects/car $ npm install raspi-wiringpi

> raspi-wiringpi@1.0.6 install /home/pi/projects/car/node_modules/raspi-wiringpi
> sh install-wiringpi.sh

Installing I2C tools
Reading package lists... Done
[... 116 lines cut ...]
  CXX(target) Release/obj.target/addon/src/read.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory '/home/pi/projects/car/node_modules/raspi-gpio/build'

> raspi-wiringpi@1.0.6 install /home/pi/projects/car/node_modules/raspi-wiringpi
> sh install-wiringpi.sh

Wiring Pi is already installed, skipping

car@1.0.0 /home/pi/projects/car
├─┬ raspi@2.0.0
│ └── nan@2.2.0
├─┬ raspi-i2c@2.2.0
│ └─┬ i2c-bus@1.0.2
│   └── nan@2.2.0
├─┬ raspi-io@5.2.0
│ └─┬ raspi-gpio@2.1.0
│   └── nan@2.2.0
├─┬ raspi-pwm@2.1.0
│ └── nan@2.2.0
└── raspi-wiringpi@1.0.6

npm WARN car@1.0.0 No repository field.
```

nebrius, 2016-02-10:
I have one suspicion that the permission changes in npm run scripts with v3 might be at play here. I'll try and investigate once I get some free time. That said...it may be possible for me to remove the dependency on wiringpi entirely, and it may make sense to move to that in the long run.

Thanks for all the bug reports everyone. Hopefully we'll find the issue yet!

Worie, 2016-03-10:
Hi, just bought B+ and it appears that i have the same problem as described above. 
When trying to install raspi-io from npm:

npm install raspi-io
> raspi-wiringpi@1.0.6 install /usr/local/lib/node_modules/raspi/node_modules/raspi-wiringpi
> sh install-wiringpi.sh

Wiring Pi is already installed, skipping

> raspi@2.0.0 install /usr/local/lib/node_modules/raspi
> node-gyp rebuild

make: Entering directory '/usr/local/lib/node_modules/raspi/build'
  CXX(target) Release/obj.target/addon/src/addon.o
In file included from ../src/addon.cc:26:0:
../node_modules/nan/nan.h:328:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
addon.target.mk:83: recipe for target 'Release/obj.target/addon/src/addon.o' failed
make: *** [Release/obj.target/addon/src/addon.o] Error 1
make: Leaving directory '/usr/local/lib/node_modules/raspi/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.1.18+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /usr/local/lib/node_modules/raspi
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! raspi@2.0.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the raspi@2.0.0 install script.
npm ERR! This is most likely a problem with the raspi package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls raspi
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.1.18+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "-g" "--unsafe-perm" "raspi"
npm ERR! cwd /home/pi/rpi-car
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/rpi-car/npm-debug.log
npm ERR! not ok code 0
pi@raspberrypi:~/rpi-car $ npm install raspi-io
 
> raspi@2.0.0 install /home/pi/rpi-car/node_modules/raspi-io/node_modules/raspi
> node-gyp rebuild

make: Entering directory '/home/pi/rpi-car/node_modules/raspi-io/node_modules/raspi/build'
  CXX(target) Release/obj.target/addon/src/addon.o
In file included from ../src/addon.cc:26:0:
../node_modules/nan/nan.h:328:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
addon.target.mk:83: recipe for target 'Release/obj.target/addon/src/addon.o' failed
make: *** [Release/obj.target/addon/src/addon.o] Error 1
make: Leaving directory '/home/pi/rpi-car/node_modules/raspi-io/node_modules/raspi/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.1.18+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/rpi-car/node_modules/raspi-io/node_modules/raspi
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN deprecated graceful-fs@1.2.3: graceful-fs version 3 and before will fail on newer node releases. Please update to graceful-fs@^4.0.0 as soon as possible.
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 

> execSync@1.0.2 install /home/pi/rpi-car/node_modules/raspi-io/node_modules/raspi-i2c/node_modules/execSync
> node install.js

module.js:340
    throw err;
          ^
Error: Cannot find module '/home/pi/rpi-car/node_modules/raspi-io/node_modules/raspi-i2c/node_modules/execSync/install.js'
    at Function.Module._resolveFilename (module.js:338:15)
    at Function.Module._load (module.js:280:25)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 

> raspi-pwm@2.3.1 install /home/pi/rpi-car/node_modules/raspi-io/node_modules/raspi-pwm
> node-gyp rebuild

I have a clean install of Raspbian Jessie.
nodejs version : v0.10.29
npm version: 1.4.21
gcc version: 4.9.2 

The solutions mentioned above didn't solve my problem, perhaps someone has some tips that can help get it to work? Thanks in advance!

fivdi, 2016-03-10:
@Worie See https://github.com/fivdi/onoff/wiki/Node.js-v0.10.29-and-native-addons-on-the-Raspberry-Pi for solutions and a workaround to the `‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’` compile error.

nebrius, 2016-03-10:
Also, just so you are aware, node 0.10 support will be removed from raspi-io in the near future

BTBlueSkies, 2016-03-19:
Same issue here. I have a clean install on the rasp-pi 2 today, then removed the origional nodejs and compiled the new 4.4.0. I installed johnny-five then raspi-io and received this:

npm WARN deprecated graceful-fs@1.2.3: graceful-fs version 3 and before will fail on newer node releases. Please update to graceful-fs@^4.0.0 as soon as possible.

I then 'npm graceful-fs --save' and no errors. 

However I see both in the listing.. npm ls | grep gra .. returns
pi@raspberrypi:~/c9_workspace/NoteCheckout $ npm ls | grep gra
├── graceful-fs@4.1.3
│ │ │ │ │ │ └── graceful-readlink@1.0.1
│ │ │ │ │ └── graceful-fs@4.1.2
│ │ │   │ ├── graceful-fs@4.1.2
│ │ │ │ └── graceful-readlink@1.0.1
  │ │     └── graceful-fs@1.2.3

I am not sure what to make of it.. but thought it may help.

nebrius, 2016-03-21:
@BTBlueSkies fortunately that's actually not an error you need to worry about :). It's a dependency of execSync, which is used as a shim on Node 0.10, and if you're not running Node 0.10 it doesn't matter if it fails :)

All that said, I just released raspi-io 6.0.1 (and 6.0.0), which has a brand new build system and removes support for Node 0.10 entirely. Can everyone who was having issue please try the new version and see if it helps?

I've tried to reproduce these issues myself and it always works for me :-/

If you try raspi-io 6 and it still doesn't work, can you give me the following information?
- Node version and architecture (e.g. `Node 5.9.0 armv7`) and how you installed it
- Raspberry Pi model (e.g. `Raspberry Pi 2 Model B`)
- Filename of the image you flashed to the SD card (e.g. `2016-02-26-raspbian-jessie.img`)

BTBlueSkies, 2016-03-21:
I'd be happy to give this a go.

Can you advise me on this idea.. I can remove the raspi-io and the
johnny-five via npm. Then I can re-install them. Do you think this is a
valid test? If not, I would be happy to reset the pi to a clean image and
try it with a fresh system.

Thoughts?

nebrius, 2016-03-21:
If you don't mind trying it on a fresh system, that would be best I think. There's a lot of caching that goes on by npm and some of the install scripts that makes a second run usually perform differently. Thanks for offering to help!

BTBlueSkies, 2016-03-22:
Okay.. A clean install of noobs.1.8.0 for a raspberry pi 2.

Here was my process...

sudo apt-get remove node-red
sudo apt-get remove nodejs

wget https://nodejs.org/dist/v4.4.0/node-v4.4.0.tar.gz
tar -xzf node-v4.4.0.tar.gz
cd node-v4.4.0
./configure
make
sudo make install

sudo npm install raspi-io

..................................

this is the output of the npm install. I stopped here. If you'd like logs
or other resources, let me know. I have left the system at this state.

...................................

pi@raspberrypi:~ $ sudo npm install raspi-io

> raspi-wiringpi@1.0.6 install
/home/pi/node_modules/raspi-io/node_modules/raspi/node_modules/raspi-wiringpi
> sh install-wiringpi.sh

Wiring Pi is already installed, skipping

|
> raspi-gpio@2.2.2 install
/home/pi/node_modules/raspi-io/node_modules/raspi-gpio
> node-gyp rebuild

gyp WARN EACCES user "root" does not have permission to access the dev dir
"/root/.node-gyp/4.4.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir
"/home/pi/node_modules/raspi-io/node_modules/raspi-gpio/.node-gyp"
make: Entering directory
'/home/pi/node_modules/raspi-io/node_modules/raspi-gpio/build'
  CXX(target) Release/obj.target/addon/src/addon.o
  CXX(target) Release/obj.target/addon/src/init.o
  CXX(target) Release/obj.target/addon/src/write.o
  CXX(target) Release/obj.target/addon/src/read.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory
'/home/pi/node_modules/raspi-io/node_modules/raspi-gpio/build'
npm WARN cannot run in wd raspi@2.0.0 node-gyp rebuild
(wd=/home/pi/node_modules/raspi-io/node_modules/raspi)

> raspi-pwm@3.0.0 install
/home/pi/node_modules/raspi-io/node_modules/raspi-pwm
> node-gyp rebuild

gyp WARN EACCES user "root" does not have permission to access the dev dir
"/root/.node-gyp/4.4.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir
"/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/.node-gyp"
make: Entering directory
'/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/build'
  CXX(target) Release/obj.target/addon/src/addon.o
  CXX(target) Release/obj.target/addon/src/init.o
  CXX(target) Release/obj.target/addon/src/write.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory
'/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/build'
[... 94 lines cut ...]
"raspi-io"
npm ERR! node v4.4.0
npm ERR! npm  v2.14.20
npm ERR! code ELIFECYCLE

npm ERR! serialport@2.0.6 install: `node-pre-gyp install
--fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@2.0.6 install script 'node-pre-gyp
install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project
with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log

BTBlueSkies, 2016-03-22:
One more thing... I can do this over and over to help get a clean install.
Just let me know :)

nebrius, 2016-03-22:
Ahhh, so you're using the NOOBs OS not Raspbian. I seem to remember hearing once that NOOBs is based on a really outdated version of Raspbian. Switching to Raspbian should clear up all of your issues (it's the only OS that's technically supported by raspi-io).

If you want to stick with NOOBs though, I'd start by checking the output of `gcc -v`, and make sure it's version 4.8 or newer. If it's not, that's the cause of the compilation errors and you'll need to upgrade gcc.

fivdi, 2016-03-22:
Does installing with `npm install raspi-io` rather than `sudo npm install raspi-io` improve things?

BTBlueSkies, 2016-03-22:
NOOBs is an image that installs Raspbian. It has the debian Jessey kernel
and gcc version 4.9 or so. I believe it is their newest image of Raspbian.

I will try the raw image as well. I just like the installers on the NOOBs
image as they make it easy to say, reflash to a new install :).

BTBlueSkies, 2016-03-24:
Hey, I have setup a with a new raw Raspbian image. Node.js is compiling now
and should be done here in a short bit.

Question. Would you like me to do anything specific?

I was planning on the next two commands..

*sudo make install*

*sudo npm install raspi-io*

You asked me to try the npm install raspi-io without the sudo in the last
exchange. I will try that first, then move fwd with the sudo if that doesnt
work.

If you have anything you would like me to alter, let me know.

nebrius, 2016-03-24:
I'm curios to see the difference. I suspect there is a difference...somewhere...probably on something small...but a difference nonetheless.

BTBlueSkies, 2016-03-24:
Success!

I still have that other sd card as I left it. I am going to try something..
I think I may have been in an improper directory when I ran the install.

I will try it again on the other system and let you know.

output below
-------------------------------------

pi@raspberrypi:~/Raspi.io $ npm install raspi-io

> raspi-wiringpi@1.0.6 install
/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi/node_modules/raspi-wiringpi
> sh install-wiringpi.sh

Wiring Pi is already installed, skipping

|
> raspi@2.0.0 install
/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi
> node-gyp rebuild

make: Entering directory
'/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi/build'
  CXX(target) Release/obj.target/addon/src/addon.o
  CXX(target) Release/obj.target/addon/src/init.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory
'/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi/build'

> raspi-pwm@3.0.0 install
/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-pwm
> node-gyp rebuild

make: Entering directory
'/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-pwm/build'
  CXX(target) Release/obj.target/addon/src/addon.o
  CXX(target) Release/obj.target/addon/src/init.o
  CXX(target) Release/obj.target/addon/src/write.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory
'/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-pwm/build'

> raspi-gpio@2.2.2 install
/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-gpio
> node-gyp rebuild

make: Entering directory
'/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-gpio/build'
  CXX(target) Release/obj.target/addon/src/addon.o
  CXX(target) Release/obj.target/addon/src/init.o
  CXX(target) Release/obj.target/addon/src/write.o
  CXX(target) Release/obj.target/addon/src/read.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory
'/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-gpio/build'

> i2c-bus@1.0.2 install
/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-i2c/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory
'/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-i2c/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory
'/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-i2c/node_modules/i2c-bus/build'

> serialport@2.0.6 install
/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-serial/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory
'/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-serial/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
  COPY
/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-serial/node_modules/serialport/build/Release/node-v46-linux-arm/serialport.node
  TOUCH Release/obj.target/action_after_build.stamp
make: Leaving directory
'/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-serial/node_modules/serialport/build'

> raspi-i2c@3.0.0 postinstall
/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-i2c
> echo "Enabling I2C at boot time, you may be asked for your password" &&
sudo env "PATH=$PATH" script/enable_i2c.js

Enabling I2C at boot time, you may be asked for your password
Checking if I2C is enabled at boot time
Enabled I2C at boot time
YOU MUST REBOOT YOUR PI BEFORE USING I2C!

> raspi-serial@1.3.1 postinstall
/home/pi/Raspi.io/node_modules/raspi-io/node_modules/raspi-serial
> echo "Disabling serial port login at boot time, you may be asked for your
password" && sudo env "PATH=$PATH" script/enable_serial.js

Disabling serial port login at boot time, you may be asked for your password
Checking if serial port login is enabled
Serial port login does not appear to be enabled, skipping
raspi-io@6.0.1 node_modules/raspi-io
├── raspi-board@3.0.0
├── raspi-led@1.3.2 (raspi-peripheral@1.5.2)
├── raspi@2.0.0 (raspi-wiringpi@1.0.6, nan@2.2.0)
├── raspi-pwm@3.0.0 (raspi-peripheral@1.5.2, nan@2.2.0)
├── raspi-gpio@2.2.2 (raspi-peripheral@1.5.2, nan@2.2.0)
├── raspi-i2c@3.0.0 (ini-builder@1.0.3, raspi-peripheral@1.5.2,
i2c-bus@1.0.2)
└── raspi-serial@1.3.1 (raspi-peripheral@1.5.2, serialport@2.0.6)
pi@raspberrypi:~/Raspi.io $

BTBlueSkies, 2016-03-24:
> On the SD card from the other, with the NOOBS install, here are the
> results.
>
> npm install raspi-io: Same success as this last time.
> sudo npm install raspi-io: same fail as the first time.
>
> -------------
> Hope that helps, and thanks for your contributions to this stuff :)
>
> Smiles,
> Karl
>
> On Wed, Mar 23, 2016 at 9:44 PM, Karl Easterly <k.e@bigtexansoftware.com>

Sorry.. one clarification..

The second 'sudo npm install raspi-io' failed the same as the first time..
this was on the new card (without the noobs).

In short.. sudo and the package node-gyp seem to have some permission
related constraints. I suspect this per the output messages.

........

In the future, if you need anything tested on clean systems, ping me. I am
a programmer, unix/win, by trade and happy to help another with their
testing, especially on tools I actively use :)

nebrius, 2016-04-11:
Sorry, this kinda slipped off my radar for a while. I'm not surprised running with sudo causes problems...it's generally recommended that you only use sudo to install global packages, and even then only as a last resort.

If you don't use sudo, which for the record I never use sudo, does that help?

BTBlueSkies, 2016-04-12:
No worries... I appreciate the followup.

Sudo, not using sudo.. in this world of open source.. there is no right
answer. So many projects have no concept of why sudo even exist.. yet many
others are spot on. I take each as they come. In this case, that's what I
read on some blog somewhere... so sudo it was :)... till you taught me on
this build it's more of a problem than a solution.

Smiles.

nebrius, 2016-10-13:
I was never able to reproduce this issue, and since there hasn't been activity in a while, I'm closing this issue. Please reopen if you're still experiencing issues!

closed 2016-10-13 by nebrius

lastmjs, 2016-11-25:
With my PI 3 Model B, installing raspi-io failed saying that wiringPi.h couldn't be found. I had to separately install wiring pi, and then I could reinstall raspi-io again and it worked.

nebrius, 2016-11-27:
@lastmjs I just tweaked raspi-io w/ version 7.0.0 to not include Wiring Pi at all. It's included by default in Raspbian and shouldn't be necessary. Removing it simplified a lot of the installation, and hopefully that will work around this issue.