nebrius/raspi-io #45: servo.sweep() only ranges about 90 degrees

cprior, 2016-02-02:
Not sure if this is an issue with my hardware, raspi-io or johnny-five:

I have a servo attached to a Raspberry Pi Zero with the latest Raspbian Jessie image.
When I run Python scripts I can make the servo traval its ~180 degrees, up to the mechanical limits.
But with Johnny-Five or Raspi-io I only get about 90 degrees of sweep motion.

The servo is attached in both attempts to the RPi 5V&GND.

When I attach the same servo (I have actually 2 with the same behaviour) to an Arduino with StandardFirmataPlus, I get about 170 degrees of motion.

How can I influence the range of a servo, ideally I'd get 175 degrees. This is my first project in Node.js and I'd like to do this not with Python -- but I really need as close to 180 degrees as possible (rising marbles with a wheel).

Here is the stripped-down code:

```
var five = require("johnny-five");
var Raspi = require("raspi-io");

var board = new five.Board({
  io: new Raspi()
});

board.on("ready", function() {

  // board.pinMode("P1-32", five.Pin.SERVO);

  var servo = new five.Servo({
    pin: "P1-32",
    type: "standard",
    range: [0, 180],
    startAt: 90
  });

  this.repl.inject({
    board: board,
    servo: servo,
  });

});
```

And a quick video: https://youtu.be/x54wt5BWxC4

\# node --version
v4.2.4
\# uname -a
Linux raspberrypi 4.4.0-rc5+ #10 PREEMPT Thu Dec 24 22:52:14 EST 2015 armv6l GNU/Linux
\# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
\# cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
...
Hardware	: BCM2708
Revision	: 900092

rwaldron, 2016-02-02:
Can you link to or paste the python scripts that worked correctly?

rwaldron, 2016-02-02:
Here's my initial guess: the python scripts might be using a wider PWM range. If that's the case, then raspi-io needs to implement `servoConfig(pin, pwm low, pwm high)` (this should configure a pin's timer to the specified range, or store the range in memory to scale 0-180 -> min-max). That will allow you to specify the exact PWM range that you want a servo to use: 

```js
var servo = new five.Servo({
  pin: "P1-32",
  pwmRange: [pwm low, pwm high],
  startAt: 90
});
```

Hopefully @nebrius forgives me for completely forgetting to document this in the io-plugins document.

-------

But, it might be something different altogether.

cprior, 2016-02-02:
Ah, I didn't think of that, sorry -- sure:
From http://tutorials-raspberrypi.de/gpio/servo-motor-mit-dem-raspberry-pi-steuern/ I just used in interactive Python the lines
```
import RPi.GPIO as GPIO
import time

servoPIN = 26
GPIO.setmode(GPIO.BCM)
GPIO.setup(servoPIN, GPIO.OUT)

p = GPIO.PWM(servoPIN, 50) # GPIO 18 als PWM mit 50Hz
p.start(2.5)
```

and then I was able to 
`p.ChangeDutyCycle(5)` between the values 2.5 and 10.5

cprior, 2016-02-02:
Ah, it seems that
https://github.com/rwaldron/johnny-five/blob/d09cffd9e5132555924de8c5cf988a0478ce0e5c/lib/servo.js#L57
is the original J5 code in question?

nebrius, 2016-02-02:
> servoConfig(pin, pwm low, pwm high)

Soooo, yeah, never heard of that function...and it's definitely not in raspi-io.

@cprior can you post the manufacturer and model number of your servo?

cprior, 2016-02-02:
One is a Sparkfun servo, ordered from the reseller
http://www.exp-tech.de/servo-large-full-rotation-rob-09347 ,
datasheet at http://www.sparkfun.com/datasheets/Robotics/servo-360_e.pdf

Then I have three chinese from "Hobbyking": Turnigy TGY-AN13 
__85406__Turnigy_TGY_AN13_Standard_Analog_Car_Servo_14_5kg_0_13sec_58g_AR_Warehouse_.

Is there some common servo model that is recommended? I gladly order a known working model, but being new in all this jawdropping "robotics" I am not familiar with what's popular.

cprior, 2016-02-02:
One is an Adafruit servo, ordered from the reseller
http://www.exp-tech.de/analog-feedback-servo 

And then I have three chinese from "Hobbyking": Turnigy TGY-AN13 
__85406__Turnigy_TGY_AN13_Standard_Analog_Car_Servo_14_5kg_0_13sec_58g_AR_Warehouse_.

Is there some common servo model that is recommended? I gladly order a known working model, but being new in all this jawdropping "robotics" I am not familiar with what's popular.

rwaldron, 2016-02-02:
> Soooo, yeah, never heard of that function...and it's definitely not in raspi-io.

Like I said...

>> Hopefully @nebrius forgives me for completely forgetting to document this in the io-plugins document.

0_0

nebrius, 2016-02-02:
That servo looks like it should work fine. I'm going to go out on a limb here and say that this is a bug in Raspi IO. I've been worried for a while that the clocking values needed to set the PWM won't remain constant with new values, and it's possible my fears have finally been realized. Sadly, my RPi zero is _still_ backordered (😒), so I haven't been able to test anything on the Pi Zero myself, so I can't stick an o-scope on it to confirm.

@cprior by chance, do you have another Raspberry Pi that's not a Zero you can test on? If the servo works there, then that tells me it's something with the Zero's clocking scheme, but if it's not, then something else is going on.

I'm not convinced just yet that this is just an odd servo that needs tweaking via `servoConfig`, although it's definitely something that should be implemented.

nebrius, 2016-02-02:
FWIW I learned that the python library uses software PWM, not hardware PWM, but I use hardware PWM instead of software PWM. This means they would be immune to changes in clocking but I wouldn't, if indeed that is the problem.

One of these days I should get software PWM implemented...

cprior, 2016-02-02:
I'll setup a new SD card, I have all kinds of those boards available. I have a   RPi2 that is not in use at the moment. Might take until tomorrow though, almost midnight over here.

cprior, 2016-02-03:
On an RPi2 with the latest DietPi (stripped down Raspbian) and Node 4.2.4:

Same range of only about 90 degrees.

I have ordered a HS-485HB because  found it metioned in other issues.
Sorry if I caused a confusion -- but sometimes asking a question is my best bet in troubleshooting.

Will report back with RPi Zero and a HS-485HB this weekend (being a commuter the parcel won't reach me earlier).

Is there anything else that I could test with the RPi Zero?

nebrius, 2016-02-03:
Thanks for testing @cprior! Ok, so maybe it just is that servo after all. Let me know how the testing with the other servo goes! I suspect at this point that I do just need to implement servoConfig. @rwaldron do you think you could get that documented in https://github.com/rwaldron/io-plugins?

Thanks for the offer to test stuff too! If you have time, can you do a standard GPIO test with an LED, and also try and talk to something over I2C (temperature sensor, accelerometer, etc if you have one)?

rwaldron, 2016-02-04:
Back in 2014 I posted a video demonstrating that the 485HB is a "short range" servo (out of the box) https://www.youtube.com/watch?v=IcfBQYkxrZw

cprior, 2016-02-13:
I have ordered and received a 180-degree servo and it works fine from within Johnny Five on a Raspberry Pi Zero, the model is  Servo DES 718 BB, MG180 Grad 20 mm http://www.graupner.de/de/products/01642515-4b83-448d-8515-ba6d5e0a3d9d/7947/product.aspx

I also have tested the LED.blink and the sequence of blink(1000) is indeed a second: https://www.youtube.com/watch?v=UScStxuS79o

Still have to test a I2C component as all I have at hand is a RGB sensor which is not supported by Johnny Five (yet), the math is a bit involved anyways.
https://www.youtube.com/watch?v=UScStxuS79o&feature=youtu.be

But I should have some I2C equipment at the other flat -- being weekend commuter and such :(
Stuff if always in the other flat. *grrr* ;)

nebrius, 2016-02-15:
Hehe, there's no rush, and thank you for testing out what you have! So it's good to know the other servo worked fine too, just need to get that one method implemented.

rwaldron, 2016-02-15:
@cprior that's awesome—thanks for testing and reporting :)

nebrius, 2016-02-19:
servoConfig (along with servoWrite) are now implemented as of raspi-io 5.3.0! Let me know if that fixes things for you :)

nebrius, 2016-03-21:
Closing, as the fix is in. If there's still any issues, feel free to reopen

closed 2016-03-21 by nebrius

cprior, 2016-04-12:
Only today I managed to test it all again: The above code on a fresh installation of

- Raspbian Jessie 2016-03-18-raspbian-jessie
- on a Pi Zero
- node v4.1.2
- johnny-five@0.9.39
- raspi-io@6.0.1

does sweep about 180 degrees! *yay*
The servo is not of the explicitly-180-kind, but the one that used to
- sweep about 180 deg with Python
- and less with j5&&raspi-io

Thank you **very** much!