nebrius/raspi-io #46: Reading from Analog > Digital

nick-jonas, 2016-02-03:
I'm trying to read data from 2x10k potentiometers that are wired to the 1st two inputs of an MCP3008, and then the MCP30008 is hooked up to 4 GPIO pins on the RPi (GPIO23, 24, 25, 16).  

How do I go about doing this?  The only reference code I could find on johnny-five is for the Grove shield and Edison board:

    var rotary = new five.Sensor("A0");

    rotary.scale(0, 255).on("change", function() {
        // do stuff with this.value
    });

How do I create two rotary instances (what would I use for the GPIO pins)?

fivdi, 2016-02-03:
The MCP3008 is an SPI device but Johnny-Five and Raspi-IO currently don't support SPI devices. What should work is the PCF8591 which is an I2C device. I2C is supported. See http://johnny-five.io/examples/expander-PCF8591/

fivdi, 2016-02-03:
See also https://github.com/rwaldron/io-plugins/issues/3

nick-jonas, 2016-02-04:
Hmm tried using node-spi from @RussTheAerialist as mentioned in that thread but getting issues on the install with node-gyp:

https://github.com/RussTheAerialist/node-spi/issues/25

I'll order a PCF8591, and if there are any other ideas of things to try with the MCP3008, let me know - thanks!!

fivdi, 2016-02-04:
@nick-jonas what version of Node.js are you using? https://github.com/RussTheAerialist/node-spi supports v0.10 and v0.12, [but not v4 or higher](https://github.com/RussTheAerialist/node-spi/issues/21#issuecomment-154093397)

nick-jonas, 2016-02-04:
@fivdi v0.10.36

nebrius, 2016-02-19:
Hi @nick-jonas! Just so you're aware, using node-spi and this SPI device won't make the analog data available to Johnny-Five. You can certainly still use it, but you'll have to parse the SPI data yourself, and wire up your own event logic independent of Johnny-Five :(

If at all possible, I'd recommend going with the I2C backpack that @fivdi mentioned.

nebrius, 2016-03-21:
I believe the issue was addressed in the previous comments. If there is still an issue that needs to be fixed, feel free to reopen.

closed 2016-03-21 by nebrius