nebrius/raspi-io #47: Trying to get Ultimate GPS HAT working

stursby, 2016-02-16:
So I have an [Ultimate GPS HAT](https://www.adafruit.com/products/2324) hooked up to a Raspberry Pi (v2 Model B). I'm able to get raw sensor data following [Adafruit's guide](https://learn.adafruit.com/downloads/pdf/adafruit-ultimate-gps-hat-for-raspberry-pi.pdf), however, I can't seem to figure out how to hook it up with Johnny-Five.

According to the Adafruit guide "The Ultimate GPS uses the following GPIO pins: TXD RXD #4 and EEDATA/EECLK".

So I'm wondering how I can hook this all up??

Here's my sample app for reference:

````js
var five = require('johnny-five'); // dtex/johnny-five#gps
var Raspi = require('raspi-io');
var board = new five.Board({
  io: new Raspi()
});

board.on('ready', function() {

  var gps = new five.GPS(['4', '4']);
  gps.on('change', function(data) {
    console.log(data);
  });
  
});
````

Any help would be much appreciated! Thanks.

rwaldron, 2016-02-17:
Raspi-io needs to be updated to support the `serial*` API extensions. Since they're still relatively new, there hasn't been any urgency to implement wide support... I guess that's changed now ;)

nebrius, 2016-02-17:
I'm hoping to have a good chunk of time to work on raspi-io later this week, I'll see if I can work some of this in too.

stursby, 2016-02-17:
Ok, thanks for the quick responses @rwaldron & @nebrius !!

dtex, 2016-02-17:
@stursby Are you using the GPS branch from my repo? You adventurous soul! You'll of course need those serial methods in rasp-io but you should also change your initialization to:
````
var gps = new five.GPS({
  pins: ['4', '4'],
  breakout: "adafruitUltimateGPS"
});
````
I now have a much more basic GPS module that will be the default and whose code will work across a wider range of devices.

stursby, 2016-02-17:
@dtex, yes, I'm using [your branch](https://github.com/dtex/johnny-five/tree/gps). I'm still getting the `TypeError: Cannot read property 'SW_SERIAL0' of undefined` error. I think the biggest issue is I'm not sure what the pin initializers are.

dtex, 2016-02-17:
My comment wasn't meant to supersede Rick's. rasp-io does not yet support the serial methods that were recently added to firmata. That's your biggest obstacle right now. Basically every method on [this page](https://github.com/rwaldron/io-plugins) that begins with "serial" along with some other definitions like the one that's causing this error to be thrown.

rwaldron, 2016-02-17:
@dtex as a side note: `adafruitUltimateGPS` needs to be changed to "const case" ;) (I will mention this on the PR for that)

nebrius, 2016-03-09:
So I just published [raspi-serial](https://github.com/nebrius/raspi-serial), which is most of the work needed to get the `serial*` methods implemented!

stursby, 2016-03-09:
@nebrius Awesome! I'll have to check it out, thanks for the heads up :+1:

nebrius, 2016-03-21:
Serial support is in with raspi-io 6.0.1! Let me know if it works for you!

nebrius, 2016-03-27:
I'm going to go ahead and close this issue, since it should be resolved. Feel free to reopen it if you run into anymore issues, and thanks again for the bug report!

closed 2016-03-27 by nebrius