nebrius/raspi-io #49: RaspberryPi 3 support?

LinusU, 2016-03-04:
The new RaspberryPi 3 is out \o/

But is it supported? We should probably check that it is, and then either fix it or just update the documentation.

nebrius, 2016-03-04:
It isn't supported yet, because we need to do two things first:

1) Verify that the pinout is the same as the A+/B+/2/etc (it probably is), and update https://github.com/nebrius/raspi-board/blob/master/index.js#L728 as necessary
2) Get the board revision(s) as returned from `cat /proc/cpuinfo` and add it to https://github.com/nebrius/raspi-board/blob/master/index.js#L38

Do you have a Pi 3 yet? If so, would you mind running `cat /proc/cpuinfo` and pasting the output here? Mines still back-ordered :-/

nebrius, 2016-03-04:
Also, thanks for filing this issue! It's a good reminder that we need to do the above steps :)

LinusU, 2016-03-04:
The requested output. I'm playing with it now and would love to get it working, I'll post updates here :)

```text
$ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2709
Revision	: a02082
Serial		: 000000009970f34b
```

LinusU, 2016-03-04:
Currently having a problem with finding `wiringPi.h`

```text
$ npm i --save raspi-io
npm WARN deprecated graceful-fs@1.2.3: graceful-fs version 3 and before will fail on newer node releases. Please update to graceful-fs@^4.0.0 as soon as possible.

> i2c-bus@1.0.2 install /home/pi/bathroom-hack/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory '/home/pi/bathroom-hack/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory '/home/pi/bathroom-hack/node_modules/i2c-bus/build'

> raspi-gpio@2.1.0 install /home/pi/bathroom-hack/node_modules/raspi-gpio
> node-gyp rebuild

make: Entering directory '/home/pi/bathroom-hack/node_modules/raspi-gpio/build'
  CXX(target) Release/obj.target/addon/src/addon.o
  CXX(target) Release/obj.target/addon/src/init.o
../src/init.cc:27:22: fatal error: wiringPi.h: No such file or directory
 #include <wiringPi.h>
                      ^
compilation terminated.
addon.target.mk:91: recipe for target 'Release/obj.target/addon/src/init.o' failed
make: *** [Release/obj.target/addon/src/init.o] Error 1
make: Leaving directory '/home/pi/bathroom-hack/node_modules/raspi-gpio/build'
```

nebrius, 2016-03-04:
Thanks, that output is perfect!

About the build error, it sounds like it's the same as in #41 :(. To get around this for now, you can run `npm install raspi-wiringpi` manually.

LinusU, 2016-03-04:
Seems to be related to #41. Running `npm install raspi-wiringpi` fixed it

LinusU, 2016-03-04:
Next problem: `Error: Cannot find module '../build/Release/addon'`
Solution:

```sh
cd node_modules/raspi
npm install
```

Could probably be a `node-gyp rebuild` instead of `npm install`...

LinusU, 2016-03-04:
Next problem: `Error: Unknown board revision a02082`
Solution: *pending*

nebrius, 2016-03-04:
Hmm...haven't seen that cannot find module error before. The base npm install should have handled it?

LinusU, 2016-03-04:
Yes it really should have... No idea why it didn't...

nebrius, 2016-03-05:
ok, everything is published and the Raspberry Pi 3 should now be supported! Let me know if it works

fivdi, 2016-03-05:
[blinky](http://johnny-five.io/examples/raspi-io/) worked fine on a Pi 3 with Raspi-io v5.4.0.

fivdi, 2016-03-05:
After taking a closer look there appear to be some inconsistencies. [blinky](http://johnny-five.io/examples/raspi-io/) does work fine on a Pi 3 with Raspi-io v5.4.0 and Raspbian Jessie 2016-02-26. However, Raspbian Jessie 2016-02-26 comes with WiringPi v2.31 preinstalled but v2.31 doesn't appear to support the Pi 3.
```
pi@raspberrypi:~ $ gpio -v
gpio version: 2.31
Copyright (c) 2012-2015 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Unknown08, Revision: 02, Memory: 1024MB, Maker: Sony 
  Device tree is enabled.
  This Raspberry Pi supports user-level GPIO access.
    -> See the man-page for more details

pi@raspberrypi:~ $ gpio readall
Oops - unable to determine board type... model: 8
```
Because Raspbian Jessie 2016-02-26 comes with WiringPi v2.31 preinstalled, [raspi-wiringpi](https://github.com/nebrius/raspi-wiringpi/blob/master/install-wiringpi.sh) will not install WiringPi. On the other hand, when raspi-wiringpi does decide to install WiringPi, it will install the cloned version at https://github.com/nebrius/wiringPi which is currently v2.25.

In addition, with [this commit](https://git.drogon.net/?p=wiringPi;a=commit;h=b0a60c3302973ca1878d149d61f2f612c8f27fac) on 2016-02-29 Pi 3 support appears to have been added to WiringPi v2.32. There is no mention of v2.32 at http://wiringpi.com/.

nebrius, 2016-03-05:
raspi-wiring pi has a lot of issues right now, sadly. See #41, which I think dovetails into these issues.

I'm not exactly sure what the best course of action is yet, but I hope to spend some time on it this week. Part of me is thinking about scrapping raspi-wiringpi entirely and relying on apt-get for it, but that of course doesn't help with the versioning issue you mentioned @fivdi, so I don't know. It may be better to just fix the bugs in the script and continue that way, but installation reliability will probably always be weaker than using apt-get.

So raspi-wiringpi was installing 2.25 because I have it install from a mirror of wiringpi I put on GitHub. I did this because there have been reliability issues with the main wiringpi repo, which is hosted on drogon's personal server and has suffered from more downtime compared to GitHub. Sadly I've forgotten to update it in a while, but I updated it just now so it will install 2.32.

ajfisher, 2016-03-07:
Just an FYI - my batch of RPi3s has just arrived (mine are made by Farnell in PRC if that is of any use) and they have a different revision.

Will send a micro patch for it.

girliemac, 2016-06-14:
Hi Bryan, forget what I tweeted, sorry, I must be half-sleeping. I still do have the installation problem. 😭 
I decided to fresh-install everything after messing stuff up, and I did!

I have:
- Raspberry Pi 3
- Raspbian Jesse 2016-05-27
- node v4.5.5
- npm v3.9.5
- gcc v4.9.2

and here's the entire output:

```bash
npm WARN deprecated graceful-fs@1.2.3: graceful-fs v3.0.0 and before will fail on node releases >= v7.0. Please update to graceful-fs@^4.0.0 as soon as possible. Use 'npm ls graceful-fs' to find it in the tree.

> i2c-bus@0.11.3 install /home/pi/KittyCam/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory '/home/pi/KittyCam/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory '/home/pi/KittyCam/node_modules/i2c-bus/build'

> raspi-gpio@1.4.0 install /home/pi/KittyCam/node_modules/raspi-gpio
> node-gyp rebuild

make: Entering directory '/home/pi/KittyCam/node_modules/raspi-gpio/build'
  CXX(target) Release/obj.target/addon/src/addon.o
In file included from ../../nan/nan_new.h:190:0,
                 from ../../nan/nan.h:74,
                 from ../src/addon.cc:26:
../../nan/nan_implementation_12_inl.h: In static member function ‘static NanIntern::FactoryBase<v8::Signature>::return_t NanIntern::Factory<v8::Signature>::New(NanIntern::Factory<v8::Signature>::FTH, int, NanIntern::Factory<v8::Signature>::FTH*)’:
../../nan/nan_implementation_12_inl.h:181:76: error: no matching function for call to ‘v8::Signature::New(v8::Isolate*, NanIntern::Factory<v8::Signature>::FTH&, int&, NanIntern::Factory<v8::Signature>::FTH*&)’
   return v8::Signature::New(v8::Isolate::GetCurrent(), receiver, argc, argv);
                                                                            ^
../../nan/nan_implementation_12_inl.h:181:76: note: candidate is:
In file included from /home/pi/.node-gyp/4.4.5/include/node/node.h:42:0,
                 from ../src/addon.cc:25:
/home/pi/.node-gyp/4.4.5/include/node/v8.h:4675:27: note: static v8::Local<v8::Signature> v8::Signature::New(v8::Isolate*, v8::Local<v8::FunctionTemplate>)
   static Local<Signature> New(
                           ^
/home/pi/.node-gyp/4.4.5/include/node/v8.h:4675:27: note:   candidate expects 2 arguments, 4 provided
In file included from ../src/addon.cc:26:0:
../../nan/nan.h: At global scope:
../../nan/nan.h:165:25: error: redefinition of ‘template<class T> v8::Local<T> _NanEnsureLocal(v8::Local<T>)’
 NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                         ^
../../nan/nan.h:160:25: note: ‘template<class T> v8::Local<T> _NanEnsureLocal(v8::Handle<T>)’ previously declared here
 NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                         ^
../../nan/nan.h:557:13: error: ‘node::smalloc’ has not been declared
     , node::smalloc::FreeCallback callback
             ^
../../nan/nan.h:557:35: error: expected ‘,’ or ‘...’ before ‘callback’
     , node::smalloc::FreeCallback callback
                                   ^
../../nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(char*, size_t, int)’:
../../nan/nan.h:561:50: error: ‘callback’ was not declared in this scope
         v8::Isolate::GetCurrent(), data, length, callback, hint);
                                                  ^
../../nan/nan.h:561:60: error: ‘hint’ was not declared in this scope
         v8::Isolate::GetCurrent(), data, length, callback, hint);
                                                            ^
../../nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(const char*, uint32_t)’:
../../nan/nan.h:568:67: error: call of overloaded ‘New(v8::Isolate*, const char*&, uint32_t&)’ is ambiguous
     return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
                                                                   ^
../../nan/nan.h:568:67: note: candidates are:
In file included from ../../nan/nan.h:25:0,
[... 41 lines cut ...]
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "raspi-io"
npm ERR! node v4.4.5
npm ERR! npm  v3.9.5
npm ERR! code ELIFECYCLE

npm ERR! raspi-gpio@1.4.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the raspi-gpio@1.4.0 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the raspi-gpio package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs raspi-gpio
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls raspi-gpio
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/KittyCam/npm-debug.log
```

Thank you for looking! 💫

fivdi, 2016-06-16:
@girliemac the compile errors are occurring because and attempt is being made to install raspi-gpio@1.4.0 with Node.js v4.x. raspi-io depends on raspi-gpio, but raspi-gpio@1.4.0 is quite old. It was released in July 2015 and doesn't support Node.js v4.x, which was released in September 2015.

I'm guessing here but think you want to install RPi-KittyCam. Is this correct?

If so, RPi-KittyCam depends on [`"raspi-io": "^3.3.4"`](https://github.com/girliemac/RPi-KittyCam/blob/master/package.json#L12) which was also released in July 2015 and depends on raspi-gpio@1.4.0.

If RPi-KittyCam is updated to depend on raspi-io@6.0.1 rather than raspi-io@3.3.4, the installation should work.

nebrius, 2016-06-16:
Yep, that'll do it. Current version of raspi-io is 6.0.1. The breaking changes across the various versions since 3.x are pretty small, and are documented in https://github.com/nebrius/raspi-io/blob/master/CHANGELOG.md

girliemac, 2016-06-16:
I am trying to re-create the RPi-KittyCam I wrote last year, but with newer hardware, OS, and software (node and everything upgraded!). It used be on Pi 2 with Node 0.12, but I am upgrading it with Pi 3 and Node 4. So, I am fresh-installing the newest raspi-io and all its dependencies.

I could independently install the latest raspi-gpio fine, but `npm install raspi-io` still fails 😢 

Actually, I did try installing the same exact Raspbian, node, and npm versions, on Pi 2, raspi-io is installed just fine. It only failed for Pi 3.

fivdi, 2016-06-17:
@girliemac running `npm install raspi-io` doesn't necessarily install the latest version of raspi-io (which is currently raspi-io@6.0.1.) For example, if `npm install raspi-io` is called in a directory containing a `package.json` file, then npm will look at `package.json` to see if there is a dependency called raspi-io. If there is such a dependency, npm will attempt to install the version specified in that dependency. Is this the case on you're Pi 3?

nebrius, 2016-06-17:
can you paste the output of `npm outdated` here?

girliemac, 2016-06-19:
@fivdi Ahhhhh, you're so right. 
For some reasons, I thought I was freshly installing everything brand-new, but no, I totally forgot that I kept an old package.json!

Thank you so much for your time, @fivdi and @nebrius,
now raspi-io is installed just fine 😸 
I knew this would be caused by something stupid I did.

closed 2016-07-12 by nebrius