nebrius/raspi-io #50: Trying to access ACT LED on raspi 2B

philraj, 2016-03-09:
Hi,

I'm trying to get the ACT LED on my Raspberry Pi 2B to blink. Everything I've read mentions the ACT LED being controlled via GPIO pin 47, and I've set the ACT LED to trigger on gpio using the following command:

`echo gpio | sudo tee /sys/class/leds/led0/trigger`

Then if I echo 1/0 to its brightness it turns on and off as expected.
Now I'm trying to use the following code to start the blinking:
```
var five = require("johnny-five");
var Raspi = require("raspi-io");
var board = new five.Board({
  io: new Raspi()
});

board.on("ready", function() {
  var led = new five.Led("GPIO47");
  led.strobe();
});
```

But raspi throws an error:
```
>> /home/pi/robo-interface/node_modules/raspi-io/lib/index.js:297
        throw new Error('Unknown pin "' + pin + '"');
              ^
Error: Unknown pin "null"
    at Raspi.value (/home/pi/robo-interface/node_modules/raspi-io/lib/index.js:297:15)
    at Raspi.value (/home/pi/robo-interface/node_modules/raspi-io/lib/index.js:315:45)
    at Raspi.pinMode (/home/pi/robo-interface/node_modules/raspi-io/lib/index.js:304:21)
    at Led.Controllers.DEFAULT.initialize.value (/home/pi/robo-interface/node_modules/johnny-five/lib/led/led.js:71:17)
    at new Led (/home/pi/robo-interface/node_modules/johnny-five/lib/led/led.js:185:10)
    at Board.<anonymous> (/home/pi/robo-interface/robo.js:10:13)
    at Board.emit (events.js:129:20)
    at process._tickDomainCallback (node.js:381:11)
```

Is it just impossible to access this pin using raspi? Please let me know if I'm doing something wrong.

nebrius, 2016-03-09:
So bad news, good news, bad news.

Bad news: raspi-io only allows access to pins that are connected to headers. This is by design. You can see the full listing of pins that are supported at https://github.com/nebrius/raspi-io/wiki/Pin-Information at. Now why it says "null" instead of "GPIO47" in the error I don't know, that needs to be fixed.

Good news: the ACT LED is set up as the default LED, so you can get access to it by doing:

```
var led = new five.Led();
```

Bad news: that seems to be broken last time I tried it: https://github.com/nebrius/raspi-io/issues/37

philraj, 2016-03-09:
What a rollercoaster of emotions... Oh well, thanks anyway. It was just for quick testing purposes until my motor control board ships. I'm mostly just glad that it's a bug and not something I messed up. Did you ever find any leads on fixing it? I'm guessing it's not a huge priority.

nebrius, 2016-03-09:
I haven't started investigating it yet, so nothing...it's probably something simple.

Can we keep this issue open to track the "null" bug? That one's new to me.

philraj, 2016-03-10:
Sure, why not.

nebrius, 2016-03-21:
Both issues, the LED and the "null" thing, are fixed with raspi-io 6.0.1. Let me know if it works for you!

nebrius, 2016-03-27:
I'm going to go ahead and close this issue, since it should be resolved. Feel free to reopen it if you run into anymore issues, and thanks again for the bug report!

closed 2016-03-27 by nebrius

philraj, 2016-03-27:
Oh right, I was going to say I'll test it when I can. Don't have access to a raspberry pi at the moment sadly. I'm sure it's all good though. Thanks again.