nebrius/raspi-io #51: Interfacing with Adafruit Motor HAT control board?

philraj, 2016-03-11:
Hello again,

So my motor control board came in today [(Adafruit Motor HAT)](https://www.adafruit.com/products/2348) and much to my dismay, the recommended way of interfacing with it is through their Python library. While it's actually not so bad once you get everything set up, I was hoping there might be a nice way to use johnny-five and raspi-io to access the motor controls. Something about calling Python commands from node kinda rubs me the wrong way.

Anyway, if you have any experience with this board or anything similar to it, and can maybe offer some hints as to how to control it by using raspi-io functionality, please let me know. I'll keep doing research but I figured I might as well ask since you're the Raspberry Pi Javascript guy.

Thanks!

nebrius, 2016-03-12:
Hi @philraj! This is something that Johnny-Five should be capable of doing in general. If you take a look at the [motor docs](http://johnny-five.io/api/motor/), you can see that this API does support motor control boards. Johnny-Five comes with a few pre-configured sets of options for common boards, but you can also create a custom one.

It looks like your motor controller uses the PCA9685 I2C interface chip, which is good because Johnny-Five has support for it. You might even be able to use the Adafruit Arduino shield config at http://johnny-five.io/api/motor/#i2c. I'd play around and see what happens.

Let me know if you figure it out!

rwaldron, 2016-03-12:
Presumably this will have different pins mapped on the shield, so that motor config probably won't work. Never fear—we can just figure out the mapping from this: https://github.com/adafruit/Adafruit-Motor-HAT-Python-Library/blob/master/Adafruit_MotorHAT/Adafruit_MotorHAT.py#L168-L194

philraj, 2016-03-12:
Wow, thank you so much guys! Will try this tomorrow.

rwaldron, 2016-03-12:
@philraj @nebrius I just ordered one of these to help test :)

philraj, 2016-03-13:
Hoho, success!
```
var m1 = new five.Motor({
  pins: {
    pwm: 8,
    dir: 9,
    cdir: 10
  },
  address: 0x60,
  controller: 'PCA9685'
});
```
This is all you need to address each motor (just substitute the pin numbers with the relevant numbers from the motor hat library that @rwaldron linked). I had issues at first because I was just passing the pins alone to the motor constructor, but then I was reminded that this is all going through I2C and not GPIO.

By the way, before I was including the address and controller properties, it was giving me the error `Error: Unknown pin` with one of the pin numbers. Is there a reason it says that instead of just silently doing nothing?

philraj, 2016-03-13:
Actually looking over the pin numbers again, the ADAFRUIT_V2 config has exactly the values needed for controlling the motor hat.

rwaldron, 2016-03-14:
Whoa, that's awesome—great work! (We should add this shield to the motor docs)

nebrius, 2016-03-14:
> By the way, before I was including the address and controller properties, it was giving me the error Error: Unknown pin with one of the pin numbers. Is there a reason it says that instead of just silently doing nothing?

I'm a fan of fail early, fail often, that's why. When someone sees this error, that means there's a bug in their code. I think developers would rather know that the bug exists than not know :)

philraj, 2016-03-14:
@nebrius I feel the same way, I just meant I don't understand how it knows I'm accessing invalid pins (I thought it would behave like when you tell it to strobe an LED that isn't connected). Is it something to do with I2C?

nebrius, 2016-03-14:
> I feel the same way, I just meant I don't understand how it knows I'm accessing invalid pins (I thought it would behave like when you tell it to strobe an LED that isn't connected). Is it something to do with I2C?

An invalid pin is an identifier that doesn't map to any known peripheral. So, for example, if you did `new five.Led('GPIO50')`, you'd get this error. It's not because nothing is connected to GPIO50, it's because there is no GPIO50.

philraj, 2016-03-14:
Right so I guess my confusion was that I figured if you don't specify a controller and an address, it would just assume you were referencing gpio pins (I was originally using valid gpio references).

rwaldron, 2016-03-14:
@philraj 

> Right so I guess my confusion was that I figured if you don't specify a controller and an address, it would just assume you were referencing gpio pins (I was originally using valid gpio references).

Yes, you are exactly right!

Let me dive into this a bit for you...

Without specifying the `controller` \*, the Raspberry Pi doesn't know that you mean `8`, `9` & `10` on that shield. Think of the `PCA9685` as a kind of "mini board"—take a look at this: 

[image]

(That's from this example: http://johnny-five.io/examples/expander-PCA9685/)

The PCA9685 chip shown here is what runs on the Adafruit Motor Hat that you're using. PCA9685 is a commonly used "io expander" chip that provides 16 PWM pins (which is really awesome).

\* You probably don't need the `address`, unless you've knowingly changed the address by soldering any of the address jumpers together.

philraj, 2016-03-14:
Ah ok. Thanks for the explanation!

closed 2016-03-17 by philraj