nebrius/raspi-io #52: [Rpi B+] Pin "GPIO13" does not support mode "3" (PWM)

Worie, 2016-03-13:
Hi, recently i have bought a Pololu Dual MC33926 Motor Driver for Raspberry Pi. (https://www.pololu.com/product/2755)

The problem is that when i want to set the speed of second motor (in the link provided above there are instructions that I am supposed to use GPIO13 to set its speed )all i get is 
`Pin "GPIO13" does not support mode "3" 
`
The error occurs whenever I try to set it to PWM mode.
`board.pinMode("GPIO13", five.Pin.PWM);
`
However, there was a python library provided for the driver i'm using, so I had to try and check if it's working properly. And it is - https://github.com/pololu/dual-mc33926-motor-driver-rpi . When i run the example script all motors are working properly. They are setting the GPIO13 into PWM with 

`wiringpi2.pinMode(13, wiringpi2.GPIO.PWM_OUTPUT)
`
So I wonder what is the problem and how to solve it. I prefer using JS over Python and I don't know the mechanics behind the raspi-io or the python library mentioned above, so i thought ill post it as an issue here.

Thanks in advance!

My config:
Raspberry Pi B+ rev 02
2016-02-26-raspbian-jessie
node v0.10.29 
gpio version: 2.31

joshmmo, 2016-03-13:
I was just about to create an issue on this. I am experiencing the exact same problem. I'm using https://www.pololu.com/product/2753

According to the wiki, (https://github.com/nebrius/raspi-io/wiki/Pin-Information) GPIO19 (Wiring pin 24) is PWM1, however looking at other documentation, http://pinout.xyz/ PWM1 is on GPIO13.

So I believe the wiki is wrong. And attempting use PWM on GPIO13 (Wiring pin 23) gives this error:

> Error: Pin "23" does not support mode "3"
>     at Raspi.value (/home/pi/piRobot/node_modules/raspi-io/lib/index.js:322:15)
>     at Raspi.pinMode (/home/pi/piRobot/node_modules/raspi-io/lib/index.js:304:21)
>     at Motor.initialize (/home/pi/piRobot/node_modules/johnny-five/lib/motor.js:599:11)
>     at new Motor (/home/pi/piRobot/node_modules/johnny-five/lib/motor.js:589:10)
>     at Board.<anonymous> (/home/pi/piRobot/app.js:16:18)
>     at emitNone (events.js:72:20)
>     at Board.emit (events.js:166:7)
>     at doNTCallback0 (node.js:417:9)
>     at process._tickDomainCallback (node.js:387:13)

joshmmo, 2016-03-14:
Looks like the issue is in raspi-board repo?

https://github.com/nebrius/raspi-board/blob/master/lib/index.js#L321

I believe PWM1 should be set on 23...not 24.

nebrius, 2016-03-14:
It turns out PWM1 works like PWM0, in that it's exposed on _two_ pins, not one. PWM1 should be set to 23 _and_ 24, just like PWM0 is set on 1 _and_ 26. This means there is good news and bad news.

Good news is, you can switch your code to use 24 instead of 23 and it'll work just fine.

Bad news, the fix will actually be a breaking change. In the case of multi-pin peripherals, it defaults to the lower number pin, meaning that adding this fix will cause `PWM1` to resolve to 23 instead of 24. This could break people's code that use `PWM1` to indicate the port, so we have to do a major version bump.

There's some other work in progress, due out soon, that also has some breaking changes, and tl;dr, we need to wait until the other stuff is ready to go, so this fix will take a while (week at minimum, probably a few weeks).

joshmmo, 2016-03-14:
The motor driver shield I linked above does not cover pin 24 unfortunately. It leaves 6 pins alone. I will just make the local change myself for now. Hopefully it's as simple as editing the lines I linked to raspi-board.

Worie, 2016-03-14:
Thanks guys, I understand the issue now :) 

I've been attempting to change the settings of pins in raspi-board too, joshmmo - but unfortunetly it did'nt work for me - if you manage to get it to work, could you please share the changes you have made?

joshmmo, 2016-03-14:
Yeah I got it working. I am on mobile so this will probably look bad but add own to the following code:

23: {
    pins: ['GPIO13', 'pwm', 'P1-33'],
    peripherals: ['gpio', 'pwm']
  },

joshmmo, 2016-03-14:
I don't see an edit button on mobile but that edit is to this file: 

https://github.com/nebrius/raspi-board/blob/master/lib/index.js#L321

nebrius, 2016-03-14:
Your close, it should look like this:

```
23: {
  pins: ['GPIO13', 'PWM1', 'P1-33'],
  peripherals: ['gpio', 'pwm']
},
```

Worie, 2016-03-14:
Oh god - it turned out I wasnt editing the the file inside the lib directory.. I guess it was too late to think for me then :) Anyway - thanks! The solution is great (for now i guess)

nebrius, 2016-03-21:
Should be fixed with raspi-io 6.0.1. Let me know if it works for you!

nebrius, 2016-03-27:
I'm going to go ahead and close this issue, since it should be resolved. Feel free to reopen it if you run into anymore issues, and thanks again for the bug report!

closed 2016-03-27 by nebrius