nebrius/raspi-io #53: Error: Cannot find module '../build/Release/addon'

philraj, 2016-03-14:
So I saw another issue with this same problem, but it didn't seem like it was resolved. I just replaced Raspbian with Arch ARM, and although everything else seems to be working great, I get this error when I try to run the script that instantiates the board:
```
module.js:341
    throw err;
    ^

Error: Cannot find module '../build/Release/addon'
    at Function.Module._resolveFilename (module.js:339:15)
    at Function.Module._load (module.js:290:25)
    at Module.require (module.js:367:17)
    at require (internal/module.js:16:19)
    at Object.<anonymous> (/home/pi/workspace/robo-interface/node_modules/raspi/lib/index.js:30:29)
    at Module._compile (module.js:413:34)
    at Object.Module._extensions..js (module.js:422:10)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Module.require (module.js:367:17)
```
Do you have any clue as to what might be going wrong here? Unfortunately Raspbian was giving me massive problems with streaming video, so I was really hoping there could be a solution to this error.

nebrius, 2016-03-14:
I'm not sure in this specific case what's causing that, but Arch is pretty unsupported in general. See https://github.com/nebrius/raspi-io/issues/24

philraj, 2016-03-14:
Hmm alright. There is no /build folder in /node_modules/raspi, so I guess something failed during the installation? Oh well.

nebrius, 2016-03-14:
> I guess something failed during the installation?

That would be my guess too. Sorry about that.

If you feel like digging and hacking, maybe post the install log here, and submit a PR with Arch support?

philraj, 2016-03-15:
Sure, here's the log:
```
npm install raspi-io
npm WARN deprecated graceful-fs@1.2.3: graceful-fs version 3 and before will fail on newer node releases. Please update to graceful-fs@^4.0.0 as soon as possible.

> i2c-bus@1.0.2 install /home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory '/home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory '/home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/i2c-bus/build'

> raspi-gpio@2.2.1 install /home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/raspi-gpio
> node-gyp rebuild

make: Entering directory '/home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/raspi-gpio/build'
  CXX(target) Release/obj.target/addon/src/addon.o
  CXX(target) Release/obj.target/addon/src/init.o
  CXX(target) Release/obj.target/addon/src/write.o
  CXX(target) Release/obj.target/addon/src/read.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory '/home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/raspi-gpio/build'

> raspi-pwm@2.3.1 install /home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/raspi-pwm
> node-gyp rebuild

make: Entering directory '/home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/raspi-pwm/build'
  CXX(target) Release/obj.target/addon/src/addon.o
  CXX(target) Release/obj.target/addon/src/init.o
  CXX(target) Release/obj.target/addon/src/write.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory '/home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/raspi-pwm/build'

> raspi-wiringpi@1.0.6 install /home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/raspi-wiringpi
> sh install-wiringpi.sh

Wiring Pi is already installed, skipping\n

> raspi@2.0.0 install /home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/raspi
> node-gyp rebuild

make: Entering directory '/home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/raspi/build'
  CXX(target) Release/obj.target/addon/src/addon.o
  CXX(target) Release/obj.target/addon/src/init.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory '/home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/raspi/build'

> execSync@1.0.2 install /home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/execSync
> node install.js

[execsync v1.0.2] Attempting to compile native extensions.
[execSync v1.0.2]
    Native code compile failed!!

> raspi-i2c@2.3.2 postinstall /home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/raspi-i2c
> echo "Enabling I2C at boot time, you may be asked for your password" && sudo env "PATH=$PATH" script/enable_i2c.js

Enabling I2C at boot time, you may be asked for your password
Checking if I2C is enabled at boot time
I2C is already enabled at boot time
raspi-gpio@2.2.1 /home/pi/workspace/robo-interface/node_modules/raspi-gpio
`-- raspi-io@5.4.1  extraneous
```
So it fails to compile the native code, but isn't very descriptive about it. Is there any way I could do this step manually myself?

nebrius, 2016-03-15:
So the thing that failed to compile was execSync, which is expected on Node 0.12 or newer. It doesn't look like there were any other compile issues, so I dunno. Maybe try and see which module it can't load the native extension for and see what the directory structure is like in that folder in node_modules?

philraj, 2016-03-15:
Where would I start looking to see which module it can't load? Also are you saying if I downgrade to Node 0.11 it might compile?

nebrius, 2016-03-15:
> Where would I start looking to see which module it can't load?

The stack trace shows it here:

```
at Object.<anonymous> (/home/pi/workspace/robo-interface/node_modules/raspi/lib/index.js:30:29)
```

So check in `node_module/raspi/build`

> Also are you saying if I downgrade to Node 0.11 it might compile?

`execSync` is an API built-in to Node, but wasn't added until 0.12. The error you see is the shim for execSync in Node 0.10 failing to compile because you're on Node > 0.10, which is normal and expected, so you can ignore it.

Basically the execSync thing is a red herring and unrelated to the runtime error you're seeing.

philraj, 2016-03-15:
Oh I see. 

About checking `node_modules/raspi/build`, I don't think there's anything to check (there is no such folder!). So I guess this is going to be very hard to investigate...

nebrius, 2016-03-15:
Oh woa!!

I looked closer at what modules are being built where. For some reason, all of these modules are being built as dependencies of `raspi-gpio`, but, just, no.

You can tell by looking at lines that look like, for example

```
> raspi-pwm@2.3.1 install /home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/raspi-pwm
> node-gyp rebuild
```

This says it's building v2.3.1 of raspi-pwm at `/home/pi/workspace/robo-interface/node_modules/raspi-gpio/node_modules/raspi-pwm`. I have _no idea_ why it's being installed there. These modules aren't dependencies of raspi-gpio.

Upgrading to NPM v3 will fix this for you (`npm install -g npm`), but that is _so bizarre_ 😖

nebrius, 2016-03-15:
Also, just out of curiosity, what version of Node and NPM are you currently using?

philraj, 2016-03-15:
Very strange indeed, nice catch. These are the version:
```
$ npm --version
3.8.1
$ node --version
v5.8.0
```
So I guess upgrading won't fix anything? Is it possible _down_grading will fix that? I know Arch keeps things extremely up to date, perhaps _too_ up to date?

nebrius, 2016-03-15:
At this point, your guess is as good as mine. I have no idea what NPM is doing here.

philraj, 2016-03-15:
Here's a question: is it normal that there's a /node_modules/ folder inside the node_modules/raspi/ path?
(`project/node_modules/raspi/node_modules/`)

nebrius, 2016-03-15:
In NPM < 3 yes, but >=3 no. Maybe `rm -rf node_modules` and start from scratch?

philraj, 2016-03-15:
Ok so good news / bad news. I removed node_modules and ran npm install, and now it has no problem loading raspi-io (or at least it gives no errors). The REPL loads up and everything seems fine. Only thing is, now the motor.fwd() command is doing nothing. Am I forgetting some step in the installation? It's been a long day.

nebrius, 2016-03-15:
Are you sure you selected the pin you thought you selected, given that raspi-io's [pin naming scheme](https://github.com/nebrius/raspi-io/wiki/Pin-Information) is kinda complicated?

philraj, 2016-03-16:
Almost positive, since I went back to the same working setup I was using when I was on Raspbian. It's just pulling the motor configs from the premade `SHIELD_CONFIGS.ADAFRUIT_V2`. Maybe if I go back to specifying the pins explicitly... in fact let's see if blinking an LED works...

philraj, 2016-03-16:
Alright so the LED blinks! This is great. Time to figure out why the motors aren't budging...

philraj, 2016-03-16:
Hey it's working! Turns out I made a bunch of silly mistakes, including not instantiating the motors within `board.on('ready')`. Don't know why that worked the first time around (I guess some kind of unreliable timing differences).

So raspi-io officially _can_ work on Arch Linux. Don't know what all those npm problems were about but wiping the node_modules folder and starting over definitely fixed it.

nebrius, 2016-03-17:
That's good to know, and I'm glad you got it working! I would have thought some of the post-install scripts that use `apt-get` would have borked the install for sure :)

philraj, 2016-03-17:
Yep everything's working great now. Got turn signals on my robot and everything. Thanks for all the help, you really got me out of a bind! This is part of a project which I'm presenting on Friday (web-controlled nodebot with variable-speed joystick control and a socket queue which allows each connected client to control it for 30 seconds before being moved to the back of the line). Looks like the presentation will go off without a hitch now.

nebrius, 2016-03-17:
Woo 🎉 Best of luck with your presentation!

closed 2016-03-20 by philraj

Sleepingbug, 2019-12-06:
i have a similar problem, but not the same:

root@78eaa057247b:/home/howard/licode-all/licode-master# exit2019-12-06 06:06:03.922  - INFO: ErizoAgent - message: launched new ErizoJS, erizoId: 53bb30d8-89d4-036d-930c-f539672953c8
[erizo-53bb30d8-89d4-036d-930c-f539672953c8] internal/modules/cjs/loader.js:797
    throw err;
    ^

Error: Cannot find module './../../erizoAPI/build/Release/addon'
Require stack:
- /home/howard/licode-all/licode-master/erizo_controller/erizoJS/erizoJS.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:794:15)
    at Function.Module._load (internal/modules/cjs/loader.js:687:27)
    at Module.require (internal/modules/cjs/loader.js:849:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/home/howard/licode-all/licode-master/erizo_controller/erizoJS/erizoJS.js:90:11)
    at Module._compile (internal/modules/cjs/loader.js:956:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:973:10)
    at Module.load (internal/modules/cjs/loader.js:812:32)
    at Function.Module._load (internal/modules/cjs/loader.js:724:14)
    at Function.Module.runMain (internal/modules/cjs/loader.js:1025:10) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/howard/licode-all/licode-master/erizo_controller/erizoJS/erizoJS.js'
  ]
}

how to fix it ?
ubuntu 16.04 64-bit
nodejs version: v12.12.0
npm version:6.11.3