nebrius/raspi-io #56: Raspi-IO - Symbol is not defined

punitganshani, 2016-04-09:
Faced Symbol is not defined issue  

```
/home/pi/code/matrix/node_modules/raspi-io/node_modules/raspi-i2c/lib/index.js:96
var devices = Symbol('devices');
              ^
ReferenceError: Symbol is not defined
    at Object.<anonymous> (/home/pi/code/matrix/node_modules/raspi-io/node_modules/raspi-i2c/lib/index.js:96:15)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/pi/code/matrix/node_modules/raspi-io/lib/index.js:21:17)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
```

Version installed 

```
raspi-io@6.0.1 node_modules/raspi-io
├── raspi-board@3.0.0
├── raspi-led@1.3.2 (raspi-peripheral@1.5.2)
├── raspi-pwm@3.0.0 (raspi-peripheral@1.5.2, nan@2.2.1)
├── raspi-gpio@2.2.2 (raspi-peripheral@1.5.2, nan@2.2.1)
├── raspi@2.0.0 (raspi-wiringpi@1.0.6, nan@2.2.1)
├── raspi-i2c@3.0.0 (ini-builder@1.0.3, raspi-peripheral@1.5.2, i2c-bus@1.0.2)
└── raspi-serial@1.3.1 (raspi-peripheral@1.5.2, serialport@2.0.6)

```

fivdi, 2016-04-09:
@punitganshani which version of Node.js is being used?

fivdi, 2016-04-09:
@punitganshani and which version of Raspi-io is being used? The current version of Raspi-io requires Node.js v0.12 or higher.

punitganshani, 2016-04-09:
Default ones on RPI3:

Node: v0.10.44
NPM: 2.15.0

fivdi, 2016-04-09:
I'm afraid Raspi-io dropped support for Node.js v0.10.x recently. See the [System Requirments](https://github.com/nebrius/raspi-io#system-requirements) section of the readme. Is it possible for you to upgrade to Node.js v4 or higher?

punitganshani, 2016-04-09:
@fivdi That worked. I missed that

closed 2016-04-09 by punitganshani