nebrius/raspi-io #57: Board never closes

wannabeCitizen, 2016-04-11:
Using this library with johnny-five. However, when I finish and go to shutdown the connection to the raspberry pi I get a message in the console `Board Closing` but this just hangs forever.

I've been just manually going in and running a `kill -9` on the process. This isn't ideal but I couldn't find other ways to more gracefully close the connection. 

Any suggestions?

nebrius, 2016-04-11:
Can you tell me how you're attempting to close the app?

Normally, I use `process.exit()` when I want the app to exit.

wannabeCitizen, 2016-04-11:
When I ran in repl mode, I used process.exit(), but it still just hung.
Now I'm running out of repl mode and I'm just sending it a termination signal (`CTRL-C`).

Any other suggestions for closing it?

nebrius, 2016-04-11:
That's strange, `ctrl-c` works for me, as does `process.exit()`. Can you tell me the following?

- Node version
- Node architecture (e.g. armv6 or armv7)
- npm version
- Raspberry Pi version
- Which OS and version (e.g. Raspbian 2016-03-18)
- Wiring Pi version, from `gpio -v`

wannabeCitizen, 2016-04-11:
Node version 4.2.1
npm 2.14.7
Pi 2 B v1.1
Raspian (should be latest version, I ran sudo apt-get upgrade)
GPIO 2.32

I have a few other boards of the same type lying around so I can test it on another one

wannabeCitizen, 2016-04-11:
One thing I'm realizing is a confounding variable is I'm also listening for events on a HID device (a USB RFID reader). So maybe somehow that's the thing that's still lingering and preventing the process from closing? 

I'll run a few tests. The only reason I suspected the raspio is because it hangs at the "Board Closing" line with no other info.

nebrius, 2016-04-11:
Can you figure out which kernel you have? `uname -r` IIRC. Running `sudo apt-get upgrade` doesn't necessarily upgrade to the latest release, just like how running it won't upgrade Ubuntu 14.04 to 15.10.

I'm not sure that really matters anymore though, I bet it's the HID device. Raspi IO doesn't do any sort of cleanup code, and doesn't make use of the `beforeExit` event. Raspi IO doesn't make a connection to anything, unlike say the Arduino, so there's nothing to close or cleanup.

wannabeCitizen, 2016-04-11:
Ah Ok, so potentially I need to figure out how handle device.close() during shutdown. Sounds like that's my issue. I'll go ahead and close this issue since it's likely not your issue.

You're saying that johnny-five though has a `beforeExit` that I could potentially work with?

closed 2016-04-11 by wannabeCitizen

nebrius, 2016-04-11:
> You're saying that johnny-five though has a beforeExit that I could potentially work with?

It's built in to Node.js itself: https://nodejs.org/dist/latest-v5.x/docs/api/process.html#process_event_beforeexit

Best of luck!