nebrius/raspi-io #58: ENOENT: no such file or directory, open \'/sys/class/leds/led0/trigger\

goliatone, 2016-05-06:
I'm running an application inside docker on a raspberry pi 3, and I get the following error:

```
Error: ENOENT: no such file or directory, open '/sys/class/leds/led0/trigger'
    at Error (native)
    at Object.fs.openSync (fs.js:546:18)
    at Object.fs.writeFileSync (fs.js:1149:15)
    at new LED (/usr/src/app/node_modules/raspi-led/lib/index.js:60:21)
    at Raspi.value (/usr/src/app/node_modules/raspi-io/lib/index.js:325:34)
    at Raspi.pinMode (/usr/src/app/node_modules/raspi-io/lib/index.js:304:21)
    at /usr/src/app/node_modules/raspi-io/lib/index.js:234:17
    at Array.forEach (native)
    at Object.Call (/usr/src/app/node_modules/es6-shim/es6-shim.js:279:14)
    at Array.forEach (/usr/src/app/node_modules/es6-shim/es6-shim.js:1269:17)
```
I believe this is due to the [Hypriot][1] OS image being faulty- still in beta- but I'm wondering if this has been reported before, also curious on how this is being used. 

[1]:http://blog.hypriot.com/downloads/

nebrius, 2016-05-06:
I haven't heard of this exact issue before with LED, but there have been [issues with Hypriot in the past](https://github.com/nebrius/raspi-io/issues/27).

Can you do me a favor and see where the path stops existing? I.e. does `/sys/class/leds` exist, and does `/sys/class/leds/led0` exist?

goliatone, 2016-05-06:
`/sys/class/leds/` is there, but I can't `sudo mdkir` a led0 directory. 

PS: I also ran into the same issue as #27, the solution provided by Snorlock solved those issues.

nebrius, 2016-05-06:
I wouldn't expect you to be able to make the directory, the `/sys/class/*` files are for configuring hardware, and a directory/file missing is equivalent to a missing entry in `/dev`, which means the hardware doesn't exist (as far as the kernel knows). Even if we could create the directory/file, it wouldn't fix the problem.

Is there anything else inside of `/sys/class/leds`?

goliatone, 2016-05-06:
As I was posting about not being able to write to `/sys/class/` I was thinking that it actually made sense. 
There is nothing inside `/sys/class/leds`.

nebrius, 2016-05-06:
Is it possible for you to switch to switch from Hypriot to Raspbian?

goliatone, 2016-05-06:
well, I have an application deployed on 10+ RPi 2 all using Hypriot/docker, I'm currently doing the next deployment with 20+ RPi3, with a lot of the deployment infrastructure relying on docker. I think using Raspbian would be possible but not the best scenario.

nebrius, 2016-05-07:
Hrm...so the fix for this is to rework https://github.com/nebrius/raspi-led/blob/master/index.js so that it detects which OS it's on and use whatever mechanism Hypriot exposes instead of the one Raspian exposes.

Unfortunately I am very swamped at the moment and don't have time to investigate it myself. Is this something you might be interested in looking into?

goliatone, 2016-05-07:
For now I forked both raspi-io and raspi-led, modified raspi-led to check for led0 and do nothing if it's not available. Seems to work, but now I ran into a different issue. SerialPortBinding cant find `dev/ttyAMA0`, which is not present either. 
If I figure out what Hypriot exposes I will update this issue.
Thanks!

nebrius, 2016-05-09:
FYI `/dev/ttyAMA0` is hard-coded in [raspi-serial](https://github.com/nebrius/raspi-serial/blob/master/index.js#L34) and also is based on Raspbian.

derekrushforth, 2016-08-23:
Having the same issue while on a Raspi 3 model B running Raspbian Jessie. Looks like `/sys/class/leds` is empty for me as well.

nerfherder101, 2016-08-26:
me too
and I am on rasberian

nebrius, 2016-08-26:
to clarify @derekrushforth, the directory `/sys/class/leds` does exist, but is empty, correct?

derekrushforth, 2016-08-26:
@nebrius yep, that's correct.

nebrius, 2016-09-20:
I just published a new version of raspi-led that adds a guard around file I/O. Let me know if this fixes your issues.

nebrius, 2016-10-06:
Closing since it should have been fixed. Feel free to reopen if the issue persists

closed 2016-10-06 by nebrius