nebrius/raspi-io #60: node-serialport issue

iam-black, 2016-05-17:
Trying to install Johnny-Five and Raspi IO on RPi following this guide https://github.com/nebrius/raspi-io/wiki/Getting-a-Raspberry-Pi-ready-for-NodeBots . 
After `npm install johnny-five raspi-io` I get  
`node-pre-gyp ERR! Tried to download: https://github.com/voodootikigod/node-seria                          lport/releases/download/3.1.2/serialport-v3.1.2-node-v46-linux-arm.tar.gz

node-pre-gyp ERR! Pre-built binaries not found for serialport@3.1.2 and node@4.4.4 (node-v46 ABI) (falling back to source compile with node-gyp)
`.
Is it OK?

nebrius, 2016-05-17:
Yes that's ok, it'll still work, although I'm a little surprised that's happening. @reconbot, any ideas?

reconbot, 2016-05-17:
There are a lot of arm platforms that fall under `linux-arm` that aren't compatible. So we opted not to do do it.

reconbot, 2016-05-17:
A personal note, I'm not fully aware of all these differences but if start to identify them and figure out if they're exposed to node, I can ensure the prebuilt binaries happen and that the downloaders `prebuilt` and `node-pre-gyp` support them. I'm hesitant to say "raspi gets this download", "tessel 2 gets this other one", but I'm happy to say `linux-armxxx` gets this download and is compiled with these flags disabled, etc.

fivdi, 2016-05-17:
@reconbot Node.js appears to be able to get away with [3 Linux arm builds](https://nodejs.org/dist/latest/), armv6l, arm7l, and arm64. I've used the armv7l builds on both the Raspberry Pi and the BeagleBone Black.

reconbot, 2016-05-17:
Is it exposed to nodejs which version a processes is running?

fivdi, 2016-05-17:
I'm not sure if it's reliably exposed in a documented way. Here's some output from Node.js v6.1.0 on a Raspberry Pi 2 running the armv7l build:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.9-v7+ #884 SMP Fri May 6 17:28:59 BST 2016 armv7l GNU/Linux
pi@raspberrypi:~ $ node -v
v6.1.0
pi@raspberrypi:~ $ node
> process.arch
'arm'
> process.config.variables.node_byteorder
'little'
> process.config.variables.arm_version
'6'
> 
```
I'd imagine that the `l` in `armv7l` is exposed as `process.config.variables.node_byteorder`. What I don't understand is why `process.config.variables.arm_version` is `6`. I would have expected it to be `7`.

asit569, 2016-05-18:
Try to install bcm2835

reconbot, 2016-06-25:
@asit569 why?

nebrius, 2016-08-02:
I'm going to go ahead and close this issue, please reopen it if you're still having problems @iam-black

closed 2016-08-02 by nebrius