nebrius/raspi-io #61: ReferenceError: Symbol is not defined

martinschuberteu, 2016-05-23:
Hi,
when I'm trying to run your led example I'm getting the following error:

pi@berry:~/johnny $ nodejs led.js

/home/pi/johnny/node_modules/raspi-i2c/lib/index.js:96
var devices = Symbol('devices');
              ^
ReferenceError: Symbol is not defined
    at Object.<anonymous> (/home/pi/johnny/node_modules/raspi-i2c/lib/index.js:96:15)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/pi/johnny/node_modules/raspi-io/lib/index.js:21:17)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)

I'm using a Raspberry Pi 3 B (same error on a 1 B) with raspbian 8/Jessie and the following:

- gcc version 4.9.2 (Raspbian 4.9.2-10)
- nodejs v0.10.29
- npm 3.9.2

any idea what the problem could be? Thank you!

closed 2016-05-23 by martinschuberteu

nebrius, 2016-05-23:
For posterity and anyone else who looks at this issue, Node 0.12 is required by Raspi IO, otherwise you get this error.

martinschuberteu, 2016-05-23:
Yes, apologies for opening an issue. I've tried it with nodejs 6 and got an ugly stack trace before opening the issue, but then succeeded with nodejs 4 shortly after.

nebrius, 2016-05-23:
No worries at all. I often open an issue or ask a question, just to find that going through the process of formulating the question led me to the answer. I just wanted to mark the solution in case anything else who found this issue from, say, a Google search wouldn't be left wondering what the solution was.