nebrius/raspi-io #62: Module Extension Error

krupper, 2016-07-04:
Hi,
I had installed the package regular, but when I'm trying to use it with the example Code from the repository:
```
var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi()
});

board.on('ready', function() {

  // Create an Led on pin 7 (GPIO4) on P1 and strobe it on/off
  // Optionally set the speed; defaults to 100ms
  (new five.Led('P1-7')).strobe();

});
```

I get the following error every time:
```
module.js:356
  Module._extensions[extension](this, filename);
                               ^
Error: /home/pi/nestor/node_modules/raspi/build/Release/addon.node: undefined symbol: node_module_register
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/pi/nestor/node_modules/raspi/lib/index.js:30:29)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)

```
I run the script as sudo and node is in version 6.2.2 and gcc in version 4.9.2 on a Raspberry Pi  2 B+ with Raspbian 4.9.2 installed. Do you have an idea why this is happening and how I can fix it?

nebrius, 2016-07-05:
Are you using nvm or any other sort of Node.js version manager or do a major version upgrade of Node after installing? Typically this error happens when raspi-io is installed using one version of Node and run using another. There is C++ code in raspi-io that can only run against the version of V8 it was compiled for (which is upgraded every major Node version).

krupper, 2016-07-08:
Yes, I'm using nvm (v0.31.1) but I installed raspi-io with the same node version while I'm using it right now. What do you think? Should I uninstallt raspi-io and retry the installation?

nebrius, 2016-07-08:
I would first uninstall nvm and then reinstall rasp-io. It's tricky to get Node versions to line up right with nvm

krupper, 2016-07-08:
I unload nvm, remove rasp-io and reinstall it but I get the same error again :-(

nebrius, 2016-07-08:
Yeah I need to figure out why it's doing that...it doesn't hurt your program though (for now), it's just an annoying message.

closed 2016-07-14 by krupper