nebrius/raspi-io #63: Raspberry Pi 3 and Node.js ARM Versions

lyzadanger, 2016-07-07:
Hi! I followed the instructions in [the wiki](https://github.com/nebrius/raspi-io/wiki/Getting-a-Raspberry-Pi-ready-for-NodeBots#installing-nodejs) and installed the ARMv7 Node.js on my Raspberry Pi 3 per:

> If you are on a Raspberry Pi 2 or Raspberry Pi 3, copy the ARMv7 link, otherwise copy the ARMv6 link.

First of all, this isn't a bug report! It worked, swimmingly. Happily using `johnny-five` to control an attached Arduino Uno (I know that `raspi-io` isn't used in this scenario, but regardless, I used your lovely and helpful Wiki. Also, I _was_ able to use `raspi-io` via `johnny-five` to control the Pi's GPIO directly—it was stable, yay!—but the pinout on the Pi is so baffling that it was slowing me down so I opted to use an Uno for simplicity).

Anyway! The ARMv7 Node works fine, but Pi 3s actually have ARMv8 chips ([Source](https://www.raspberrypi.org/products/raspberry-pi-3-model-b/)). Wondering if the instructions should be updated _or_, alternately, if there is some reason that ARMv8 would _not_ be a good idea on the Pi 3.

nebrius, 2016-07-07:
TIL about the Raspberry Pi 3. I guess I assumed it was the same processor as the 2, not sure why. Thanks updated!

The pins are confusing on the RPi, it's pretty annoying. If I updated the pin information table with graphics to help visually orient where the pins are, would that help? BTW, I use https://www.adafruit.com/products/2028 to help with pintouts and it's _so_ helpful

lyzadanger, 2016-07-07:
The pins are indeed mind-melting. I have the SparkFun variant of the Cobbler (the Pi Wedge). The thing I find so confusing is: which is the _reference_ pin number? The actual pin from the Pi? The GPIO number mapping? The Wiring Pi abstraction? Gah! 

[There's a whole site dedicated to visual representation of Pi pinouts](https://pinout.xyz/) but I find I am _more_ confused after looking at that, not less. Finding the answer to, e.g., "which pins support PWM?" is a tortuous affair :). Let me know if I can be of help.

nebrius, 2016-07-07:
@lyzadanger have you seen the tables I created at https://github.com/nebrius/raspi-io/wiki/Pin-Information? They show the names that raspi-io/johnny-five expects and how to cross reference them between the reference number (which is the Wiring Pi abstraction number), the physical pin number, and the peripheral name. Anything that accepts a pin number can accept any of these designations, so:

```
const servo = new five.Servo('PWM1');
const led = new five.Led('P1-19');
const button = new five.Button('GPIO18');
```

P.S. I just updated the table to make the physical pin numbers more obvious

lyzadanger, 2016-07-07:
I sure have! And thank you! Given that there are actually two pins defined per line, is there any trick of GH table markdown that can make a clearer delineation between the left and right sides of the table? Would it be lame to have one line per pin (I realize it would make for a really long table, vertically)?

For the `Peripherals` column information, when you have a value like:

`GPIO14/TXD0`

I'm parsing this as: _This pin is mapped to GPIO14 and also supports TX._  Yes?

What do the `0`'s  and `1`s at the end of the values mean (and sorry if this is outside the scope of this thread!)? e.g.:

`GPIO20/MOSI1`

vs.

`GPIO18/PWM0`

Maybe some of this is an ignorance thing on my end...like, I don't know what `ID_SC` is, but that's not _your_ fault!

nebrius, 2016-07-07:
> I sure have! And thank you! Given that there are actually two pins defined per line, is there any trick of GH table markdown that can make a clearer delineation between the left and right sides of [...]

I'll see if I can add a clearer divider between them. I did it this way because the table exactly matches how the header looks when viewed from above, so you can count rows/columns to figure out which pin is which on the Pi itself.

> For the Peripherals column information, when you have a value like:

> GPIO14/TXD0

> I'm parsing this as: This pin is mapped to GPIO14 and also supports TX. Yes?

Not exactly, `TXD0` is the name of a specific pin function, in this case the transmit pin for UART port 0, not generic support for a type of peripheral. It's the same reason I list `GPIO14` instead of `GPIO`. `MOSI1` is the MOSI pin for SPI port 1, and `PWM0` is PWM port 0. `ID_SC` and `ID_SD` are for doing some custom EEPROM stuff, and probably shouldn't be listed on that chart.

nebrius, 2016-07-07:
OK, I updated  the pin page to include a description of what all the peripheral identifiers mean, let me know what you think!

lyzadanger, 2016-07-07:
This is most, most excellent! Would you trust me to make a couple of small formatting changes to the wiki page? GH wikis are hard because there’s no analog to a PR.

nebrius, 2016-07-07:
@lyzadanger I'd love for some help! Collaborator invite sent :)

And yes, I really wish we could do PR's to wikis too

nebrius, 2016-08-02:
Hey @lyzadanger, do you think everything in this issue has been resolved, and can we close it?

lyzadanger, 2016-08-02:
Oh, for sure!

closed 2016-08-02 by lyzadanger