nebrius/raspi-io #64: problem with i2C reading from GrovePI+

bblumberg, 2016-07-08:
Hi

I am experimenting with the GrovePi+ and Johnny5 and am running into an intermittent problem in my first test case (reading a light sensor). I am running the latest raspian and using Node 6.2.1

When I start the program it will start reading the sensor values but will crash after reading some number of values (seemingly random number). This is the console output and I have attached the little test program as well. Any help would be much appreciated. 

Parethetically, after reading the exchange about the difficulties of communicating in a NodeJS way with the GrovePI, I am nervous about whether it is a good solution in either case.

      if (register === undefined) {
               ^

Error: EIO: i/o error, write
    at Error (native)
    at Object.fs.writeSync (fs.js:780:20)
    at Bus.i2cWriteSync (/home/bbox/dev/j5/node_modules/i2c-bus/i2c-bus.js:340:13)
    at I2C.writeSync (/home/bbox/dev/j5/node_modules/raspi-i2c/lib/index.js:332:34)
    at Raspi.i2cWrite (/home/bbox/dev/j5/node_modules/raspi-io/lib/index.js:559:19)
    at Expander.<anonymous> (/home/bbox/dev/j5/node_modules/johnny-five/lib/expander.js:1305:19)
    at tryOnTimeout (timers.js:224:11)
    at Timer.listOnTimeout (timers.js:198:5

[j5Grove.js.zip](https://github.com/nebrius/raspi-io/files/355149/j5Grove.js.zip)

nebrius, 2016-07-12:
Hmm...that error is thrown when the device cannot be written to, and usually happens because there isn't a device present on that address. My first guess is that it's a hardware issue of some sort. How are you powering your Raspberry Pi? Perhaps it isn't providing enough power? Have you double checked all of your connections to make sure you have good ground connections?

bblumberg, 2016-07-12:
Thanks. I will investigate further and get back to you. To answer your
immediate question, I was using a 5.25V/2.4A power supply, and a Grove
sensor plugged into the GrovePI+, but I will try another sensor and see if
I have the same problem.

I'm a real believer in the NodeJS way, so after I had read your back and
forth about bringing up the board and looking at the javascript
implementation the grove folks provided, I was perhaps too quick to dismiss
the GrovePi+ when I ran into issues.

bb

nebrius, 2016-07-14:
Another possible reason, that @fivdi brought up in the gitter chat, is that the Raspberry Pi doesn't support I2C clock skewing. Do you know if the GrovePi is fast enough to support the baudrate you're using (defaults to 100000)?

Can you try editing `/boot/config.txt` and changing the baud rate from 100000 to 10000?

bblumberg, 2016-07-14:
Hi

I will try that later today. That seems like a great explanation for what I
was seeing because it was acting as if there was a timing issue of some
sort. And I do remember you having note to the effect of lowering the baud
rate.

Thanks for following up.

bb

bblumberg, 2016-07-14:
Indeed, @fivdi was absolutely correct. When I decreased the baud rate to 10000, and rebooted, the problem went away, when I restored the baud rate to 100000, the problem reappeared, so I think that is the bug and the fix.  So I am closing the issue. Thanks again!!!!

closed 2016-07-14 by bblumberg

nebrius, 2016-07-14:
🎉  @fivdi is definitely the I2C expert around here 😄