nebrius/raspi-io #65: can't use GrovePi Expander and raspi-io together: "Error: EIO: i/o error, write"

The-Alchemist, 2016-08-08:
### Stack Trace

* GroveP is connected
* ``i2cdetect -y 1`` shows the GrovePi at address 0x04

```
>> fs.js:658
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^

Error: EIO: i/o error, write
    at Error (native)
    at Object.fs.writeSync (fs.js:658:20)
    at Bus.i2cWriteSync (/home/pi/node_modules/raspi-io/node_modules/raspi-i2c/node_modules/i2c-bus/i2c-bus.js:340:13)
    at I2C.writeSync (/home/pi/node_modules/raspi-io/node_modules/raspi-i2c/lib/index.js:332:34)
    at Raspi.i2cWrite (/home/pi/node_modules/raspi-io/lib/index.js:559:19)
    at Expander.Controllers.GROVEPI.pinMode.value (/home/pi/node_modules/johnny-five/lib/expander.js:1219:17)
    at Expander.Controllers.GROVEPI.initialize.value (/home/pi/node_modules/johnny-five/lib/expander.js:1188:20)
    at new Expander (/home/pi/node_modules/johnny-five/lib/expander.js:1868:10)
    at Board.<anonymous> (/home/pi/grove-lcd-rgb-edison.js:10:20)
    at emitNone (events.js:72:20)
    at Board.emit (events.js:166:7)
    at nextTickCallbackWith0Args (node.js:420:9)
    at process._tickDomainCallback (node.js:390:13)
```

### Code

```
var five = require("johnny-five");
var raspi = require('raspi-io');

var board = new five.Board({
    io: new raspi()
});

board.on("ready", function() {

    var expander = new five.Expander({
        controller: "GROVEPI",
        address: 0x04,
        bus: "i2c-1"
    });

});

```

...and then:
* the GrovePi's red RST LED comes on, and stays on
* ``i2cdetect -y 1``does NOT show the GrovePi anymore :(
* reboot does not fix it, must hard reset

## Specs
* node v4.4.7
* npm 2.15.8
* Raspberry Pi 3
* Raspbian Jessie
* Linux 4.4.14-v7+ #896 SMP Sat Jul 2 15:09:43 BST 2016 armv7l GNU/Linux
* johnny-five 0.9.62
* raspi-io 6.1.0

nebrius, 2016-08-15:
That's...strange. If i2cdetect isn't showing it, this sounds like a slightly catastrophic failure, but unfortunately I have no idea why.

Some generic troubleshooting tips that might help:
- Try upgrading to node 6 and see if that makes a difference
- Try [lowering the I2C baudrate](https://github.com/nebrius/raspi-io#i2c-notes)

@fivdi do you have any more ideas on what could be causing this?

fivdi, 2016-08-28:
Despite the fact that `i2cdetect -y 1` initially works, I'd say the baud rate is too high.

The-Alchemist, 2016-08-28:
@fivdi : I have everything at the defaults (i.e., I started with a clean Raspbian Jessie image).  What's the best way to change the baud rate?

I can probably try this out later today. :)

fivdi, 2016-08-28:
The default baud rate is 100,000. Change it to 10,000 to see if it works.

After installing `raspi-io`, edit `/boot/config.txt` and change the following line:
```
dtparam=i2c_arm_baudrate=100000
```
to:
```
dtparam=i2c_arm_baudrate=10000
```
Note that the Pi needs a reboot for the change to take effect.

The-Alchemist, 2016-08-30:
Strangely, I can't reproduce the red RST light on the GrovePI error anymore.  I had some bad SD cards, so I maybe that was the cause of it??

But with the baud rate of 10,000, everything works great, thanks @fivdi and @nebrius !

For the record, this is the code I tested with:

```javascript
var five = require("johnny-five");
var raspi = require('raspi-io');

var board = new five.Board({
    io: new raspi()
});

board.on("ready", function() {

var virtual = new five.Board.Virtual(
    new five.Expander("GROVEPI")
  );

  // Plug the Rotary Angle sensor module
  // into the Grove Shield's A0 jack
  var rotary = new five.Sensor({
      pin: "A1",
      board: virtual});

  // Plug the LED module into the
  // Grove Shield's D6 jack. See
  // grove-led for more information.
  var led = new five.Led({
     pin: "D6",
     board: virtual});

  // Plug the Rotary Angle sensor module
  // into the Grove Shield's A0 jack
  var rotary = new five.Sensor({pin: "A0", board: virtual});

  // Set scaling of the Rotary angle
  // sensor's output to 0-255 (8-bit)
  // range. Set the LED's brightness
  // based on the value of the sensor.
  rotary.scale(0, 255).on("change", function() {
       	led.brightness(this.value);
  });

  // Plug the LCD module into any of the
  // Grove Shield's I2C jacks.
  var lcd = new five.LCD({
    controller: "JHD1313M1"
  });

  // Set scaling of the Rotary angle
  // sensor's output to 0-255 (8-bit)
  // range. Set the LCD's background
  // color to a RGB value between
  // Red and Violet based on the
  // value of the rotary sensor.
  rotary.scale(0, 255).on("change", function() {
    var r = linear(0xFF, 0x4B, this.value, 0xFF);
    var g = linear(0x00, 0x00, this.value, 0xFF);
    var b = linear(0x00, 0x82, this.value, 0xFF);

    lcd.bgColor(r, g, b);
  });

});

// [Linear Interpolation](https://en.wikipedia.org/wiki/Linear_interpolation)
function linear(start, end, step, steps) {
  return (end - start) * step / steps + start;
}

```

P.S. If I get around to it, I'll post some examples using the Raspberry Pi + GrovePi because ``docs/``.

closed 2016-08-30 by The-Alchemist

outan, 2017-05-16:
```
var five = require("johnny-five");
var raspi = require('raspi-io');
var board = new five.Board({
      io: new raspi()
});

board.on("ready", function() {
    var virtual = new five.Board.Virtual(
       new five.Expander("GROVEPI")
    );

    var led = new five.Led({
        pin: "D4",
        board: virtual
    });

    led.on();
});
```

sudo node test.js
```
1494937528166 Available RaspberryPi-IO
1494937528197 Connected RaspberryPi-IO
1494937528211 Repl Initialized
fs.js:718
    return binding.writeBuffer(fd, buffer, offset, length, position);
                   ^
Error: Unknown system error -121: Unknown system error -121, write
    at Object.fs.writeSync (fs.js:718:20)
```

I still got the same error above.

**Specs**

- node v7.2.1
- npm 3.10.10
- Raspberry Pi 3
- Raspbian Jessie
- johnny-five 0.10.6
- raspi-io 7.2.4

nebrius, 2017-05-22:
Crossposting for posterity:

So this is a thing: https://github.com/rwaldron/johnny-five/issues/1343#issuecomment-302909593.

There's a fix out that _may_ fix your issue too. Can you update to the latest version of raspi-i2c (one of raspi-io's dependencies) when you get a chance and let me know if you still see this issue?