nebrius/raspi-io #68: Fatal error in, line 0

styoe, 2016-09-03:
I am using raspberry 3 with electron, and after following the tutorial i get this error
#
# Fatal error in , line 0
# unreachable code
#

==== C stack trace ===============================

 1: V8_Fatal
 2: v8::internal::ToBooleanStub::Types::UpdateStatus(v8::internal::Handle<v8::internal::Object>)
 3: v8::internal::ToBooleanStub::UpdateStatus(v8::internal::Handle<v8::internal::Object>)
 4: v8::internal::ToBooleanIC::ToBoolean(v8::internal::Handle<v8::internal::Object>)
 5: v8::internal::Runtime_ToBooleanIC_Miss(int, v8::internal::Object**, v8::internal::Isolate*)

const serialport = require('serialport'),
    raspi = require('raspi-io'),
    five = require('johnny-five');

const board = new five.Board();

const io = new raspi();

As soon as i initialize the raspi io, i get this error.

Electron: 1.1.2
Node: 6.5

but i tried with different versions

nebrius, 2016-09-06:
What version of Node does Electron 1.1.2 use? Since Raspi IO has native modules, you have to make sure that your Node version and Electron's node version are identical in practice.

rwaldron, 2016-09-06:
@nebrius does Raspi-IO also need to be rebuilt for electron?

nebrius, 2016-09-06:
@rwaldron Raspi IO doesn't need to be built prior to publishing for electron, that step is _just_ babel (I don't use node-pre-gyp, although I probably should one of these days). the raspi-io package itself doesn't have any native code and is fine, but raspi-gpio, raspi-pwm, raspi, i2c-bus, and maybe a few others do, which get compiled at install time.

rwaldron, 2016-09-06:
Oh, got it

nebrius, 2016-10-13:
Closing this as the issue is pretty stale. Please reopen if you still need help!

closed 2016-10-13 by nebrius