nebrius/raspi-io #69: GPIO input and output doesn't appear to be working.

nkolban, 2016-09-25:
I am using the latest versions of the packages and my Pi 3B kernel reports - 4.4.13-v7+

I am running the following script:

```
"use strict";

var five = require('johnny-five');
var Raspi = require('raspi-io');
 
var board = new five.Board({
  io: new Raspi()
});
 
board.on('ready', function() {
  console.log("Board ready!");
  this.pinMode("GPIO17", five.Pin.OUTPUT);
  this.digitalWrite("GPIO17", 0);
});
```

The result of which should set GPIO17 to output with a low value.  As I ran the program, I wasn't seeing the expected results in the electronics so tested further.  Before running, I ran "gpio readall" and for GPIO17 found that it has a "1" value.  I ran the program and then re-ran "gpio readall" and it remains at a "1" level.  Are there any known issues at the current releases?  Is there a mechanism to enable additional diagnostics?

nebrius, 2016-10-13:
Sorry for the delay, I've been pretty swamped lately.

Looking into the code further, there must be something else going on. When Raspi IO is initialized, it force writes a 0 to the output at https://github.com/nebrius/raspi-io/blob/master/index.js#L275.

I don't have a way to test at the moment because I'm traveling, but once I can get my hands on some LEDs or a multimeter, I'll check it out.

nebrius, 2016-11-25:
I finally had a chance to dig into this and found the bug. It's fixed in v7.0.0. Let me know if it works for you!

nebrius, 2017-01-22:
Closing this issue. If you run into any other issues, let me know!

closed 2017-01-22 by nebrius