nebrius/raspi-io #70: Stepper motor support

techniq, 2016-09-29:
Any plans to support stepper motors, especially driver mode?

I have an Arduino johnny-five project I'm looking to port to a Raspberry PI.

nebrius, 2016-09-30:
Not directly, no, because the Raspberry Pi can't directly control a stepper motor. Stepper motors require very specially timed pulses, something the Raspberry Pi can't do unfortunately. To get the timing right, you either have to be running on bare-metal without an OS (such as the Arduino), or you have to use hardware PWM. The Raspberry Pi doesn't have enough PWM ports on it to drive a stepper motor though, and since it runs an OS, we can't fake it with GPIO and have it be stable because the OS can interrupt timers.

That said, pretty much any time you want to use a stepper motor with any microcontroller, not just the Raspberry Pi, the preferred way is to use an extension board, such as [this one here](https://www.adafruit.com/products/2348). These boards are connected to the Raspberry Pi using I2C, which is already supported by raspi-io.

Let me know if you have any other questions!

techniq, 2016-09-30:
Thanks for the quick reply @nebrius .

I'm currently using an Arduino and a [DRV8825](https://www.pololu.com/product/2133) with johnny-five.  Using the DRV8825, I'm able to step a motor "manually" by turning a pin `on` then `off`, like so:

```js
board.on('ready', function() {
  board.pinMode(7, board.MODES.OUTPUT);
  let spinPulse = true;
  setInterval(() => {
    board.digitalWrite(7, spinPulse ? 1 : 0);
    spinPulse = !spinPulse;
  }, 0);

  board.pinMode(8, board.MODES.OUTPUT);
  let direction = true;
  setInterval(() => {
    board.digitalWrite(8, direction ? 1 : 0);
    direction = !direction;
  }, 1000);
});
```
What this does is step the stepper as fast as possible using pin 7, and also change the direction every second using pin 8.

I've recently started using johnny-five's `Stepper` component with the `DRIVER` type, which has a nicer API and includes speed (rpm), acceleration, deceleration, and more, but the principal is the same.

```js
  var stepper = new five.Stepper({
    type: five.Stepper.TYPE.DRIVER,
    stepsPerRev: 200,
    pins: {
      step: 7,
      dir: 8 
    }
  });

  stepper.rpm(20).ccw().step(100, function {
    // ...
  });
```

My question is, could `raspi-io` support just the Stepper type `Stepper.TYPE.DRIVER`?.

Also, I've seen https://github.com/nkolban/node-stepper-wiringpi which looks to provide stepper motor control for Raspberry PI.  Would this be helpful in adding support?

dtex, 2016-09-30:
@techniq With the DRV8825 this is all just digital writes. Why wouldn't it work with rasp-io?

techniq, 2016-09-30:
@dtex I'll be honest, I haven't tried just yet.

I'm currently driving my Arduino with my Raspberry PI, but I saw [this](https://github.com/nebrius/raspi-io/blob/master/index.js#L799-L805), [this](https://github.com/rwaldron/johnny-five/blob/master/lib/stepper.js#L237), and [stepper support - no](http://johnny-five.io/platform-support/#raspberry-pi-2-model-b) and assumed it wouldn't work (yet).

dtex, 2016-09-30:
Hmmm, I'll need to try it out myself as I haven't actually done it yet either.

RonB, 2016-12-31:
Hi,
I'm using an adafruit motorhat and would also like support for steppers. I would be willing to try and port the python library of [adafruit](https://github.com/adafruit/Adafruit-Motor-HAT-Python-Library) to add to johnny5, allthough i am really new to this kind of programming.
Is this a good idea and if so, anyone care to help?
Ronald

nebrius, 2017-01-03:
> I'm using an adafruit motorhat and would also like support for steppers. I would be willing to try and port the python library of adafruit to add to rasp-io and johnny5, allthough i am really new to this kind of programming.
Is this a good idea and if so, anyone care to help?

Looking at the product, it should already work with the Raspberry Pi. The Adafruit Motor HAT uses I2C to communicate, which is already supported in raspi-io. Digging deeper, the HAT uses the [PCA9685](http://www.nxp.com/products/interfaces/ic-bus-portfolio/ic-led-display-control/16-channel-12-bit-pwm-fm-plus-ic-bus-led-controller:PCA9685), which is already a [supported expander](http://johnny-five.io/api/expander/) in Johnny-Five.

So, there's nothing to port, no assembly needed! :)

rwaldron, 2017-01-03:
@nebrius I think we still need a Stepper class controller (which would use the expander internally), but that's blocked by the Stepper class rewrite, which is necessary because it brings in controller definition support. However! Ignore all of that, one could definitely use an Expander object to implement a Stepper plugin that's specific to this shield.

RonB, 2017-01-05:
@nebrius and @rwaldron thank you for your response.

I realized that the pca9685 is already supported regarding the i2c communication, but how that ic has to write to the two motordriver ic's TB6612 needs to be addressed I think. In the Motor.js class of Johnny5 the ADAFRUIT shield is available, for the Stepper class it is not.

[image]

I've come this far:

`var raspi = require("raspi-io");
var five = require("johnny-five");
var io = new raspi();
var board = new five.Board({io: io});

board.on("ready", function () {
    
    var expander = new five.Expander({
        controller: "PCA9685",
        address: 0x60
    });
    
    var adafruitmotorhat = new five.Board.Virtual(expander);
    var stepper = new five.Stepper({
        type: five.Stepper.TYPE.DRIVER,
        stepsPerRev: 200,
        board: adafruitmotorhat,
        pins: {
            step: 11, //??
            dir: 12 //??
        }
    });
    // error stepper not supported because raspi-io io.MODES.STEPPER is not available
});`

but as the rasp-io board has stepper support turned off in general it will still report 'Stepper not supprted.' 

Can either off you point me in the right direction on where and how to write a stepper plugin specifically for this board?

//==== the other path i've walked... ==//

Another attempt I've made was to rewrite the python library from Adafruit to a node module by using the pca9685 [node driver](https://github.com/101100/pca9685) and translate the python ADAFRUIT Stepper Class. Unfortunately I'm too new in this kind of thing to get it working and I also beleive that this effort shouldn't be necessary. Just for information purposes I've included my poor attempt here: 

`
var i2cBus = require("i2c-bus");
var Pca9685Driver = require("pca9685").Pca9685Driver;

// initialize Stepper

function Stepper(options) {

    options = options || {};
    options.motornum = options.motornum || 1;
    options.steps = options.steps || 200;
    options.address = options.adress || 0x60;
    options.frequency = options.frequency || 50;

    var stepper = {};

    var pwmOptions = {
        frequency: options.frequency,
        address: options.address,
        i2c: i2cBus.openSync(1),
        debug: true
    };

    stepper.pwm = new Pca9685Driver(pwmOptions, function (err) {
        if (err) {
            console.error("Error initializing PCA9685");
            process.exit(-1);
        }
    });
[... 196 lines cut ...]
                  this._timerId = null;
                  if (callback) {
                    callback();
                  }
                  return;
                } // End of stepsLeft <= 0
            });
            
            lateststep = this.oneStep(direction, stepstyle);
            stepsLeft--;
            
            if (stepstyle === this.MICROSTEP) {
                // this is an edge case, if we are in between full steps, lets just keep going
                // so we end on a full step
                while (lateststep !== 0 && lateststep !== this.MICROSTEPS) {
                    lateststep = this.oneStep(direction, stepstyle);
                    // delay...
                }
            }
        }
    };
    return stepper;
}

module.exports = Stepper;`

rwaldron, 2017-01-05:
>  In the Motor.js class of Johnny5 the ADAFRUIT shield is available, for the Stepper class it is not.

This is what I was referring to w/r to `Stepper` class undergoing changes for new "controller definition" support. 

> // error stepper not supported because raspi-io io.MODES.STEPPER is not available

That's correct, because the `Stepper` class is still a very old implementation that pre-dates the decoupling that "controller definitions" provides. 

> the other path i've walked... 

A similar thing could be made using an `Expander` instance to implement a `Stepper` class. I haven't actually tested this, but I'm pretty sure it should work or is very close: 

1. Make a file called `stepper-hat.js`
```js
var five = require("johnny-five");

var Board = five.Board;
var Fn = five.Fn;

var TAU = Fn.TAU;
var priv = new Map();
var pins = [
  {
    PWMA: 8,
    AIN2: 9,
    AIN1: 10,
    PWMB: 13,
    BIN2: 12,
    BIN1: 11,
  },
  {
    PWMA: 2,
    AIN2: 3,
    AIN1: 4,
    PWMB: 7,
    BIN2: 6,
    BIN1: 5,
  }
];

var coilstates = [
  [1, 0, 0, 0],
  [1, 1, 0, 0],
  [0, 1, 0, 0],
  [0, 1, 1, 0],
  [0, 0, 1, 0],
  [0, 0, 1, 1],
  [0, 0, 0, 1],
  [1, 0, 0, 1],
];

var motors = [
  1, 2
];

function Stepper(opts) {
  if (!(this instanceof Stepper)) {
    return new Stepper(opts);
  }

  Board.Component.call(
    this, opts = Board.Options(opts)
  );

  var address = opts.address || 0x60;
  var frequency = opts.frequency || 1600;
  var motor = opts.motor || 1;
  var pins = Object.assign({}, pins[motor - 1]);
  var rpm = opts.rpm || 30;
  var steps = opts.steps || 200;
[... 244 lines cut ...]
};

module.exports = Stepper;
```

2. Your main program: 
```js

var five = require("johnny-five");
var raspi = require("raspi-io");
var Stepper = require("./stepper-hat.js");

var board = new five.Board({ io: new raspi() });

board.on("ready", function() {
  var stepper = new Stepper({ bus: 1, address: 0x60 });

  // Defaults to: 200 steps, 30 rpm
  stepper.steps(200, Stepper.FORWARD, Stepper.SINGLE, function() {
    console.log("Hopefully this fires when the complete move is done");
  });
});
```js

(I only used 0x60 in that code because you did, otherwise I would've used the default PCA9685 address of 0x40. Change if necessary.)

RonB, 2017-01-06:
@rwaldron  
Wow, this is brilliant. I've had to make some adjustments but it works... Very nice, thanks a lot. 

Any plans to include this in the standard Johnny5 lib? I'm using node-red so I want to include it in node-red-contrib-gpio :-)

`
var five = require("johnny-five");
var Board = five.Board;
var Fn = five.Fn;
var priv = new Map();
var pinassignment = [
    {
        PWMA: 8,
        AIN2: 9,
        AIN1: 10,
        PWMB: 13,
        BIN2: 12,
        BIN1: 11
    },
    {
        PWMA: 2,
        AIN2: 3,
        AIN1: 4,
        PWMB: 7,
        BIN2: 6,
        BIN1: 5
    }
];

var coilstates = [
    [1, 0, 0, 0],
    [1, 1, 0, 0],
    [0, 1, 0, 0],
    [0, 1, 1, 0],
    [0, 0, 1, 0],
    [0, 0, 1, 1],
    [0, 0, 0, 1],
    [1, 0, 0, 1]
];

var motors = [
    1, 2
];

function Stepper(opts) {
    if (!(this instanceof Stepper)) {
        return new Stepper(opts);
    }

    Board.Component.call(
            this, opts = Board.Options(opts)
            );

    var address = opts.address || 0x60;
    var frequency = opts.frequency || 1600;
    var motor = opts.motor || 1;
    var pins = Object.assign({}, pinassignment[motor - 1]);
    var rpm = opts.rpm || 30;
    var steps = opts.steps || 200;

    var state = {
        address: address,
        currentstep: 0,
        frequency: frequency,
        interval: null,
        motor: motor,
        pins: pins,
        secondsPerStep: 0,
        steps: steps
    };
[... 217 lines cut ...]
                // This is the edge case discussed in
                // Adafruit_MotorHAT.py...

                // If the last latest step was BETWEEN a
                // full step, then we'll take another step.
                //
                // But we're not done with this movement,
                // so we don't call the callback yet.
                if (lateststep !== 0 &&
                        lateststep !== Stepper.MICROSTEPS_8) {
                    this.steps(1, direction, style, callback);
                }
            } else {
                callback();
            }
        }

        numberOfSteps--;
    }.bind(this), ms);

    return this;
};

module.exports = Stepper;
`

nebrius, 2017-01-06:
> Any plans to include this in the standard Johnny5 lib? I'm using node-red so I want to include it in node-red-contrib-gpio :-)

Once the stepper rewrite lands in Johnny-Five, my understanding is that this shouldn't be necessary. Until that lands, there's not really a place for this in raspi-io or J5, based on how the division between IO plugins and J5 works.

That said, it _may_ be possible to get _full_ stepper support into raspi-io itself, now that we have support for software PWM. I'd need to do some testing to see if software PWM is stable enough for stepper motors, but I have the feeling it may be from what I've seen of it so far.

So let's game this out. If we decide to get this into Raspi IO, the first thing we need to do is add base stepper support into Raspi IO. This is a unique case where we _don't_ need to create a new Raspi.js plugin, because stepper motors aren't actually a unique peripheral (they're just PWM, not an all-together new peripheral). If you haven't seen the architecture discussion in [CONTRIBUTING](https://github.com/nebrius/raspi-io/blob/master/CONTRIBUTING.md#raspi-io-and-raspijs-modular-architecture), I recommend taking a peak.

This means that we would need to modify [these lines over in raspi-io-core](https://github.com/nebrius/raspi-io-core/blob/master/src/index.js#L923-L930) that first checks if software PWM is enabled (we'll probably need to store a flag on the Raspi IO Core instance, as it's currently just a parameter in the constructor), and then if software PWM is enabled, it does the right PWM things, otherwise it throws an error saying software PWM needs to be enabled.

Once that's done, my understanding is that the expander should just work, right @rwaldron?

rwaldron, 2017-01-06:
Probably safest to just make the thing we hashed out here into a plugin that can be `npm install`ed—would that work?

nebrius, 2017-01-06:
> Probably safest to just make the thing we hashed out here into a plugin that can be `npm install`ed—would that work?

I like that idea. We can release it quickly, and there's no risk to the existing plugin or worries about handling some controllers but not others or the confusion of not support stepper motors directly since it's a separate module. I can then add a blurb to Raspi IO's README about it.

@RonB if you can get a repo setup on your GitHub account with this code and drop the link to it here, we can help you flesh out the non-code bits necessary to get this wrapped into an npm module, if you'd like.

RonB, 2017-01-08:
Tanks both of you for the constructive support.
@nebrius I Will do that. First I will do some extensive testing and make sure the motor runs fine with different parameters and scenario's. Do hou have a convention for naming the repo/node module ie node-stepper-adafruit-motorhat?
@rwaldron 
I think I run into a problem if I want to combine the module with the standaard dc motor class as both will try to lock the PCA9685 address? Would it be a beter idea to create a module for the entire board and include steppen and dc motor support?

rwaldron, 2017-01-09:
>I think I run into a problem if I want to combine the module with the standaard dc motor class as both will try to lock the PCA9685 address? Would it be a beter idea to create a module for the entire board and include stepper and dc motor support?

There shouldn't be any such locking with address, since `Expander.get(...)` will either create a new Expander for a given I2C address or reuse an already existing instance associated with that address.

nebrius, 2017-01-09:
> I Will do that. First I will do some extensive testing and make sure the motor runs fine with different parameters and scenario's. Do you have a convention for naming the repo/node module ie node-stepper-adafruit-motorhat?

I don't have a naming convention except for the core libraries for raspi-io. Pick whatever you prefer! I would include something in the name indicating it's specifically for the raspberry pi, maybe `pi-adafruit-motorhat-stepper`?

Triadager, 2017-10-10:
Hello,

I'm sorry to bother, but I wanted to control stepper drivers (A4988, DRV8825 and TB6600) with johnny-five. Why is the stepper class disabled for the raspberry pi? My understanding is that it should work without any (major) issues, since all the RasPi has to do is sending a few pulses to the pins. Could somebody explain whats going on there? Am I missing something?

I highly appreciate your help!

Kind regards, 

Triadager

dtex, 2017-10-10:
Update on stepper.

There is a [PR](https://github.com/firmata/firmata.js/pull/179) open for the new stepper API in firmata.js. I think it's nearly ready to land, we're just seeing some failures on Travis that need to be ironed out. I don't think the API itself will change at all.

I'm working on the new Stepper class for Johnny-Five. It will include support for both the [accelStepper](http://www.airspayce.com/mikem/arduino/AccelStepper/) library that can be bundled with [configurableFirmata](http://firmatabuilder.com/) and a slower J5 only solution that will just uses digital pin writes. 

For the former to work in rasp-io we would need to port the accelStepper library so that we can replicate that functionality. Hopefully someone will step up and want to take that on. I hope to have the J5 class done in the next couple of months.

For the latter to work in rasp-io there will need to be one small change made to reflect a change I'm planning to submit for firmata.js. I need to test a hypothesis first.

nebrius, 2017-10-10:
So I think I may be missing something, but I cannot for the life of me figure out how to download accelStepper so I can peak at the files.

Anyways.

My plate is really full with Raspi IO work for the rest of the year, so I won't be able to take on this work, unfortunately. Is this something you would be interested in working on @Triadager? I can help walk you through how to integrate it into Raspi IO

Triadager, 2017-10-10:
@nebrius 
Regarding accelStepper files: I think you might be looking for this: http://www.airspayce.com/mikem/arduino/AccelStepper/AccelStepper-1.57.zip 
It's the package with examples. Maybe you didn't see the link on the page? 

I would love to, but since I only got into touch with both the Raspberry Pi and Node.js a couple of weeks ago and don't really have any other programming experience, I might not be the right guy for that. I still need to learn loads of things. However, if you still think I could be of any help let me know, I would love to contribute!

At this point I just want to thank all of you big time for the effort you put into this. It's awesome.

nebrius, 2017-10-11:
Ah, thanks for the link @Triadager! Sorry we couldn't be of more help to you right now. And I sadly do think you're right, this _particular_ project will probably require more experience than most changes require :-/

nebrius, 2017-11-22:
I'm going to go ahead and close this issue due to inactivity. Feel free to reopen if needed!

closed 2017-11-22 by nebrius

a1xon, 2018-03-10:
Isn't it possible to run a stepper with a simple driver like shown in this example? http://johnny-five.io/examples/stepper-driver/

a1xon, 2018-03-10:
I still don't get why it's more complicated to talk to a stepper driver with an RPi than with an Arduino? Can somebody explain?

nebrius, 2018-03-11:
> I still don't get why it's more complicated to talk to a stepper driver with an RPi than with an Arduino? Can somebody explain?

Super short tl;dr:

1) the Arduino and the Raspberry Pi support different kinds of hardware because they are different kinds of devices. The Arduino is more geared towards embedded applications than the Arduino is...Broadcom and Atmel designed their respective chips for different things. There's no such thing as a perfect, one-size-fits-all design in the hardware world. Each company made their choices for their respective chips, and we can't change that (you can't patch hardware).

2) The Raspberry Pi runs an operating system and the Arduino does not. This has major implications on what software is available and guarantees on timing within the OS. Having a thread scheduler vs. not having a thread schedular is really significant when it comes to timing of things, and timing is really important when it comes to certain hardware operations. It makes it so that many hardware protocols _cannot_ be emulated in software when there is a thread scheduler, but _can_ be implemented when there is not a thread scheduler. When there's a thread scheduler, the timing is such that you can guarantee that a `setTimeout` will take _no shorter than x ms_, but there's no guarantee that it won't take longer thn x ms.

Comparing the Raspberry Pi and the Arduino is a lot like comparing ANSI C (not even C++) to JavaScript. They both have their strengths and differences, but they are radically different languages.

dtex, 2018-03-11:
I have successfully controlled a stepper connected to an Arduino from Johnny-Five running on my Macbook Pro without using stepper specific c code on the Arduino (i.e. firmataStepper or firmataAccelSteper). 

Here's a video https://twitter.com/dtex/status/922597004500504576

I'm using just standardFirmata and JS to toggle the pins. This code will run on a Pi w/o any changes to rasp-io but I have no clue how well it will perform.

I've seen decent performance on my Macbook (about 500 steps per second) but I know and accept that my step timing is not gauranteed to be smooth. For instance, when the V8 garbage collector does its thing the event loop grinds to a halt. How long that takes depends on so many things we may as well say it's nondeterministic. It might be a couple milliseconds, it might be less. When it does kick in there'll be a hitch in the stepper's giddyup (sorry, the rodeo is in town this week). So far those events are undetectable by my eye, but my program is doing nothing but spinning a single stepper.

Generally performance will depend on how quickly the Pi can run through the event loop (really needs to be more than once every millisecond on average to achieve 500 steps per second smoothly). It will also depend on the interface between the CPU and GPIO because we are toggling lots of pins really quickly.

All this code lives in a complete refactor of the Johnny-Five stepper class. It's not even ready for first commit in my mind but if you are adventurous, have time to kill, and want to try it out let me know and I will share the code in a gist. So far you can only spin the motor, but it's a start.

BTW, I'm working on getting my first Pi up and running with Johnny-Five so I'll start testing it myself soon enough. I have another J5 refactor that's turned into a real hairball and has led me to back burner this stepper stuff for a while.

a1xon, 2018-06-09:
@dtex any updates? I ran servos with the RPi via a simple python script without any problems. But it feels stupid calling a python script that cares for the servos while working with johnny five.

a1xon, 2018-06-11:
I'm sorry. Of course I meant steppers and not servos. I followed [this tutorial](https://www.youtube.com/watch?v=LUbhPKBL_IU&t=593s) for the python script. But for my project I've decided I will use a continuous servo so I can keep the code in plain JavaScript. Thank you so much for your work on jonny-five. If you need testers for your code I'm happy to fry a pi for you.

oligriffiths, 2020-05-08:
Hey @RonB @rwaldron @nebrius what's the state of play with this?

I recently received the Adafruit Motor Controller HAT https://www.adafruit.com/product/2348 and was hoping to use the stepper motor functionality with J5. Is this natively supported now or does it still require the custom Stepper class as outlined above?

Thanks

nebrius, 2020-05-18:
Hi @oligriffiths. There hasn't been any progress on this issue since the last posting date. I'm afraid that raspi-io isn't as active as it used to be, and isn't likely to be implemented.