nebrius/raspi-io #73: Support for software PWM

andrewn, 2016-10-17:
Raspi-io currently supports the Raspberry Pi's PWM pin via the [`nebrius/raspi-pwm`](/nebrius/raspi-pwm) library that wraps Wiring Pi.

It would be nice to have a way to fall back to [Wiring Pi's Software PWM](http://wiringpi.com/reference/software-pwm-library/) if the pin selected doesn't provide hardware PWM.

This would allow Johnny Five's [`LED#brightness`](http://johnny-five.io/api/led/) and [`LED.RGB`](http://johnny-five.io/api/led.rgb/) to work transparently on the Pi since they both require PWM.

tralves, 2016-10-17:
This kind of exists through the package [raspi-soft-pwm](https://github.com/tralves/raspi-soft-pwm).

This package uses [pigpio](https://github.com/fivdi/pigpio), which in turn uses the [pigpio C library](https://github.com/joan2937/pigpio).

andrewn, 2016-10-17:
Oops. :-)

Looks like I've duplicated your work by creating another [`raspi-soft-pwm`](https://github.com/andrewn/raspi-soft-pwm) but using Wiring Pi as the basis instead. I even picked the same name but didn't try to push to NPM otherwise I would have discovered your package!

I think having Raspi-IO support either one of these approaches would be useful.

nebrius, 2016-11-29:
Support for raspi-soft-pwm in raspi-io has been published w/ 7.1.0! Thanks again for all of your help @tralves and @andrewn!

closed 2016-11-29 by nebrius