nebrius/raspi-io #75: i2cConfig delay should be in microseconds, not milliseconds

RandScullard, 2016-10-24:
The raspi-io module is taking the delay option passed to i2cConfig and passing it directly to setTimeout as a number of milliseconds. However, according to the firmata documentation, i2cConfig is supposed to take a delay in microseconds: https://github.com/firmata/firmata.js#i2c

The result is that when I run the johnny-five thermometer example, it waits 50 _seconds_ after initialization before the **data** event starts firing, instead of 50 _milliseconds_ as intended:

    var five = require("johnny-five");
    var Raspi = require("raspi-io");
    var board = new five.Board({
      repl: false,
      io: new Raspi()
    });
    
    board.on("ready", function() {
      var thermometer = new five.Thermometer({
        controller: "HTU21D"
      });
    
      thermometer.on("data", function() {
        console.log(this.celsius + "°C", this.fahrenheit + "°F");
      });
    });

I am using a Raspberry Pi 3 Model B running Jessie with Node v6.9.1.

rwaldron, 2016-10-24:
Nice find! Here's the io-plugins info about it: https://github.com/rwaldron/io-plugins#i2cconfigoptions

nebrius, 2016-11-01:
Fixed with raspi-i2c 3.1.1

closed 2016-11-01 by nebrius