nebrius/raspi-io #76: After installing raspi-io, node-red-node-pi-mcp3008 readings...

aleonnet, 2016-11-04:
After installing raspi-io, node-red-node-pi-mcp3008 readings only outcomes 0.
Note that It worked perfecly before raspi-io installations. Also, if I remove it, mcp3008 comes back to work.
What can I do to avoid this scenario? is this some kind of bug?
Thanks

fivdi, 2016-11-05:
Here's a guess: During the initialization phase of a Johnny-Five/Raspi-IO application all available GPIOs on the Raspberey Pi are automatically configured as GPIO outputs, irrespective of whether the Johnny-Five/Raspi-IO application actually use those GPIOs or not. On the other hand, node-red-node-pi-mcp3008 requires that the SPI specific GPIOs be configured as SPI GPIOs, not as GPIOs outputs. They were actually configured as SPI GPIOs at boot time but the fact that Johnny-Five/Raspi-IO reconfigures everything to be a GPIO output then breaks node-red-node-pi-mcp3008.

aleonnet, 2016-11-05:
Would some expert check and correct it?

nebrius, 2016-11-05:
@fivdi is one of the best experts around here :). Given that it's I2C, he's correct, it's initialized to GPIO at start, and there is no way using raspi-io to change it to SPI because it doesn't support SPI under the hood either.

I have plans to support SPI sooner rather than later because I want to use it for a project of my own, so hang tight! It'll probably be a few weeks though, just FYI.

If you need something sooner, could you try an I2C ADC instead? https://www.adafruit.com/products/1083

aleonnet, 2016-11-05:
@nebrius I'm sure you and @fivdi are great experts! Guys, awesome news! :) I'll hang tight for sure!
Thanks a lot for wonderful job and support.

fivdi, 2016-11-06:
Maybe it would be a good idea if Johnny-Five/Raspi-IO allowed the application to optionally specify which pins the application uses. Then only these pins would be automatically configured as GPIO outputs during the startup phase. All other pins would be left untouched. This would enable Johnny-Five/Raspi-IO to play well with other modules which currently isn't the case. Initializing all pins on an Arduino is a reasonable thing to do as  only the firmata sketch will be running on the Arduino. Initializing all pins on Linux boards like the Raspberry Pi is a less reasonable thing to do as it can prevent other processes (or other modules in the same process as Johnny-Five)  from using pins that are not used by Johnny-Five.

nebrius, 2016-11-07:
> Maybe it would be a good idea if Johnny-Five/Raspi-IO allowed the application to optionally specify which pins the application uses. Then only these pins would be automatically configured as GPIO [...]

I like this! I think the only difference I would make would be to invert it and say "here's a list of pins to _not_ use." I say this because most people probably would only care about avoiding a couple of pins, and considering how many pins there are on the Pi, it could get lengthy to do the positive case. Then again, we could always have a `usePins` _and_ `ignorePins` option :)

nebrius, 2016-11-23:
I just published v6.2.0 with support for `includePins` and `excludePins` configuration options. Let me know if this works for you!

aleonnet, 2016-11-25:
Hi @nebrius!
I notice that node-red-contrib-noble stops scaning BLE devices after Raspi-io deployment.
Maybe due to some Raspi-io usage of serial port. What do you think?

Also, I can't instantiate "board" because it crashes my flow (Error: Board closed.).
Could you provide a sample of code showing how to use it correctly along with you new version of Raspi-io?

Thanks!

nebrius, 2016-11-25:
You should be able to get it to work using:

```JavaScript
var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi({
    excludePins: [
      'MOSI0',
      'MISO0',
      'SCLK0',
      'CE0',
      'MOSI1',
      'MISO1',
      'SCLK1',
      'CE1',
    ];
  })
});

board.on('ready', function() {
  // Do work here
});
```

nebrius, 2016-11-25:
As for the serial thing, it's a known issue that I just filed https://github.com/nebrius/raspi-serial/issues/3 to track.

aleonnet, 2016-12-04:
@nebrius exclude pins worked!
For the serial thing, I'm trying to use bluetooth and maybe to exclude UART TXD/RXD PINS (8/10) would work... http://raspberrypi.stackexchange.com/questions/45570/how-do-i-make-serial-work-on-the-raspberry-pi3
What do you think? Anyway I'm failing with the name of the PINS. 

3 Dec 23:56:28 - [red] Uncaught Exception:
3 Dec 23:56:28 - Error: Invalid pin "TXD" specified in excludePins
at /root/.node-red/node_modules/raspi-io/lib/index.js:267:21
nodered.service: main process exited, code=exited, status=1/FAILURE

Thanks

nebrius, 2016-12-04:
> 3 Dec 23:56:28 - Error: Invalid pin "TXD" specified in excludePins

Try `TXD0` instead and it should work. FYI the full list of pins and valid pin names are [listed in the wiki](https://github.com/nebrius/raspi-io/wiki/Pin-Information).

nebrius, 2017-01-22:
Closing this issue, since it seems to be resolved. Feel free to reopen if there's still an issue!

closed 2017-01-22 by nebrius

vallettea, 2018-06-28:
This works but it should be documented in the readme as it is a bit difficult to find the information

nebrius, 2018-06-28:
> This works but it should be documented in the readme as it is a bit difficult to find the information

Can you clarify which part specifically is difficult to find?

vallettea, 2018-06-28:
Sorry! My mistake I to much focused `mcp-spi-adc` documentation and didn't realized that everything is clear in yours.
By the way thanks a lot for this lib !