nebrius/raspi-io #77: Pin 8 does not support mode 1

DannyBoyBroadSword, 2016-11-23:
```
var Raspi = require("raspi-io");
var five = require("johnny-five");
var pitft = require("pitft");
var sleep = require('sleep');
var
//var pitftTouch = require('pitft-touch');

var WriteVolumeAddress=0x4B;
var ReadStrokeAddress=8;
var volumeLevel=31;
var isStarted=false;

var fb = pitft("/dev/fb1"); // Returns a framebuffer in direct mode.  See the clock.js example for double buffering mode
//var touchscreen = pitftTouch("/dev/input/touchscreen");

var xMax = fb.size().width;
console.log(xMax);
var yMax = fb.size().height;
console.log(yMax);

// Clear the screen buffer
fb.clear();
fb.color(1, 1, 1); // Set the color to whit
fb.font("fantasy", 12); // Use the "fantasy" font with size 12

fb.text(0,20,"This will never work.",false);

console.log("Starting Board");

var board = new five.Board({
  io: new Raspi()
});

board.on("ready", function() {
  //Prepare i2c Data
  new five.Pin('SDA0',five.Pin.i2c);

  //prepare buttons from TFT
  var buttonOne = new five.Button(0);
  var buttonTwo = new five.Button(3);
  var buttonThree = new five.Button(4);
  var buttonFour = new five.Button(2);

  //prepare imu
  var imu = new five.IMU({
    controller: "MPU6050"
  });

  //prepare gps
  var gps = new five.GPS({
    breakout: "ADAFRUIT_ULTIMATE_GPS",
    pins: ['P1-8', 'P1-10']
  });

  //create a function that reads the Volume.
  function readVolume(address){
    return defaultVolumeLevel;
  }

  //create a cunction that writes Volume
  function writeVolume(address,volume){
    return board.io.i2cwrite(address,volume)
    //sample writeVolume(WriteVolumeAddress,defaultVolumeLevel);
  }

  //create a function that read's stroke rate.
  function readStrokeRate(address){
    return board.io.i2cread(address,4,true);
    //sample readStrokeRate(ReadStrokeAddress);
  }

  //default volume
  writeVolume(writeVolumeAddress,volumeLevel);

  buttonOne.on("press", function() {
    writeVolume(WriteVolumeAddress,VolumeLevel+1);
    VolumeLevel = VolumeLevel+1
    console.log( "Increased Volume" );
  });

  buttonTwo.on("press", function() {
    writeVolume(writeVolumeAddress,VolumeLevel-1);
    VolumeLevel = VolumeLevel-1
    console.log( "Decreased Volume" );
  });

  buttonThree.on("press", function() {
    if (isStarted == false){
      //startSomething();
      console.log("Started");
    }else{
      //stopSomething();
      console.log("Stopped");
    }
  });

  buttonThree.on("hold", function() {
    //resetTimer();
    isStarted = false;
    console.log( "Reset" );
  });

  buttonFour.on("press", function() {
    console.log( "IDK fam" );
    //doSumting();
  });

  touchscreen.on("data.touch", function() {
    var screenX = ((270 - 50) / (693 - 3307)) * data.x + 320;
    var screenY = ((190 - 50) / (3220 - 996)) * data.y;
    console.log("  screenX   : ", screenX);
    console.log("  screenY  : ", screeny);
  });

  gps.on("change", function() {
    console.log("GPS Data");
    console.log("  Speed   : ", gps.speed);
    console.log("  Course  : ", gps.course);
    console.log("  latitude   : ", gps.latitude);
    console.log("  longitude  : ", gps.longitude);
    console.log("  altitude   : ", gps.altitude);
    console.log("MPU6050 Roll Data");
    console.log("  pitch   : ", imu.gyro.pitch);
    console.log("MPU6050 Starting Acceleration");
    console.log("  Acceleration   : ", imu.accelerometer.acceleration);
    console.log("MPU6050 Temp");
    console.log("  fahrenheit   : ", imu.thermometer.fahrenheit);
    console.log("Volume");
    console.log("  Volume   : ",volumeLevel);
    console.log("Stroke Rate");
    console.log("  Stroke Rate   : ",readStrokeRate(ReadStrokeAddress));
    console.log("--------------------------------------");
  });
});
```
My Code above.

My error's
```
23.11.16 00:42:11 [-0500] 320
23.11.16 00:42:11 [-0500] 240
23.11.16 00:42:11 [-0500] Starting Board
23.11.16 00:42:11 [-0500] 1479879731387 Device(s) RaspberryPi-IO
23.11.16 00:42:11 [-0500] 1479879731470 Connected RaspberryPi-IO
23.11.16 00:42:11 [-0500] 1479879731512 Repl Initialized 
23.11.16 00:42:11 [-0500] GJ>> G/usr/src/app/node_modules/raspi-io/lib/index.js:366
23.11.16 00:42:11 [-0500] throw new Error('Pin "' + pin + '" does not support mode "' + mode + '"');
23.11.16 00:42:11 [-0500] ^
23.11.16 00:42:11 [-0500]
23.11.16 00:42:11 [-0500] Error: Pin "8" does not support mode "1"
23.11.16 00:42:11 [-0500] at Raspi.value (/usr/src/app/node_modules/raspi-io/lib/index.js:366:15)
23.11.16 00:42:11 [-0500] at Raspi.pinMode (/usr/src/app/node_modules/raspi-io/lib/index.js:348:21)
23.11.16 00:42:11 [-0500] at Pin.set (/usr/src/app/node_modules/johnny-five/lib/pin.js:82:17)
23.11.16 00:42:11 [-0500] at new Pin (/usr/src/app/node_modules/johnny-five/lib/pin.js:90:13)
23.11.16 00:42:11 [-0500] at Board.<anonymous> (/usr/src/app/server.js:36:3)
23.11.16 00:42:11 [-0500] at emitNone (events.js:91:20)
23.11.16 00:42:11 [-0500] at Board.emit (events.js:185:7)
23.11.16 00:42:11 [-0500] at _combinedTickCallback (internal/process/next_tick.js:67:7)
23.11.16 00:42:11 [-0500] at process._tickDomainCallback (internal/process/next_tick.js:122:9)
```

I think the tutorial code was too generous in saying:
```
var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi()
});
 
board.on('ready', function() {
  new five.Pin('SDA');
  board.io.i2cWrite(0x18, 0x5, 'hello');
});
``` 

I Found that here: `https://www.npmjs.com/package/raspi-i2c`

nebrius, 2016-11-23:
The error code you're getting is accurate, pin 8 in fact does _not_ support mode 1.

Let's break down what's happening here. When you create a pin via `new five.Pin` and you pass it a pin number, you are creating a _digital input_ pin (this is mode 1). You are passing in `SDA` to the pin constructor, which maps to pin 8 under the hood and is an I2C pin. Now here's the thing, as stated in the [Raspi IO README](https://github.com/nebrius/raspi-io/#i2c-notes), the I2C pins _cannot_ be used for digital input (or output).

It looks like you're trying to use raspi-io to communicate over I2C, so to fix your code all you need to do is delete the line that says `new five.Pin('SDA');` so that it looks like this:

```JavaScript
var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi()
});
 
board.on('ready', function() {
  board.io.i2cWrite(0x18, 0x5, 'hello');
});
```

Let me know if this fixes it for you.

nebrius, 2016-11-29:
Closing for now, since I don't think there's a bug here. Feel free to reopen if you want!

closed 2016-11-29 by nebrius