nebrius/raspi-io #78: need to be able to pass in serial config options

monteslu, 2016-12-19:
Since node serialport is being used for the serial* methods, we need to be able to pass config options from the raspi-io constructor on down to the serial port constructor.

The defaults in serial port now lock the port in linux which has the effect of only allowing us to initialize raspi-io once.  This breaks my node-red implementation in that I can't make changes and redeploy a flow.

nebrius, 2016-12-21:
Which config options do you want to pass down?

monteslu, 2017-01-06:
doh, I missed your reply @nebrius 

The main issue is that serialport now defaults to locking the port which means if we try re-initializing raspi-io in the same node instance we're blocked.

nebrius, 2017-01-06:
I have two ideas for working around this. One is to set `lock` to `false`, the other is to listen to the `beforeExit` event and close the port there, but leaving it locked. The former is simpler, and I would guess would be more reliable, but I also worry about unintended side effects, especially if someone re-enables TTY console over serial.

nebrius, 2017-01-22:
I just published 3.0.4 of raspi-serial, which includes the `beforeExit` approach to closing the port. Let me know if this works for you!

monteslu, 2017-01-23:
It works!

Thanks, @nebrius

closed 2017-01-23 by monteslu

nebrius, 2017-01-23:
Woo!