nebrius/raspi-io #79: pi msg : string [46] TypeError: Cannot read property 'name' of null

RonB, 2016-12-28:
Hi,

i am getting a strange error in a simple node-red flow trying to control a dc-motor. I use a pi-3 with an adafruit motorhat v2. A simple dashboard with a slider that sends a value of 0 to 255 to the johnny5 node in which i set the motor speed.

The flow:

`[{"id":"57c1a75.9387458","type":"ui_slider","z":"e00e6236.010bd","name":"speed","label":"Snelheid","group":"9b8accf3.638c9","order":0,"width":"3","height":"1","passthru":false,"topic":"Speed","min":0,"max":"255","x":400,"y":250,"wires":[["2992a1d4.f0577e"]]},{"id":"2992a1d4.f0577e","type":"johnny5","z":"e00e6236.010bd","name":"dcmotor","func":"var configs = five.Motor.SHIELD_CONFIGS.ADAFRUIT_V2;\nvar motor3 = new five.Motor(configs.M3);\nmotor3.stop();\nnode.on('input', function(msg) {\n    motor3.forward(msg.payload);\n});","board":"54a00bc0.b92e54","noerr":0,"x":610,"y":249,"wires":[[]]},{"id":"9b8accf3.638c9","type":"ui_group","z":"e00e6236.010bd","name":"adafruit board","tab":"9b3b082c.545b18","disp":true,"width":"12"},{"id":"54a00bc0.b92e54","type":"nodebot","z":"e00e6236.010bd","name":"pi","username":"","password":"","boardType":"raspi-io","serialportName":"","connectionType":"local","mqttServer":"","socketServer":"","pubTopic":"","subTopic":"","tcpHost":"","tcpPort":"","sparkId":"","sparkToken":"","beanId":"","impId":"","meshbluServer":"https://meshblu.octoblu.com","uuid":"","token":"","sendUuid":""},{"id":"9b3b082c.545b18","type":"ui_tab","z":"e00e6236.010bd","name":"Home","icon":"dashboard"}]`

At first init of node-red the flow runs fine and i can control the motor. When I click deploy in the editor it immediately returns an error in the debug tab and on the console:

`28-12-2016 15:09:22 pi
msg : string [46]
TypeError: Cannot read property 'name' of null`

pi is the name of the raspberry pi board so it looks like the error comes from the board.

The motor is after that unresponsive and I have to restart node-red on the pi.

Any help would be appreciated.
Ronald

nebrius, 2016-12-28:
I'm sorry to hear you're having trouble. Unfortunately I know next to nothing about Node red, and I'm not able to infer enough about what's going on here to say what's wrong.

Do you know if it's possible to get the full stack trace for the two errors?

cc @monteslu

RonB, 2016-12-28:
Ok, thats one of the things which is very difficult to debug. I will try to walk a different path.  Thanks for your swift reply.

nebrius, 2017-04-05:
Closing due to inactivity. Feel free to reopen if you're able to gain more insight into what's going on.

closed 2017-04-05 by nebrius