nebrius/raspi-io #80: Error when installing when Raspberry 3

ppopadiyn, 2017-01-04:
Hi

I tried installing raspi-io on raspberry 3. And that's the error what I get:

```
npm ERR! Linux 4.4.34-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "raspi-io"
npm ERR! node v6.9.2
npm ERR! npm  v3.10.9
npm ERR! code ELIFECYCLE
npm ERR! **raspi@3.0.4 install: `node-gyp rebuild`**
npm ERR! Exit status 1
```

Any ideas? Few months ago I tried it on Raspberry 2 without any issues.

Thanks in advance

nebrius, 2017-01-04:
Can you include the full output of `npm install`? Also, are you running the November 2016 release of Raspbian?

ppopadiyn, 2017-01-04:
My bad, after a detail look at the npm install log, I saw that I am missing wiringPi. I am using the minimal image - RASPBIAN JESSIE LITE and seems that wiringPi is not pre-installed there. After installing it, ```npm install raspi-io``` completed successfully.

closed 2017-01-04 by ppopadiyn

nebrius, 2017-01-04:
Oh, good to know. I haven't really played with the Lite version, and so I didn't know it didn't come with Wiring Pi. I'll add a note to the README. Thanks!