nebrius/raspi-io #81: wiringPiISR: unable to open /sys/class/gpio/gpio4/value: No such file or directory

alancairns, 2017-01-09:
I've been trying to use johnny-five on a Raspberry Pi 3 with node.js and also with node-red without any success. So I have tried to install and use it from a clean install as follows:

on iMac:
- sdformatter
- Download and copy Noobs v 2.1 0 to microSD card

on Raspberry Pi
- Installed Raspian
- sudo apt-get update
- sudo apt-get upgrade
- update-node.js-and-node-red (as advised on node-red site)
- node -v  (reports v4.7.2)
- npm -v  (reports 2.15.11)

(instructions from http://johnny-five.io/examples/raspi-io/)
- npm install johnny-five raspi-io
- create led-blink.js
	var five = require("johnny-five");
	var board = new five.Board();

	board.on("ready", function() {

  		var led = new five.Led(13);

  		// "blink" the led in 500ms on-off phase periods
  		led.blink(500);
	});

- sudo node led-blink.js
	1483954014083 Device(s) RaspberryPi-IO  
	wiringPiISR: unable to open /sys/class/gpio/gpio4/value: No such file or directory

nebrius, 2017-01-09:
That's strange, this error is coming from the underlying library used for GPIO. Can you do me a favor and run `gpio edge gpio4 rising` from your Raspberry Pi's command line to see if it gives you the same error? If so, it would seem you're installation of wiring pi is broken.

alancairns, 2017-01-10:
Thanks for the quick response. When I run gpio edge gpio4 rising there is no output from the command and I'm returned to the command prompt.

I hope the following helps rather than confuses:

- sudo reboot
- ls -l /sys/class/gpio/
	-rwxrwx--- 1 root gpio 4096 Jan  9 10:10 export
	lrwxrwxrwx 1 root gpio    0 Jan  9 10:10 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0
	lrwxrwxrwx 1 root gpio    0 Jan  9 10:10 gpiochip100 -> ../../devices/platform/soc/soc:virtgpio/gpio/gpiochip100
	-rwxrwx--- 1 root gpio 4096 Jan  9 10:10 unexport

- gpio exports
- gpio edge gpio4 rising
- gpio exports
	GPIO Pins exported:
   	0: in   1  rising  

- ls -l /sys/class/gpio/
	total 0
	-rwxrwx--- 1 root gpio 4096 Jan 10 08:50 export
	lrwxrwxrwx 1 root gpio    0 Jan 10 08:50 gpio0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio0
	lrwxrwxrwx 1 root gpio    0 Jan 10 08:48 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0
	lrwxrwxrwx 1 root gpio    0 Jan 10 08:48 gpiochip100 -> ../../devices/platform/soc/soc:virtgpio/gpio/gpiochip100
	-rwxrwx--- 1 root gpio 4096 Jan 10 08:48 unexport

- cat app.js
	var five = require("johnny-five");
	var Raspi = require("raspi-io");
	var board = new five.Board({
 	io: new Raspi()
	});

	board.on("ready", function() {
  	var led = new five.Led("P1-13");
 	 led.blink();
	});

- sudo node app.js
	1484038988967 Device(s) RaspberryPi-IO  
	wiringPiISR: unable to open /sys/class/gpio/gpio4/value: No such file or directory

- ls -l /sys/class/gpio/
	total 0
	-rwxrwx--- 1 root gpio 4096 Jan 10 09:03 export
	lrwxrwxrwx 1 root gpio    0 Jan 10 08:50 gpio0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio0
	lrwxrwxrwx 1 root gpio    0 Jan 10 09:03 gpio17 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio17
	lrwxrwxrwx 1 root gpio    0 Jan 10 09:03 gpio18 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio18
	lrwxrwxrwx 1 root gpio    0 Jan 10 09:03 gpio22 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio22
	lrwxrwxrwx 1 root gpio    0 Jan 10 09:03 gpio23 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio23
	lrwxrwxrwx 1 root gpio    0 Jan 10 09:03 gpio24 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio24
	lrwxrwxrwx 1 root gpio    0 Jan 10 09:03 gpio25 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio25
	lrwxrwxrwx 1 root gpio    0 Jan 10 09:03 gpio27 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio27
	lrwxrwxrwx 1 root gpio    0 Jan 10 08:48 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0
	lrwxrwxrwx 1 root gpio    0 Jan 10 08:48 gpiochip100 -> ../../devices/platform/soc/soc:virtgpio/gpio/gpiochip100
	-rwxrwx--- 1 root gpio 4096 Jan 10 08:48 unexport

Now tried fresh install:
Copied Noob V2.1.0 to newly formatted microSD Card

Powered up and installed Raspian OS

Copied app.js to the Raspberry Pi 3

- ls -l /sys/class/gpio/
	total 0
	-rwxrwx--- 1 root gpio 4096 Jan 10 09:54 export
	lrwxrwxrwx 1 root gpio    0 Jan 10 09:54 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0
	lrwxrwxrwx 1 root gpio    0 Jan 10 09:54 gpiochip100 -> ../../devices/platform/soc/soc:virtgpio/gpio/gpiochip100
	-rwxrwx--- 1 root gpio 4096 Jan 10 09:54 unexport

- gpio edge gpio4 rising
- ls -l /sys/class/gpio/
	total 0
	-rwxrwx--- 1 root gpio 4096 Jan 10 10:00 export
	lrwxrwxrwx 1 root gpio    0 Jan 10 10:00 gpio0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpio0
	lrwxrwxrwx 1 root gpio    0 Jan 10 09:54 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0
	lrwxrwxrwx 1 root gpio    0 Jan 10 09:54 gpiochip100 -> ../../devices/platform/soc/soc:virtgpio/gpio/gpiochip100
	-rwxrwx--- 1 root gpio 4096 Jan 10 09:54 unexport

nebrius, 2017-01-11:
Hmmm, that doesn't look like the file structure I would have expected. Can you try running with Raspbian instead of Noobs and see if that fixes it?

alancairns, 2017-01-11:
This is the procedure I followed to install and run Raspian

- downloaded 2017-01-11-raspian-jessie.img
- used ApplePi-Baker to restore 2017-01-11-raspian-jessie.img to microSD card
- powered on Raspberry Pi 3
- Raspberry Pi Configuration | Interfaces
         Enabled: Camera, SSH, VNC, SPI, I2C, Serial, 1-Wire
- sudo apt-get update
- sudo apt-get upgrade
- update-node.js-and-node-red
- sudo npm install johnny-five raspi-io

====================output from sudo npm install johnny-five raspi-io ==================
> raspi-soft-pwm@2.0.2 preinstall /home/pi/node_modules/raspi-io/node_modules/raspi-soft-pwm
> sh install-pigpio.sh

pigpio is already installed, skipping

 
> raspi-pwm@3.1.2 install /home/pi/node_modules/raspi-io/node_modules/raspi-pwm
> node-gyp rebuild

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/4.7.2"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/tmp/.node-gyp"
make: Entering directory '/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/build'
  CXX(target) Release/obj.target/addon/src-native/addon.o
  CXX(target) Release/obj.target/addon/src-native/init.o
  CXX(target) Release/obj.target/addon/src-native/write.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory '/home/pi/node_modules/raspi-io/node_modules/raspi-pwm/build'
npm WARN cannot run in wd raspi@3.0.4 node-gyp rebuild (wd=/home/pi/node_modules/raspi-io/node_modules/raspi)

> raspi-gpio@3.1.0 install /home/pi/node_modules/raspi-io/node_modules/raspi-gpio
> node-gyp rebuild

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/4.7.2"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/tmp/.node-gyp"
make: Entering directory '/home/pi/node_modules/raspi-io/node_modules/raspi-gpio/build'
  CXX(target) Release/obj.target/addon/src-native/addon.o
  CXX(target) Release/obj.target/addon/src-native/init.o
  CXX(target) Release/obj.target/addon/src-native/listener.o
  CXX(target) Release/obj.target/addon/src-native/write.o
  CXX(target) Release/obj.target/addon/src-native/read.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory '/home/pi/node_modules/raspi-io/node_modules/raspi-gpio/build'

> pigpio@0.3.4 install /home/pi/node_modules/raspi-io/node_modules/raspi-soft-pwm/node_modules/pigpio
> node-gyp rebuild

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/4.7.2"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/tmp/.node-gyp"
make: Entering directory '/home/pi/node_modules/raspi-io/node_modules/raspi-soft-pwm/node_modules/pigpio/build'
  CXX(target) Release/obj.target/pigpio/src/pigpio.o
  SOLINK_MODULE(target) Release/obj.target/pigpio.node
  COPY Release/pigpio.node
make: Leaving directory '/home/pi/node_modules/raspi-io/node_modules/raspi-soft-pwm/node_modules/pigpio/build'

> i2c-bus@1.2.0 install /home/pi/node_modules/raspi-io/node_modules/raspi-i2c/node_modules/i2c-bus
> node-gyp rebuild

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/4.7.2"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/tmp/.node-gyp"
make: Entering directory '/home/pi/node_modules/raspi-io/node_modules/raspi-i2c/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory '/home/pi/node_modules/raspi-io/node_modules/raspi-i2c/node_modules/i2c-bus/build'

[... 278 lines cut ...]

modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not open moddep file '/lib/modules/4.4.34-v7+/modules.dep.bin'
child_process.js:508
    throw err;
    ^

Error: Command failed: modprobe i2c-dev
modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not open moddep file '/lib/modules/4.4.34-v7+/modules.dep.bin'

    at checkExecSyncError (child_process.js:465:13)
    at Object.execSync (child_process.js:505:13)
    at new I2C (/home/pi/node_modules/raspi-io/node_modules/raspi-i2c/dist/index.js:112:25)
    at new RaspiIOCore (/home/pi/node_modules/raspi-io/node_modules/raspi-io-core/dist/index.js:218:14)
    at new RaspiIO (/home/pi/node_modules/raspi-io/dist/index.js:26:10)
    at Object.<anonymous> (/home/pi/blink.js:4:7)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)

============ output from sudo node blink.js ================

I hope this isn't too longwinded but I'm trying to provide complete info.

Thanks again

nebrius, 2017-01-11:
Did you reboot your Pi after installing raspi-io?

alancairns, 2017-01-12:
Sorry, I had forgotten to reboot. However, after rebooting I got a similar error so decided to start again from fresh and follow the instructions from https://github.com/nebrius/raspi-io/wiki/Getting-a-Raspberry-Pi-ready-for-NodeBots

- Downloaded 2017-01-11-raspian-jessie.img
- ApplePi-Baker to restore this img file to the microSD card
- sudo apt-get upgrade
- sudo apt-get update

- wget https://nodejs.org/dist/v6.9.4/node-v6.9.4-linux-arm64.tar.xz
- tar -xf node-v6.9.4-linux-arm64.tar.xz
- sudo mv node-v6.9.4-linux-arm64.tar.xz /usr/local/node
- cd /usr/local/bin
- sudo ln -s /usr/local/node/bin/node node
- sudo ln -s /usr/local/node/bin/npm npm
- node -v
	bash: ./node: cannot execute binary file: Exec format error

I repeated the above but still got the same error so I started again.

This appears to be a known problem, see http://raspberrypi.stackexchange.com/questions/45319/install-newer-node-version-on-pi-3 so I followed the suggestion to use the curl command.

- Downloaded 2017-01-11-raspian-jessie.img
- ApplePi-Baker to restore this img file to the microSD card
- sudo apt-get upgrade
- sudo apt-get update

- curl -sL https://deb.nodesource.com/setup_6.x | sudo -E bash -
- sudo apt-get install -y nodejs
- sudo apt-get install -y build-essential
	Reading package lists... Done
	Building dependency tree       
	Reading state information... Done
	build-essential is already the newest version.
	The following packages were automatically installed and are no longer required:
  	libc-ares2 libv8-3.14.5
	Use 'apt-get autoremove' to remove them.
	0 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.
- node -v
	v6.9.4
- npm -v
	3.10.10

- mkdir node-projects
- cd node-projects
- npm install johnny-five raspi-io
- sudo reboot

- created index.js
var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi()
});

board.on('ready', function() {
  // Create an Led on pin 7 on P1 (GPIO4)
  // and strobe it on/off
  var led = new five.Led('P1-7');
  led.strobe(500);
});

- copied index.js from Mac to Raspberry pi ~/node-projects
- cd node-projects
- sudo node index.js
	1484234851323 Device(s) RaspberryPi-IO  
	1484234851363 Connected RaspberryPi-IO  
	1484234851376 Repl Initialized  
	LED blinks 9 times the stops and the ‘ node command prompt” appears
	>>
I exited node using .exit then repeated the sudo node index.js command a number of times. The result was inconsistent with the LED blinking randomly between 0 and 9 times. I'm trying to use johnny-five for the first time and assume that the led is meant blink constantly.

=========================================

Re-install and upgrade node and npm using node-red upgrade instructions

- Downloaded 2017-01-11-raspian-jessie.img
- ApplePi-Baker to restore this img file to the microSD card
- sudo apt-get upgrade
- sudo apt-get update

- node -v
	v0.10.29
- npm -v
	npm: command not found

- update-nodejs-and-nodered
- node -v
	v4.7.2
- npm -v
	2.15.11
-
 mkdir ~/node-projects
- cd ~/node-projects
- npm install johnny-five raspi-io
- sudo shutdown -h now
- power off/on

- copied index.js from Mac to Raspberry pi ~/node-projects
- cd node-projects
- sudo node index.js

I got exactly the same outcome. I exited node using .exit then repeated the sudo node index.js command a number of times. Again the result was inconsistent with the LED blinking randomly between 0 and 9 times

nebrius, 2017-01-12:
It is expected that you will get a REPL (the proper name for the Node.js command prompt). That said, it should blink consistently, assuming that there isn't high load on the CPU.

Looking at your other instructions, I'm not exactly sure on the details of what's going on, but I suspect that the `sudo apt-get install -y nodejs` line is not necessary.

Try going back to the instructions I posted, except try using the armv7 version instead of the armv8 version and see what happens.

alancairns, 2017-01-13:
Using a fresh install of 2017-01-11-raspian-jessie.img I’ve followed the instructions you posted except to use the armv7 version instead of the armv8 version. Node and npm worked.

The node version is v6.9.4 and the npm version is 3.10.10

Using sudo node index.js repeatedly, it did the same thing as the previous working versions above i.e. LED blinks a random number of times then stops with the LED off.

I connected and Arduino Uno to the Raspberry Pi 3 with a usb cable and installed the StandardFirmata sketch then used the following ard-blink.js file:

var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var led = new five.Led(13);

  this.repl.inject({
    led: led
  });

  led.strobe(1500);
});

This worked as expected, the LED continued to blink for 1.5 seconds and I was returned to a REPL prompt.

I then tried it on the Raspberry Pi 3 with the installation that I upgraded node using ‘update-nodejs-and-nodered’

The node version on this install is v4.7.2 and the npm version is 2.15.11. This was exactly the same, using index.js to blink the LED connected to the Pi, the LED blinks a random number of times then stops with the LED off. On the Arduino the LED continued to blink for 1.5 seconds.

This would seem to point to raspi-io or one or more of its dependencies.

nebrius, 2017-01-13:
I ran your code on a Raspberry Pi 3 with a fresh install of 2017-01-11-raspbian-jessie and node.js node-v6.9.4-linux-armv7l downloaded from the node.js website and it worked fine. There is some configuration issue on your end, it's not in raspi-io itself.

> This worked as expected, the LED continued to blink for 1.5 seconds and I was returned to a REPL prompt.

The behavior you describe here is _not_ what is expected. What is expected is that the LED will blink _every_ 1.5 seconds indefinitely. When the REPL comes up, you are not being _returned_ to it, that is a sign that your program is up and running. The REPL runs at the same time as your application, not after the application is finished running.

So let's go back again, when you install everything, did you run `sudo apt-get` for anything? I did not use that command once, and I recommend you try doing that just so we have a common baseline, combined with the _exact_ version of Raspbian and the _exact_ version of Node (once again, _do not_ run `sudo apt-get`).

alancairns, 2017-01-14:
I really appreciate the help you're giving me.  I think I've tracked down the issue and you are correct to say that it is a configuration issue.  Here's what I've done:

- downloaded 2017-01-11-raspian-jessie.img
- used ApplePi-Baker to restore 2017-01-11-raspian-jessie.img to microSD card
- disconnected touch screen and connect monitor
- connected ethernet cable
- powered on Raspberry Pi 3

- wget https://nodejs.org/dist/v6.9.4/node-v6.9.4-linux-armv7l.tar.xz
- tar -xf node-v6.9.4-linux-armv7l.tar.xz
- sudo mv node-v6.9.4-linux-armv7l /usr/local/node
- cd /usr/local/bin
- sudo ln -s /usr/local/node/bin/node node
- sudo ln -s /usr/local/node/bin/npm npm
- cd ~
- mkdir node-projects
- cd node-projects
- npm install johnny-five raspi-io
- shutdown

- powered on Raspberry Pi 3
- checked Raspberry Pi Preferences | Interfaces, only I2C enabled
- create file index.js
var raspi = require('raspi-io');
var five = require('johnny-five');
var board = new five.Board({
  io: new raspi()
});

board.on('ready', function() {
  // Create an Led on pin 7 on P1 (GPIO4)
  // and strobe it on/off
  var led = new five.Led('P1-7');
  led.strobe(500);
});

- cd node-projects
- sudo node index.js

This worked!!

- Raspberry Pi Preferences | Interfaces | Camera Enabled (I2C was already enabled)
- reboot This worked

- Raspberry Pi Preferences | Interfaces | SSH Enabled (reboot not required)
- connected to Pi using SSH from now on
- This worked

- Raspberry Pi Preferences | Interfaces | VNC Enabled (reboot not required)
- This worked

- Raspberry Pi Preferences | Interfaces | SPI Enabled (reboot not required)
- This worked

- Raspberry Pi Preferences | Interfaces | Serial Enabled
- reboot This worked

- Raspberry Pi Preferences | Interfaces | 1-Wire Enabled
- reboot This failed

Repeatedly enabled/disabled 1-wire. Worked when 1-wire disabled, failed when 1-wire enabled

- sudo apt-get update
- sudo apt-get upgrade

***Did not reboot

- sudo node index.js
modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not open moddep file '/lib/modules/4.4.34-v7+/modules.dep.bin'
child_process.js:526
    throw err;
    ^

Error: Command failed: modprobe i2c-dev
modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not open moddep file '/lib/modules/4.4.34-v7+/modules.dep.bin'

    at checkExecSyncError (child_process.js:483:13)
    at Object.execSync (child_process.js:523:13)
    at new I2C (/home/pi/node-projects/node_modules/raspi-i2c/dist/index.js:112:25)
    at new RaspiIOCore (/home/pi/node-projects/node_modules/raspi-io-core/dist/index.js:218:14)
    at new RaspiIO (/home/pi/node-projects/node_modules/raspi-io/dist/index.js:26:10)
    at Object.<anonymous> (/home/pi/node-projects/index.js:4:7)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)

- sudo reboot

sudo node index.js now worked if 1-Wire enabled, failed if 1-wire enabled

I normally enable everything except Remote GPIO in the Preferences as I use this Raspberry Pi for experimentation, although at the moment I don't require 1-Wire. With this disabled I can now try to learn to use johnny-five following the excellent book I recently bought 'Javascript Robotics: Building NodeBots with Johnny-Five, Raspberry Pi, Arduino and BeagleBone'

Once again thank you so much for the excellent support you've give me.

nebrius, 2017-01-14:
That explains it! 1-wire i not supported by raspi-io. Glad you got it working!

nebrius, 2017-01-22:
Closing since this was resolved. Feel free to reopen if this issue still persists.

closed 2017-01-22 by nebrius