nebrius/raspi-io #82: version mismatch expected 47, got 51

BHEADRICK, 2017-01-12:
I keep getting this error when trying to run anything using this. I've tried reinstalling node by downloading the binaries from nodejs.org as well as by using nvm, but to no avail

module.js:440
  return process.dlopen(module, path._makeLong(filename));
                 ^

Error: Module version mismatch. Expected 47, got 51.
    at Error (native)
    at Object.Module._extensions..node (module.js:440:18)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Module.require (module.js:367:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/nodeserver/node_modules/raspi/dist/index.js:29:13)
    at Module._compile (module.js:413:34)
    at Object.Module._extensions..js (module.js:422:10)
    at Module.load (module.js:357:32)

nebrius, 2017-01-12:
Have you tried `rm -r`ing your `node_modules` directory and then `npm install`ing again? This error happens when there are native modules that are compiled for one version of Node, but run against another version.

nebrius, 2017-04-05:
Closing this issue due to inactivity. If this is still an issue for you, feel free to reopen it!

closed 2017-04-05 by nebrius