nebrius/raspi-io #83: Exception thrown: No such file or directory, cannot open /dev/ttyAMA0

mwittig, 2017-01-18:
When I run the example included in the README I get the following exception:
```
1484694086366 Device(s) RaspberryPi-IO
1484694086409 Connected RaspberryPi-IO
1484694086422 Repl Initialized
>> events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: Error: No such file or directory, cannot open /dev/ttyAMA0
    at Error (native)
```

Actually I have provoked this error by removing /dev/ttyAMA0 to find an error which occurs in one of my projects. My questions are:

1) Is it possible to use raspi-io without having /dev/ttyAMA0 in place, e.g. by using a different serial device, or by deactivating serial device usage completely?

2) How can the error be handled gracefully, as it seems to happen after board creation? Putting the whole code in a try/catch block is not really an option for me.

3) Looking at the console output it does not seem possible to disable repl when using raspi-io. What do I need to do to make sure that repl is disabled? I have set "repl" to false as part of the board options (see below).

Here is a slightly modfied version of the sample code from README I have used for testing

```
var raspi = require('raspi-io');
var five = require('johnny-five');
var board;

try {
  board = new five.Board({
    io: new raspi(),
    repl: false,
  })
}
catch (error) {
  console.log("Ooops", error);
}

board.on('ready', function() {

  try {
    // Create an Led on pin 7 (GPIO4) on P1 and strobe it on/off
    // Optionally set the speed; defaults to 100ms
    (new five.Led('P1-7')).strobe();
  }
  catch  (error) {
    console.log("Ooops", error);
  }

});

board.on('error', function() {
  console.log("Ooops", error);
});
```

nebrius, 2017-01-18:
Hi @mwittig! Unfortunately it is not currently possible to disable the serial port or switch the serial device at this time.

mwittig, 2017-01-19:
Hi @nebrius, thanks for swift reply! How about question 2? 

The exception occurs after the board has been initialized (ready event has been emitted). It looks like the initalization of raspi-io is completed too early while the sub-components initialization has not been completed.

nebrius, 2017-01-19:
Unfortunately this error is not a recoverable error, even if it does happen after the "ready" event.

If you take a look at the [serial initialization code](https://github.com/nebrius/raspi-io-core/blob/master/src/index.js#L741-L749), you'll see that opening the port is thrown into an operation queue, which all serial operations are put into. This was done because Johnny-Five may reconfigure the serial port mid-execution, which requires re-opening the port. The behavior you're seeing is by design.

mwittig, 2017-01-19:
Thanks for the reply. I should have a look at your code in more detail!

However, may be you can answer the following quickly: Wouldn't it be possible to defer the io-plugin' ready state until raspi-serial has processed the serialport initialization as part of the queued operations, or does the serialport initialization happen implicitly when the first read or write operation is dequeued?

nebrius, 2017-01-19:
> Wouldn't it be possible to defer the io-plugin' ready state until raspi-serial has processed the serialport initialization as part of the queued operations, or does the serialport initialization happen implicitly when the first read or write operation is dequeued?

It's _possible_ because the `serialConfig` method is [called in the initialization code](https://github.com/nebrius/raspi-io-core/blob/master/src/index.js#L425), but it would complicate the code more than I'd prefer, and so I don't think it's worth the effort TBH. Aside from timing of this exception that's thrown (which you couldn't catch in a try-catch anyways even if it occurred before `ready`), the system behaves correctly.

mwittig, 2017-01-19:
I'll document this as a known issue as part of my project. It is an edge case anyway as most users run the software as root. The problem only occurs if access is denied for an unprivileged user or if the device does not exist for whatever reason. I may revisit the matter later and come up with change proposal for your project, but at the moment I am far too busy with other things.

nebrius, 2017-01-19:
Ahh, I see. As it turns out, root access is required for raspi-io anyways. Running as non-root is not supported, as GPIO/PWM access requires root privileges.

mwittig, 2017-01-20:
> Running as non-root is not supported, as GPIO/PWM access requires root privileges.

Well, I think that's not exactly right as the user reporting to me actually runs my project in user mode.  It is possible to use wiringPi in user mode with a few tweaks. See also 
* http://wiringpi.com/reference/setup/ 
* http://wiringpi.com/wiringpi-update-to-2-29/
* and and https://www.raspberrypi.org/forums/viewtopic.php?f=32&t=32177

nebrius, 2017-01-20:
I'm aware of that approach, and it's _definitely_ not supported by raspi-io. It's possible that some users may get it to work, but it's not supported, meaning that I won't fix bugs that come up as a result of trying.

mwittig, 2017-01-20:
Ok, that's fine. Thank you very much again for your support! I think this ticket can be closed now unless you want to take further action on this.

nebrius, 2017-01-20:
Thanks for filing the issue!

closed 2017-01-20 by nebrius