nebrius/raspi-io #84: using temperature sensors

pepemarino, 2017-01-23:
Got Raspberry Pi 3 recently and I am now testing some basics.  I am running this code to the temp sensor:

var five = require("johnny-five"),
    Raspi = require("raspi-io"),
    board = new five.Board({
      io: new Raspi()
    });    

console.log("started...");

board.on("ready", () => {
    console.log("inside of ready")
    
    var temperature = new five.Thermometer({
      controller: "TMP36",
      pin: 'P1-7'
    });

    temperature.on("change", () => {
      console.log(this.celsius + "°C", this.fahrenheit + "°F");
    });
});

however this is the error I am getting:

pi@raspberrypi:~/projects $ sudo node raspi-io.js                               1485155355258 Device(s) RaspberryPi-IO
started...
1485155355292 Connected RaspberryPi-IO
1485155355303 Repl Initialized
>> inside of ready
/home/pi/projects/node_modules/raspi-io-core/dist/index.js:503
        throw new Error('Pin "' + pin + '" does not support mode "' + mode + '"');
        ^

Error: Pin "7" does not support mode "2"
    at RaspiIOCore.value (/home/pi/projects/node_modules/raspi-io-core/dist/index.js:503:15)
    at RaspiIOCore.pinMode (/home/pi/projects/node_modules/raspi-io-core/dist/index.js:484:21)
    at Thermometer.analogHandler (/home/pi/projects/node_modules/johnny-five/lib/thermometer.js:14:11)
    at new Thermometer (/home/pi/projects/node_modules/johnny-five/lib/thermometer.js:980:10)
    at Board.board.on (/home/pi/projects/raspi-io.js:16:23)
    at emitNone (events.js:91:20)
    at Board.emit (events.js:185:7)
    at _combinedTickCallback (internal/process/next_tick.js:67:7)
    at process._tickDomainCallback (internal/process/next_tick.js:122:9)

Any idea what could be missing?

Cheers!

nebrius, 2017-01-23:
Hi @pepemarino! Unfortunately the TMP36 is an analog sensor, and the Raspberry Pi does not have any analog inputs on it. You'll need to use a different sensor that uses I2C to communicate, such as the [SparkFun TMP102](https://www.sparkfun.com/products/11931) instead.

pepemarino, 2017-01-23:
Thank you Nebrius!  On my way to Amazon!  :)

nebrius, 2017-01-23:
Best of luck!

closed 2017-01-23 by nebrius