nebrius/raspi-io #85: wiringPiSetup: Must be root. (Did you forget sudo?)

alex-def, 2017-02-14:
Raspberry Pi Zero
Raspbian: Linux raspberrypi 4.4.38+ #938 Thu Dec 15 15:17:54 GMT 2016 armv6l GNU/Linux
node: v6.9.5
npm:3.10.10

Hello,
First when i tried to install :    sudo npm install -g raspi-io --unsafe-perm --force 
I've got following error :
```
make: Entering directory '/usr/lib/node_modules/raspi-io/node_modules/raspi/build'
  CXX(target) Release/obj.target/addon/src-native/addon.o
  CXX(target) Release/obj.target/addon/src-native/init.o
../src-native/init.cc:25:22: fatal error: wiringPi.h: No such file or directory
 #include <wiringPi.h>
```
To solve this issue i've installed : 
```
sudo apt-get install wiringpi
sudo npm install -g raspi-wiringpi --unsafe-perm --force
```

Then i tried to use it with johnny-five (installed by : sudo npm install -g johnny-five --unsafe-perm --force) over node-red i ve got the following blocking error:

**wiringPiSetup: Must be root. (Did you forget sudo?)**

```
Welcome to Node-RED
===================
14 Feb 23:23:28 - [info] Node-RED version: v0.16.2
14 Feb 23:23:28 - [info] Node.js  version: v6.9.5
14 Feb 23:23:28 - [info] Linux 4.4.38+ arm LE
14 Feb 23:23:33 - [info] Loading palette nodes
pi : TTY=unknown ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/python -u /usr/lib/node_modules/node-red/nodes/core/hardware/nrgpiopy info
pam_unix(sudo:session): session opened for user root by (uid=0)
pam_unix(sudo:session): session closed for user root
14 Feb 23:24:01 - [warn] ------------------------------------------------------
14 Feb 23:24:01 - [warn] [sensehat] Error: Can't find Sense HAT python libraries. Run sudo apt-get install sense-hat
14 Feb 23:24:01 - [warn] ------------------------------------------------------
14 Feb 23:24:01 - [info] Settings file  : /home/pi/.node-red/settings.js
14 Feb 23:24:01 - [info] User directory : /home/pi/.node-red
14 Feb 23:24:01 - [info] Flows file     : /home/pi/.node-red/flows_raspberrypi.json
14 Feb 23:24:01 - [info] Server now running at http://127.0.0.1:1880/
14 Feb 23:24:01 - [debug] loaded flow revision: 51859858b072360ca45c68d82fd5d322
14 Feb 23:24:01 - [debug] red/runtime/nodes/credentials.load : no user key present
14 Feb 23:24:01 - [debug] red/runtime/nodes/credentials.load : using default key
14 Feb 23:24:01 - [info] Starting flows
wiringPiSetup: Must be root. (Did you forget sudo?)
nodered.service: main process exited, code=killed, status=6/ABRT
Unit nodered.service entered failed state.
nodered.service holdoff time over, scheduling restart.
Stopping Node-RED graphical event wiring tool....
Starting Node-RED graphical event wiring tool....
Started Node-RED graphical event wiring tool..
```

Did you have an idea ?

nebrius, 2017-02-15:
A couple of notes:

1. `sudo npm install -g raspi-wiringpi --unsafe-perm --force` is not necessary, and may even cause issues. You should only need to do `sudo apt-get install wiringpi`
1. You should not install johnny-five globally, or as root. Instead, you should run `npm install johnny-five raspi-io` from your project directory.
1. When you run the program, it must be run as root. Normally this is done via `sudo node myapp.js`, but I see you're running Node-Red, so it's different. I don't know much about Node-Red, so you'll need to consult the Node-Red documentation. FWIW, this error is coming from Wiring Pi itself, not from raspi-io.

alex-def, 2017-02-15:
ok thks for help :)

closed 2017-02-17 by alex-def