nebrius/raspi-io #87: Unable to install raspi-io

shuynh84, 2017-03-27:
I'm having problem installing raspi-io on the raspberry pi 3 Model B. After that node red runs in a continuous loop and won't load. Below are my errors:

@raspberrypi:~ $ cd ~/.node-red
pi@raspberrypi:~/.node-red $ npm install raspi-io

> raspi-soft-pwm@2.0.2 preinstall /home/pi/.node-red/node_modules/.staging/raspi-soft-pwm-bb68b3a9
> sh install-pigpio.sh

pigpio is already installed, skipping

> i2c-bus@1.2.1 install /home/pi/.node-red/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory '/home/pi/.node-red/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory '/home/pi/.node-red/node_modules/i2c-bus/build'

> pigpio@0.3.4 install /home/pi/.node-red/node_modules/pigpio
> node-gyp rebuild

make: Entering directory '/home/pi/.node-red/node_modules/pigpio/build'
  CXX(target) Release/obj.target/pigpio/src/pigpio.o
  SOLINK_MODULE(target) Release/obj.target/pigpio.node
  COPY Release/pigpio.node
make: Leaving directory '/home/pi/.node-red/node_modules/pigpio/build'

> usb@1.2.0 install /home/pi/.node-red/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v48-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@6.10.1 (node-v48 ABI) (falling back to source compile with node-gyp) 
make: Entering directory '/home/pi/.node-red/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
../libusb/libusb/os/linux_udev.c:40:21: fatal error: libudev.h: No such file or directory
 #include <libudev.h>
                     ^
compilation terminated.
libusb.target.mk:131: recipe for target 'Release/obj.target/libusb/libusb/libusb/os/linux_udev.o' failed
make: *** [Release/obj.target/libusb/libusb/libusb/os/linux_udev.o] Error 1
make: Leaving directory '/home/pi/.node-red/node_modules/usb/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.50-v7+
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/.node-red/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/home/pi/.node-red/node_modules/usb/src/binding"
gyp ERR! cwd /home/pi/.node-red/node_modules/usb
gyp ERR! node -v v6.10.1
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/.node-red/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/pi/.node-red/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/.node-red/node_modules/usb/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:886:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.50-v7+
node-pre-gyp ERR! command "/usr/bin/nodejs" "/home/pi/.node-red/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/.node-red/node_modules/usb
node-pre-gyp ERR! node -v v6.10.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.30
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/.node-red/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/pi/.node-red/node_modules/usb/src/binding' (1)

> raspi@3.0.5 install /home/pi/.node-red/node_modules/raspi
> node-gyp rebuild

make: Entering directory '/home/pi/.node-red/node_modules/raspi/build'
  CXX(target) Release/obj.target/addon/src-native/addon.o
  CXX(target) Release/obj.target/addon/src-native/init.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory '/home/pi/.node-red/node_modules/raspi/build'

> raspi-gpio@3.1.1 install /home/pi/.node-red/node_modules/raspi-gpio
> node-gyp rebuild

make: Entering directory '/home/pi/.node-red/node_modules/raspi-gpio/build'
  CXX(target) Release/obj.target/addon/src-native/addon.o
  CXX(target) Release/obj.target/addon/src-native/init.o
  CXX(target) Release/obj.target/addon/src-native/listener.o
  CXX(target) Release/obj.target/addon/src-native/write.o
  CXX(target) Release/obj.target/addon/src-native/read.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory '/home/pi/.node-red/node_modules/raspi-gpio/build'

> raspi-pwm@3.1.3 install /home/pi/.node-red/node_modules/raspi-pwm
> node-gyp rebuild

make: Entering directory '/home/pi/.node-red/node_modules/raspi-pwm/build'
  CXX(target) Release/obj.target/addon/src-native/addon.o
  CXX(target) Release/obj.target/addon/src-native/init.o
  CXX(target) Release/obj.target/addon/src-native/write.o
  SOLINK_MODULE(target) Release/obj.target/addon.node
  COPY Release/addon.node
make: Leaving directory '/home/pi/.node-red/node_modules/raspi-pwm/build'

> raspi-i2c@5.0.3 postinstall /home/pi/.node-red/node_modules/raspi-i2c
> echo "Enabling I2C at boot time, you may be asked for your password" && sudo env "PATH=$PATH" script/enable_i2c.js

Enabling I2C at boot time, you may be asked for your password
Checking if I2C is enabled at boot time
I2C is already enabled at boot time

> raspi-serial@3.0.5 postinstall /home/pi/.node-red/node_modules/raspi-serial
> echo "Disabling serial port login at boot time, you may be asked for your password" && sudo env "PATH=$PATH" script/enable_serial.js

Disabling serial port login at boot time, you may be asked for your password
Checking if serial port login is enabled
Serial port login does not appear to be enabled, skipping
/home/pi/.node-red
└─┬ raspi-io@7.2.4 
  ├── raspi@3.0.5 
  ├─┬ raspi-board@4.0.2 
  │ └── @types/node@0.0.2 
  ├─┬ raspi-gpio@3.1.1 
  │ └── raspi-peripheral@1.6.4 
  ├─┬ raspi-i2c@5.0.3 
  │ ├── i2c-bus@1.2.1 
  │ └── ini-builder@1.0.3 
  ├── raspi-io-core@1.0.2 
  ├── raspi-led@1.5.5 
  ├── raspi-pwm@3.1.3 
  ├── raspi-serial@3.0.5 
  └─┬ raspi-soft-pwm@2.0.2 
    └─┬ pigpio@0.3.4 
      └── nan@2.3.5 

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@~0.1.4 (node_modules/node-red-node-sensortag/node_modules/sensortag/node_modules/noble-device/node_modules/noble/node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/.node-red/package.json'
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/.node-red/node_modules/build/package.json'
npm WARN .node-red No description
npm WARN .node-red No repository field.
npm WARN .node-red No README data
npm WARN .node-red No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.2.0 (node_modules/usb):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.2.0 install: `node-pre-gyp install --fallback-to-build`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

nebrius, 2017-03-27:
Here's the relevant part that's failing:

```
node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v48-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@6.10.1 (node-v48 ABI) (falling back to source compile with node-gyp)
make: Entering directory '/home/pi/.node-red/node_modules/usb/build'
CC(target) Release/obj.target/libusb/libusb/libusb/core.o
CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
CC(target) Release/obj.target/libusb/libusb/libusb/io.o
CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
../libusb/libusb/os/linux_udev.c:40:21: fatal error: libudev.h: No such file or directory
#include <libudev.h>
^
compilation terminated.
libusb.target.mk:131: recipe for target 'Release/obj.target/libusb/libusb/libusb/os/linux_udev.o' failed
make: *** [Release/obj.target/libusb/libusb/libusb/os/linux_udev.o] Error 1
make: Leaving directory '/home/pi/.node-red/node_modules/usb/build'
```

It's an issue with `node-usb`, although I'm not sure why that's happening because AFAIK it's not a part of the raspi-io dependency chain. Do you have `node-usb` as a dependency in your project?

shuynh84, 2017-03-28:
I'm not really sure. I'm trying to import from a flow i see on node-red called "**Read an i2c-based mpu6050 accelerometer on a raspberry pi**" to read from an accelerometer. I'm getting an error, "Error: Cannot find module 'raspi-io'" so i tried to download the raspi-io. I'm using the Raspberry Pi 3 Version B. I'm pretty new to this but this is my flow:

[{"id":"1d335403.e0b33c","type":"tab","label":"MMS"},{"id":"69c2f3b9.18340c","type":"debug","z":"1d335403.e0b33c","name":"","active":true,"console":"false","complete":"payload","x":886,"y":297,"wires":[]},{"id":"7c2a8e5b.9030f","type":"ds18b20","z":"1d335403.e0b33c","name":"","sensorid":"28-051684d20cff","timer":"10","x":378,"y":200,"wires":[["98567958.dc8d38"]]},{"id":"2dfe1ab9.482686","type":"ui_gauge","z":"1d335403.e0b33c","name":"Temperature","group":"79b1253a.99a93c","order":0,"width":0,"height":0,"gtype":"gage","title":"Temperature Sensor","label":"Fahrenheit","format":"{{value}}","min":0,"max":"100","colors":["#00b500","#e6e600","#ca3838"],"seg1":"80","seg2":"90","x":895,"y":105,"wires":[]},{"id":"98567958.dc8d38","type":"function","z":"1d335403.e0b33c","name":"Convert","func":"msg.payload = Math.round(msg.payload *(9/5) +32);\nreturn msg;","outputs":1,"noerr":0,"x":628,"y":190,"wires":[["69c2f3b9.18340c","2dfe1ab9.482686"]]},{"id":"47b06062.ad972","type":"pimcp3008","z":"1d335403.e0b33c","name":"Current","pin":"0","dnum":"0","x":411,"y":503,"wires":[["86bcc844.4d57f8","c1eda058.74096"]]},{"id":"86bcc844.4d57f8","type":"debug","z":"1d335403.e0b33c","name":"","active":true,"console":"false","complete":"payload","x":627,"y":617,"wires":[]},{"id":"936bd36f.eb529","type":"inject","z":"1d335403.e0b33c","name":"","topic":"","payload":"","payloadType":"date","repeat":"30","crontab":"","once":false,"x":200,"y":499,"wires":[["47b06062.ad972"]]},{"id":"c1eda058.74096","type":"ui_chart","z":"1d335403.e0b33c","name":"Current","group":"79b1253a.99a93c","order":0,"width":"6","height":"5","label":"Current Sensor","chartType":"line","legend":"false","xformat":"HH:mm:ss","interpolate":"linear","nodata":"","ymin":"400","ymax":"600","removeOlder":"10","removeOlderPoints":"10","removeOlderUnit":"60","cutout":0,"colors":["#1f77b4","#aec7e8","#ff7f0e","#2ca02c","#98df8a","#d62728","#ff9896","#9467bd","#c5b0d5"],"x":626,"y":466,"wires":[[],[]]},{"id":"e42faae4.fa9548","type":"gpio out","z":"1d335403.e0b33c","name":"","state":"I2C_READ_REQUEST","pin":"","i2cDelay":"0","i2cAddress":"104","i2cRegister":"59","outputs":1,"board":"ad7d7810.e680a8","x":513,"y":967,"wires":[["94b8bfec.ece7c"]]},{"id":"262bd026.3ecca","type":"debug","z":"1d335403.e0b33c","name":"","active":true,"console":"false","complete":"false","x":730,"y":895,"wires":[]},{"id":"94b8bfec.ece7c","type":"function","z":"1d335403.e0b33c","name":"process i2c","func":"var data = msg.payload;\nvar int16 = function(high, low) {\n  var result = (high << 8) | low;\n  // if highest bit is on, it is negative\n  return result >> 15 ? ((result ^ 0xFFFF) + 1) * -1 : result;\n};\n\n\nmsg.payload = {\n    accelerometer : {\n        x: int16(data[0], data[1]),\n        y: int16(data[2], data[3]),\n        z: int16(data[4], data[5])\n    },\n    temperature : int16(data[6], data[7]),\n    gyro : {\n        x: int16(data[8], data[9]),\n        y: int16(data[10], data[11]),\n        z: int16(data[12], data[13])\n    }\n};\n\nreturn msg;","outputs":1,"x":545,"y":821,"wires":[["262bd026.3ecca"]]},{"id":"e4b723de.d8a7","type":"inject","z":"1d335403.e0b33c","name":"","topic":"","payload":"14","payloadType":"string","repeat":"","crontab":"","once":false,"x":349,"y":825,"wires":[["e42faae4.fa9548"]]},{"id":"b05604b3.1aae58","type":"gpio out","z":"1d335403.e0b33c","name":"","state":"I2C_WRITE_REQUEST","pin":"","i2cDelay":"0","i2cAddress":"104","i2cRegister":"","outputs":0,"board":"ad7d7810.e680a8","x":505,"y":1063,"wires":[]},{"id":"abb224bb.86bd68","type":"inject","z":"1d335403.e0b33c","name":"","topic":"","payload":"","payloadType":"none","repeat":"","crontab":"","once":false,"x":320,"y":906,"wires":[["8733d6f9.f20148"]]},{"id":"8733d6f9.f20148","type":"function","z":"1d335403.e0b33c","name":"i2c bytes","func":"msg.payload = [0x6B, 0x00];\nreturn msg;","outputs":1,"x":369,"y":982,"wires":[["b05604b3.1aae58"]]},{"id":"79b1253a.99a93c","type":"ui_group","z":"","name":"MMS Dashboard","tab":"79d6dbe9.e9a114","disp":true,"width":"6"},{"id":"ad7d7810.e680a8","type":"nodebot","name":"","username":"","password":"","boardType":"raspi-io","serialportName":"","connectionType":"local","mqttServer":"","pubTopic":"","subTopic":"","tcpHost":"","tcpPort":"","sparkId":"","sparkToken":"","beanId":"","impId":""},{"id":"79d6dbe9.e9a114","type":"ui_tab","z":"","name":"Home","icon":"dashboard"}]

shuynh84, 2017-03-28:
I think i got it to install but node red won't start, just keeps looping:

To find more nodes and example flows - go to http://flows.nodered.org
 
Started Node-RED graphical event wiring tool..
27 Mar 23:45:04 - [info] Dashboard version 2.3.6 started at /ui
27 Mar 23:45:06 - [info] Settings file  : /home/pi/.node-red/settings.js
27 Mar 23:45:06 - [info] User directory : /home/pi/.node-red
27 Mar 23:45:06 - [info] Flows file     : /home/pi/.node-red/flows_raspberrypi.json
27 Mar 23:45:06 - [warn] Communication server error: Error: listen EADDRINUSE 0.0.0.0:1880
27 Mar 23:45:06 - [error] Unable to listen on http://127.0.0.1:1880/
27 Mar 23:45:06 - [error] Error: port in use
nodered.service: main process exited, code=exited, status=1/FAILURE
Unit nodered.service entered failed state.
nodered.service holdoff time over, scheduling restart.
Stopping Node-RED graphical event wiring tool....
Starting Node-RED graphical event wiring tool....
Started Node-RED graphical event wiring tool..
27 Mar 23:45:09 - [info]
Welcome to Node-RED
===================
27 Mar 23:45:09 - [info] Node-RED version: v0.16.2
27 Mar 23:45:09 - [info] Node.js  version: v6.10.1
27 Mar 23:45:09 - [info] Linux 4.4.50-v7+ arm LE
27 Mar 23:45:10 - [info] Loading palette nodes
pi : TTY=unknown ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/python -u /usr/lib/node_modules/node-red/nodes/core/hardware/nrgpio.py info
pam_unix(sudo:session): session opened for user root by (uid=0)
pam_unix(sudo:session): session closed for user root
27 Mar 23:45:15 - [info] Dashboard version 2.3.6 started at /ui
27 Mar 23:45:17 - [info] Settings file  : /home/pi/.node-red/settings.js
27 Mar 23:45:17 - [info] User directory : /home/pi/.node-red
27 Mar 23:45:17 - [info] Flows file     : /home/pi/.node-red/flows_raspberrypi.json
27 Mar 23:45:17 - [warn] Communication server error: Error: listen EADDRINUSE 0.0.0.0:1880
27 Mar 23:45:17 - [error] Unable to listen on http://127.0.0.1:1880/
27 Mar 23:45:17 - [error] Error: port in use
nodered.service: main process exited, code=exited, status=1/FAILURE
Unit nodered.service entered failed state.
nodered.service holdoff time over, scheduling restart.
Stopping Node-RED graphical event wiring tool....
Starting Node-RED graphical event wiring tool....
Started Node-RED graphical event wiring tool..
27 Mar 23:45:20 - [info]

nebrius, 2017-03-28:
Ok yeah, looks like it's installed correctly. The next relevant error is:

```
28 Mar 00:20:37 - Error: EACCES: permission denied, open '/sys/class/leds/led0/trigger'
```

Are you running the app as root? Raspi IO needs to be run as root to function correctly.

shuynh84, 2017-03-28:
this is a dumb question because I'm new to raspberry pi. How would I do that? I usually click on Programming>>Node-Red. I'm using VNC viewer, and i've added **pi** to do the sudo groups.

nebrius, 2017-03-28:
> this is a dumb question because I'm new to raspberry pi. How would I do that? I usually click on Programming>>Node-Red. I'm using VNC viewer, and i've added pi to do the sudo groups.

I'm not sure, I'm not very familiar with Node-Red TBH. Normally you would do it from the command line with `sudo node foo.js`, but it sounds like you aren't running from the command line. Adding `pi` to the sudo groups isn't enough though, because all that does is _allow_ the `pi` user to run the `sudo` command. You'll need to check the Node-Red documentation to figure out how to run as root.

shuynh84, 2017-03-28:
ah ok, thank you for the help!

shuynh84, 2017-03-28:
i've tried doing sudo su to run as root, then node-red-start but still getting the same looping error. Does raspi-io have to be installed in the same folder as npm install node-red-contrib-gpio?

nebrius, 2017-03-28:
> Does raspi-io have to be installed in the same folder as npm install node-red-contrib-gpio?

I would assume so, but I'm not familiar with node-red-contrib-gpio, so I can't say for certain.

Something else to try, can you run `sudo ls /sys/class/leds/led0/trigger` and make sure that file exists?

shuynh84, 2017-03-28:
this is what it returns:

/sys/class/leds/led0/trigger

shuynh84, 2017-03-29:
pi@raspberrypi:~ $ /sys/class/leds/led0/trigger
bash: /sys/class/leds/led0/trigger: Permission denied

I'm not sure why this is happening

nebrius, 2017-03-29:
Yeah, I dunno then. My best guess is that your app still isn't being run as root, but maybe there's something else going on?

Ok, try this. At the top of your main JavaScript file, add this line:

```
console.log(`process user: ${process.env.USER}`);
```

Then report back with what that line logs to the console.

shuynh84, 2017-03-29:
what javascript file? after I've installed the raspi-io, I can no longer get back into node red :/

nebrius, 2017-03-29:
> what javascript file? after I've installed the raspi-io, I can no longer get back into node red :/

Whichever file is your main file that runs first on app startup. When you run a Node app, you run it like this: `node index.js`, and in this example,`index.js` is the main JavaScript file to be run.

shuynh84, 2017-03-29:
I'm not running any script at the beginning. When i start the raspberry pi and try to run it via root. this is what i do, then it loops and never start node red.

pi@raspberrypi:~ $ sudo su
root@raspberrypi:/home/pi# sudo node-red-start

Start Node-RED
 
Once Node-RED has started, point a browser at http://10.0.0.13:1880
On Pi Node-RED works better with the Firefox browser
 
Use   node-red-stop                          to stop Node-RED
Use   node-red-start                         to start Node-RED again
Use   node-red-log                           to view the recent log output
Use   sudo systemctl enable nodered.service  to autostart Node-RED at every boot
Use   sudo systemctl disable nodered.service to disable autostart on boot
 
To find more nodes and example flows - go to http://flows.nodered.org
 
Started Node-RED graphical event wiring tool..
29 Mar 14:46:55 - [info]
Welcome to Node-RED
===================
29 Mar 14:46:55 - [info] Node-RED version: v0.16.2
29 Mar 14:46:55 - [info] Node.js  version: v6.10.1
29 Mar 14:46:55 - [info] Linux 4.4.50-v7+ arm LE
29 Mar 14:46:57 - [info] Loading palette nodes
pi : TTY=unknown ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/python -u /usr/lib/node_modules/node-red/nodes/core/hardware/nrgpio.py info
pam_unix(sudo:session): session opened for user root by (uid=0)
pam_unix(sudo:session): session closed for user root
29 Mar 14:47:04 - [info] Dashboard version 2.3.6 started at /ui
29 Mar 14:47:07 - [info] Settings file  : /home/pi/.node-red/settings.js
29 Mar 14:47:07 - [info] User directory : /home/pi/.node-red
29 Mar 14:47:07 - [info] Flows file     : /home/pi/.node-red/flows_raspberrypi.json
29 Mar 14:47:07 - [info] Server now running at http://127.0.0.1:1880/
29 Mar 14:47:07 - [info] Starting flows
1490816827816 Device(s) RaspberryPi-IO
1490816827857 Device(s) RaspberryPi-IO
1490816827891 Device(s) RaspberryPi-IO
1490816827944 Device(s) RaspberryPi-IO
29 Mar 14:47:08 - [info] Started flows
29 Mar 14:47:08 - [red] Uncaught Exception:
29 Mar 14:47:08 - Error: EACCES: permission denied, open '/sys/class/leds/led0/trigger'
at Error (native)
at Object.fs.openSync (fs.js:641:18)
at Object.fs.writeFileSync (fs.js:1347:33)
at new LED (/home/pi/.node-red/node_modules/node-red-contrib-gpio/node_modules/raspi-led/dist/index.js:42:18)
at RaspiIOCore.value (/home/pi/.node-red/node_modules/node-red-contrib-gpio/node_modules/raspi-io-core/dist/index.js:525:34)
at RaspiIOCore.pinMode (/home/pi/.node-red/node_modules/node-red-contrib-gpio/node_modules/raspi-io-core/dist/index.js:484:21)
at /home/pi/.node-red/node_modules/node-red-contrib-gpio/node_modules/raspi-io-core/dist/index.js:408:17
at Array.forEach (native)
at /home/pi/.node-red/node_modules/node-red-contrib-gpio/node_modules/raspi-io-core/dist/index.js:337:32
at _combinedTickCallback (internal/process/next_tick.js:73:7)
at process._tickDomainCallback (internal/process/next_tick.js:128:9)
nodered.service: main process exited, code=exited, status=1/FAILURE
Unit nodered.service entered failed state.
nodered.service holdoff time over, scheduling restart.
Stopping Node-RED graphical event wiring tool....
Starting Node-RED graphical event wiring tool....
Started Node-RED graphical event wiring tool..
29 Mar 14:47:10 - [info]

nebrius, 2017-03-29:
There's a main file somewhere, but I suppose Node Red is hiding it under the hood. At this point in time, I'm out of ideas I'm afraid. I'd recommend posting an issue in the node-red GitHub repo and see if they can help.

nebrius, 2017-06-02:
I'm going to close this issue now since there's no forward progress. I hope you were able to figure something out in the Node Red repos!

closed 2017-06-02 by nebrius